arkq/bluez-alsa #286: how to set-up both a2dp-sink and hfp-hf profiles together?

amitkumar718, 2020-01-15:
I m trying to use my pi zero as a headset/handsfree unit. So if connect the pi zero to my android phone, and if I play music on my phone, it play it over the a2dp connection and if I get a call, I able to have the coversation over the SCO connection.

First of all, is this scenario possible with bluealsa today?

I am able to use bluealsa-aplay to recieve music over a2dp. I am also abe to blualsa-aplay to get sound from the voice call (over sco profile). I am also able to separately send voice over sco with
arecord | aplay -D bluealse:xxx...

But I have not been able to get them all working together.  Is it possible at all to get it to work with current/available tools in combination with bluealsa-aplay? Or Do I need to somehow extend bluealsa for it all to work together seemlessly ?

BobGraton, 2020-01-18:
I have an equivalent question (problem) here:
I can send phone music over a2dp connection to the Pi4 (Raspbian Buster, updated):
On the Android phone, "Media audio" is available , but not "Phone audio").

I recompiled bluealsa from source
[https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source](url)
with the following command:
`./configure --enable-aac --enable-ofono --enable-debug --disable-payloadcheck`
it compiles well, no dependency issue detected.

I followed the systemd integration:
[https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration](url)

and in  /etc/default/bluealsa I add
`OPTIONS="-p a2dp-sink -p a2dp-source -p hfp-ofono"`
and here is my
/etc/systemd/system/a2dp-playback.service
allowing connection with bluealsa-aplay service (see below) set with --profile-a2dp, but not with --profile-sco : 
with --profile-a2dp:
`  321 ?        Ss     0:00 /usr/sbin/ofonod -n`
 ` 356 ?        Ss     0:00 /usr/bin/python -u /usr/local/bin/a2dp-agent`
 ` 362 ?        Ssl    0:00 /usr/bin/bluealsa -p a2dp-sink -p a2dp-source -p hfp-ofono`
  `439 ?        Ssl    0:00 /usr/bin/bluealsa-aplay 00:00:00:00:00:00 -d plughw:CARD=v15,DEV=0 --profile-a2dp`

, there is music on the Pi4 coming from Android phone's VLC. No "Phone audio" from phone call.
with --profile-sco:
 `  316 ?        Ss     0:00 /usr/sbin/ofonod -n`
 ` 355 ?        Ss     0:00 /usr/bin/python -u /usr/local/bin/a2dp-agent`
`  358 ?        Ssl    0:00 /usr/bin/bluealsa -p a2dp-sink -p a2dp-source -p hfp ofono`

, bluealsa-play does not start and 
there is no music on the Pi4 coming from Android phone's VLC. No "Phone audio" from phone call.

`[Unit]
Description=A2DP Playback
After=bluealsa.service syslog.service
Requires=bluealsa.service

[Service]
ExecStartPre=/bin/sleep 3
ExecStart=/usr/bin/bluealsa-aplay 00:00:00:00:00:00 -d plughw:CARD=v15,DEV=0 --profile-sco
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=A2DP-Playback
User=root
RestartSec=1
Restart=always

[Install]
WantedBy=multi-user.target`

**Is an "ofono-agent" or "sco-agent" required?
Any direction to look at?**
 
In any case, thanks a lot for developing bluealsa for ALSA only systems!

borine, 2020-01-18:
@BobGraton - a2dp-agent *may* be part of the problem here - it claims to automatically accept a2dp requests but reject all others - so maybe it is preventing the phone from establishing a sco connection to bluealsa. Try without a2dp-agent and instead just connect the phone manually using bluetoothctl.

Then you should at least be at the same stage as @amitkumar718

BobGraton, 2020-01-18:
Thanks a lot for your fast feedback @borine !
I stopped the a2dp-agent and bluealsa-aplay with systemd:
`sudo systemctl disable bt-agent-a2dp.service`
and run bluealsa-aplay manually:
`/usr/bin/bluealsa-aplay 00:00:00:00:00:00 -d plughw:CARD=v15,DEV=0 --profile-sco`
it connects, but no more sound from VLC through "Media audio" on the Pi4. Still no "Phone audio" service visible on android phone.

I have sound when using instead:
`/usr/bin/bluealsa-aplay 00:00:00:00:00:00 -d plughw:CARD=v15,DEV=0 --profile-a2dp` 

Could ofonod be a problem? With and without I did not see any difference...

Arkq, 2020-01-18:
> I am able to use bluealsa-aplay to recieve music over a2dp. I am also abe to blualsa-aplay to get sound from the voice call (over sco profile). I am also able to separately send voice over sco with
arecord | aplay -D bluealse:xxx... Is it possible at all to get it to work with current/available tools in combination with bluealsa-aplay?

No, currently it's not possible. I thought a little bit about such feature, but I will have no use for that, so my focus is in some other areas.

> On the Android phone, "Media audio" is available , but not "Phone audio").

You need `a2dp-sink` and `hfp-hf` profiles enabled (or `hfp-ofono`). You could also use `hsp-hs`, but this profile requires [bluez patching](https://marc.info/?l=linux-bluetooth&m=157403116609637&w=2) and [this commit](https://github.com/Arkq/bluez-alsa/commit/b1ab34ec9498c3ed2c57f1090edb2997dd11a9da).

> Could ofonod be a problem? With and without I did not see any difference...

No, it should not be a problem. You should verify your setup with only `ofonod` running. Because it is sufficient to connect phone in the HFP mode. Then, you can start bluealsa with `hfp-ofono` profile to receive sound from "phone audio". And add `a2dp-sink` for A2DP connection.

BobGraton, 2020-01-19:
Thanks a lot @Arkq and @borine 

ofonod is required and the way to have "Call audio" and "Media audio" visible on the phone,
and the connection to the phone needs to be initiated from bluetoothctl from the Pi, and not the other way round as "Call audio" won't be visible.

I had ofonod running:
`$ ofonod --version`
1.21
and ofono.service running:
`$ systemctl -a |grep ofono`
  ofono.service     loaded    active   running   Telephony service

I stopped bluealsa.service and a2dp-playback.service:
`sudo systemctl stop bluealsa.service`
`sudo systemctl stop a2dp-playback.service`

and using bluetoothctl I paired / trusted / connected my android phone:
**For some reason, connecting the Pi from the phone provided only "Media audio**
**whereas connecting from Pi with bluetoothctl provided both "Call audio" and "Media audio" on the phone.**

So, at that point,
 "Call audio" was visible and enabled on the phone
 "Media audio" was visible but disabled on the phone

I started manually bluealsa with a2dp and ofono profile:
`$ sudo /usr/bin/bluealsa -i hci0 -p a2dp-sink -p hfp-ofono`
(-p hfp-hf resulted in no audio)
and "Media audio" can be enabled.

**Either** starting bluealsa-play with a2dp profile enable Music to be heard from the Pi speaker (using VLC on the android phone):
`$ /usr/bin/bluealsa-aplay 00:00:00:00:00:00 -d plughw:CARD=v15,DEV=0  --profile-a2dp`
**or**  starting bluealsa-play with sco profile enable the call to be heard from the Pi speaker (from incoming call on the android phone):
`$ /usr/bin/bluealsa-aplay 00:00:00:00:00:00 -d plughw:CARD=v15,DEV=0  --profile-sco`

and at that point, I could finally see:
`$ amixer -D bluealsa`
Simple mixer control 'Samsung Galaxy S7 - A2DP',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 127
  Front Left: Capture 127 [100%] [on]
  Front Right: Capture 127 [100%] [on]
Simple mixer control 'Samsung Galaxy S7 - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-j
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 15 [100%] [on] Capture 15 [100%] [on]
Simple mixer control 'Samsung Galaxy S7 | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 255 [255%]

and as @amitkumar718 mentioned, aplay can send audio to the incoming call on the Android phone:
`$ aplay -D bluealsa:SRV=org.bluealsa,DEV=F8:E6:3A:B0:71:14,PROFILE=sco ~/track.wav`

Thank you all for clarification!

borine, 2021-06-06:
My reading of the final comment here is that this issue appears to have been resolved. I am therefore closing it, but please feel free to re-open if necessary

karviss, 2021-06-25:
Hi,
After following all the steps mentioned by @BobGraton in his last comment, I'm able to stream music when I use `--profile-a2dp` but with `--profile-sco`, I'm unable to hear anything on the local speaker or the receiver's phone.
Can it be an issue with my adapters sco audio routing as said in [https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Bluetooth/#hspproblem:thebluetoothsinkandsourcearecreatedbutnoaudioisbeingtransmitted](url)? (I'm not exactly sure what it is) or is it something else that I'm missing!

borine, 2021-06-26:
> Can it be an issue with my adapters sco audio routing

That only applies to USB adapters, and even then bluez-alsa includes a work-around for Broadcom/Cypress . So it depends on what adapter you are using.

karviss, 2021-06-28:
Mine is not a USB adapter. I have a QCA9377.
At first, I tried the HSP/HFP with PulseAudio, but my sink-input is always in the drained state when I'm on calls. A2DP used to work fine. Thought it might be some issue with PulseAudio and switched to Bluez-Alsa. Still, I'm unable to use HSP/HFP.
Can you give me some inputs on what might be the issue!

borine, 2021-06-28:
Sorry - I completed the above edit just seconds too late! SCO routing issues can apply to any bus type.

karviss, 2021-06-28:
> "RX bytes ... sco:0" indicates no incoming packets - my QCA chip does not route SCO to HCI

Thanks for the reply. Seems this is my case.
`hci0:	Type: Primary  Bus: UART
	BD Address: 00:1F:7B:31:80:1B  ACL MTU: 1024:7  SCO MTU: 60:8
	UP RUNNING PSCAN 
	RX bytes:173828134 acl:289931 sco:0 events:3294 errors:0
	TX bytes:86776 acl:3204 sco:0 commands:60 errors:0
`
How did you resolve your issue with QCA chip?

borine, 2021-06-28:
Actually, I've just checked, and my on-board BT adapter is also a QCA9377, but my device does use the USB bus for its HCI:
```shell
$ lsusb | grep Atheros
Bus 001 Device 003: ID 0cf3:e500 Qualcomm Atheros Communications
```
I have been unable to persuade it to route incoming SCO data to the HCI, although outgoing (ie ALSA playback) does work fine with CVSD codec. I never managed to find any online info to help fix the SCO routing, but haven't looked at it for over a year now (I gave up and bought an external USB adapter that works great). If you do find anything, please post links here as it may help may others.

karviss, 2021-06-28:
Thanks for the reply.
> "RX bytes ... sco:0" indicates no incoming packets - my QCA chip does not route SCO to HCI

Seems that's my case too.

hci0:	Type: Primary  Bus: UART
	BD Address: 00:1F:7B:31:80:1B  ACL MTU: 1024:7  SCO MTU: 60:8
	UP RUNNING PSCAN 
	RX bytes:173828134 acl:289931 sco:0 events:3294 errors:0
	TX bytes:86776 acl:3204 sco:0 commands:60 errors:0

borine, 2021-10-30:
I think this issue is resolved, in as far as the hardware in use allows. Please see the wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices for advice on using HFP/HSP with BlueALSA.

closed 2021-10-30 by borine