arkq/bluez-alsa #287: How to switch audio output (with BlueAlsa) from Bluetooth to analog in GUI

rozhasi, 2020-01-21:
I'm using Mate Desktop with Raspbian Buster on Raspberry Pi 4B. PulseAdio causes all audio output via the on-board analog audio output to be mono. Therefore, I uninstalled PulseAudio completely and am using ALSA exclusively. For the GUI volume applet in System Tray I'm using **_pnmixer_** because the standard **_mate-volume-control-applet_** can't control the volume of the ALSA-based on-board analog audio on the Raspberry PI 4B for whatever reason. Everything works fine with the on-board audio output with _Mate / Raspbian Buster_and **_pnmixer_**. I can control volume of the on-board analog output using the slider in **_pnmixer_**. I can also control the volume using _**alsamixer**_. 

My next goal is to get Bluetooth audio working under Mate with Raspbian on the Raspberry Pi 4. I downloaded **_BlueALSA_** and was able to get the Bluetooth audio working with the Bose Bluetooth headphones. The problem I'm having is that I cannot figure out a way to switch from the Bluetooth headphones to the on-board audio. With the _.asoundrc_ config listed below, both **_pnmixer_** and **_mate-volume-control-applet_** see both the Raspberry Pi on-board audio and the Bluetooth headphones, but neither utility can switch from the Bluetooth headphones to the on-board analog audio. Therefore, with the _.asoundrc_  in place, I cannot send audio out of the on-board analog audio jack. If I remove the _.asoundrc_ file then the on-board analog audio works fine, but the Bluetooth headphones are not working and neither are they visible in **_pnmixer_** or in **_mate-volume-control-applet._** even when they are connected and visible in _Blueman_.  I need help with this last piece of the puzzle to be able to switch from the Bluetooth headphones to the analog audio and back. Simply disconnecting the Bluetooth headphones doesn't switch the audio to the onboard analog audio either. Thank you!

`
pcm.!default {
    type plug
    slave.pcm {
         type bluealsa
         device "2C:41:A1:EB:9B:C9"
         profile "a2dp"
    }
}
pcm.onboard {
   type hw
   slave.pcm {
        card 0
   }
}
ctl.!default {
   type bluealsa
}

ctl.onboard {
   type hw
   card 0
}

`

borine, 2020-01-31:
pnmixer appears to produce its list of available cards by iterating over all the hardware cards and then adding "default". So I guess the only way to get it to recognize bluealsa is to redefine "default" to be bluealsa. Your .asoundrc file does that, so I would expect if you select "default" as the output then sound will go through bluealsa. If the hw devices no longer work with your .asoundrc, then as a first test I would remove everything from .asoundrc *except* the pcm.!default and ctl.!default entries and see how that works.

rozhasi, 2020-02-02:
@borine 

I removed the following configuration from the .asoundrc file: 

pcm.onboard {
type hw
slave.pcm {
card 0

ctl.onboard {
type hw
card 0
}

However, it made no difference whatsoever. pnmixer and mate-volume-control still see both the (default) (which is Bluetooth headphones) and bcm2835  ALSA. When I restart the Raspberry Pi, as soon as Bluetooth headphones connect, applications can stream music to the headphones. If I disconnect the Bluetooth headphones (e.g. by switching them off), Youtube starts the spinning circle. If I close Chromium, reopen it, and start playing another video, I get the following error message in Youtube: 
Audio Render error. Please restart your computer. 

With the .asoundrc file in place (the one I posted earlier or the one with the pcm.onboard and ctl.onboard removed, I can't get audio playing out of the analog speakers even if I keep Bluetooth Headphones off upon the Raspberry Pi boot. In order for me to restore the audio out of the analog speakers, I have to remove the .asoundrc file. After I reboot, the sound plays out of the analog audio speakers, but it's no longer possible to play audio out of the Bluetooth headphones. 

I don't think this issue is with pnmixer per se. I think this is something on a lower level. 

Any help would be greatly appreciated.

borine, 2020-02-03:
> Therefore, I think this issue is not with pnmixer but at a lower level.

Yes, I guess Mate depends on PulseAudio to manage audio routing. By removing pulse, I think you have removed that ability. I don't think gsettings would help here. AFAICT pnmixer merely operates the control elements of the hardware cards (and "default" card); it does not manage any pcm elements, which is what you would need to perform audio routing.

You might have more luck finding help on a Mate Desktop forum, since I think your issue boils down to "how do I do manage audio pcm routing on Mate Desktop without PulseAudio" and is not directly related to Bluealsa.

rozhasi, 2020-02-03:
Well, that may be, but setting aside the mate desktop environment, how would I switch audio without using GUI utilities. It doesn’t seem to me that I can direct audio output out of the analog audio jack when I have an .asoundrc file. I would like to be able to reliably switch audio output bypassing GUI utilities to rule out the problem elsewhere before I focus on the Mate desktop. 

Raspberry Pi developers had to write their own audio plugin for the lxpanel to switch audio output to Bluetooth. Once they did it (I think it happened in Stretch), they stopped installing PulseAudio in their Raspbian distros. Before they developed that plugin, they had to utilize PulseAudio to route audio to connected Bluetooth devices. 

I can’t use their plugin with lxpanel in the Mate DE because it installs into the LXDE system tray, and Mate has its own System Tray, so there’s a conflict preventing me running lxpanel with a LXDE system tray plugin in it. I’m not sure why Raspbian Developers had to write their own plugin for switching audio between the onboard adapter and Bluetooth devices - if this had something to do with LXDE not being able to handle that sort of switching or if this is a lower-level issue independent of the DE running on Raspbian (Debian). 

In my case, when running Mate over Raspbian Buster, two different GUI sound mixing utilities (on mixer and mate-volume-control) can see both the onboard adapter and the Bluetooth he headphones. Moreover, alsamixer also sees both the onboard adapter and Bluetooth headphones. If, I press F6 in alsamixer and type I’m “default” I get access to the Bluetooth headphones where I can control the volume of either channel as well as the recording levels from the Bluetooth headphones microphone. Alsamixer is not part of the Mate DE, so at s lower level, both types of audio output are visible. The problem is only with Routing the sound out of the analog audio jack when the .asoundrc duke is present. 

I don’t have enough knowledge to solve this problem, and I’ve spent many many days on this already. Linux is not my area of expertise, but I’m an engineer in a different field, and with my decades long engineering And troubleshooting experience, it doesn’t appear that the issue is on the Mate DE per se until I can prove that I can route the sound successfully without using GUI utilities. If I can do it without GUI utilities, I will then shift to the troubleshooting Mate DE.

borine, 2020-02-03:
Let me walk you though the typical linux desktop audio configuration. Most Gtk based GUIs use PulseAudio as the audio server, while Qt based ones generally seem to use Artsd. Either way, the basic architecure is the same. I'll assume pulseaudio is used in this note.

- Every application that is integrated into the desktop (including the browser, video and music players etc) is configured to use the alsa "default" pcm for audio output.

- The system-wide alsa configuration defines "default" to be "pulse" which causes all applications to send their audio to the pulseaudio server.

- the pulseaudio daemon is by default configured to mix the audio from all applications together then send the result to the first ALSA hardware pcm device (hw:0,0 in ALSA parlance).

- client applications such a pavucontrol etc can change the pulse daemon settings to send the input from different combinations of applications to different output devices.

If you disable the pulseaudio service, the applications still send audio to ALSA's "default" pcm, but that is no longer defined to be "pulse". Typically it will be "hw:0,0" (probably also through "plug" and "dmix" first, but that too much detail for this discussion), but that will depend on the local alsa configuration. You can override the default in your own .asoundrc file. Note that there is now no "routing" going on - "default" is fixed in the alsa configuration.

So now, in your case with Mate on Rasbian Buster, with no .asoundrc the GUI applications send their audio to pcm "hw:0,0" which is the 3.5mm jack . All good.

With your .asoundrc, GUI applications now send audio to the bluetooth speaker via bluealsa. All good.

Remember, the browser has been configured by the distribution to use "default" for output so any other pcms you define in there make no difference to it. To get the browser (or other application) to use the jack plug you would have to change the definition of pcm.!default in your .asoundrc and possibly re-start the application to clear its alsa config cache so it reads the new config. I *think* this is what the RPi desktop audio management application does.

Desktop "mixer" applications only operate on the control elements of the cards. So you can happily change the volume on the jack plug and bluealsa, but you cannot force the browser to direct its audio output to the jack using the mixer.
 
Your desire to switch between different audio outputs while an application is running *absolutely* depends on having an audio daemon as intermediary, unless  the application itself provides some interface to select output devices - but then you would have to make the change in each application individually,

So this is not a bluealsa issue - if you wanted to switch between 3.5mm jack and HDMI you would still have exactly the same issue.

I realise this is not the answer you wanted, so I did a quick search on pi4 stereo audio and found this post which apparently has a solution using pulseaudio:
https://www.raspberrypi.org/forums/viewtopic.php?t=251455

rozhasi, 2020-02-03:
@borine 

Thank you very much for the explanation. It is one of the best technical posts I’ve read in a long time. 

Is it possible to create an .asoindrc file (or another way) that would make the onboard adapter to become the “default” when the Bluetooth headphones are not connected? 

Basically, the scenario is as follows:

1. When the Bluetooth headphones are connected, “default” is the Bluetooth headphones. 

2. When the Bluetooth headphones are disconnected, “default” is the onboard audio adapter. 

PilseAudio causes mono output out of the analog jack for some reason, so I uninstalled it for that reason.

borine, 2020-02-04:
> Is it possible to create an .asoindrc file (or another way) that would make the onboard adapter become “default” when the Bluetooth headphones are not connected?

Sure, bluealsa emits D-Bus signals "PCMAdded" and "PCMRemoved" and you could write a script / program that listens for these and modifies .asoundrc as appropriate. You would have to restart each application to get it to read the new config though. Also, I didn't mention this in my summary above, but some applications may have been configured to send audio directly through pulseaudio's native API, by-passing alsa entirely. For those, removing pulseaudio will have broken them. If you use such an appllcation, it may be possible to modify its runtime config file(s) to use alsa instead, but if you are really unlucky you might have to rebuild it from source - it just depends how the distribution has chosen to build and install it.

> PilseAudio causes mono output out of the analog jack for some reason, so I uninstalled it for that reason.

The link I gave above claims that the pulseaudio mono/stereo issue on RPi can be fixed with a simple change to one pulseaudio configuration file. I don't know if that is true, and I realise you asked for help not opinion, but I have to say that fixing a broken pulseaudio installation sounds mush easier than implementing a new, bespoke, audio management system from scratch. If your only desired outcome is to have a fully functioning Mate desktop then surely its worth giving that fix a try?

borine, 2020-02-04:
@rozhasi if you are still determined to do without pulseaudio, here is an example script to detect bt audio device connect/disconnect events and invoke a handler for each. It needs error handling before it is fit for production use,  but gives you the basics:

```
#!/bin/bash

handle_add_event() {
	# put your speaker connected routine here
	echo "Bluetooth device $1 connected to bluealsa"
}

handle_remove_event() {
	# put your speaker disconnected routine here
	echo "Bluetooth device $1 disconnected from bluealsa"

}

# create a temporary named pipe to communicate with dbus monitor
PIPE=$(mktemp -u)
mkfifo $PIPE
# attach it to unused file descriptor FD
exec {FD}<>$PIPE
# unlink the named pipe
rm $PIPE

# make sure the pipeline is shut down if this script interrupted
trap "kill %1; exec {FD}>&-; trap - EXIT; exit 1" INT
trap "kill %1; exec {FD}>&-" EXIT

# start dbus monitor in background
gdbus monitor --system --dest org.bluealsa >&$FD &

# sed command to extract the device address from the dbus signal
SED_CMD='s!.*/org/bluealsa/hci0/dev_\(.*\)/a2dp.*!\1!;s/_/:/g'

# now listen for connect/disconnect signals
while read 
do
	case "$REPLY" in
		*PCMAdded*a2dp*)
			BTADDR=$(echo "$REPLY" | sed "$SED_CMD")
			handle_add_event "$BTADDR"
			;;
		*PCMRemoved*a2dp*)
			BTADDR=$(echo "$REPLY" | sed "$SED_CMD")
			handle_remove_event "$BTADDR"
			;;
	esac
done  <&$FD
```

rozhasi, 2020-02-04:
@borine 

Thank you very much again for the time and effort you've given to this issue. 

I followed your advice and installed PulseAudio and fixed the lack of stereo output by commenting out the entire [Mapping analog-mono] section of the file /usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf

That fixed the problem of the lack of stereo with PulseAudio installed on the Raspberry Pi 4B. 

Then, I uninstalled bluealsa, rebooted the Raspberry Pi 4B, and was able to play audio from Youtube (via Chromium) to the Bluetooth headphones. I launched mate-volume-control, which is a utility/SystTray plugin that comes with the Mate desktop. I've been told that this is a stripped-down version of pavucontrol. 

mate-volume-control could see both the Bluetooth headphones and the on-board analog audio adapter. However, even though there is what appears to be a switcher between the audio output devices in mate-volume-control, I couldn't switch the music playing in Youtube via Chromium from the Bluetooth Headphones to the analog audio jack or vice versa (while the music was playing). I had to switch the output device, then close the Chromium tab, then reopen the Chromium tab and start playing the music again. Only then, the music would be playing through the device that I selected in mate-volume-control. In other words, the audio-device switcher in mate-volume-control couldn't switch the audio output in real time. 

I found another utility called Sound Switcher Indicator, which can switch the audio output in real time. This is also a utility/SysTray plugin, which automatically gets launched at startup (once the Raspberry Pi 4B is restarted upon its installation). Now I can switch the audio output in real time without having to close a Chromium tab . 

I'm very grateful for all of your advice. Even though I had spent about a week googling, trying to find out why PulseAudio was causing Mono output, I didn't find the thread that you linked to here. 

Thanks again!

fabianmendes, 2021-01-19:
> @borine
> 
> Thank you very much again for the time and effort you've given to this issue.
> 
> I followed your advice and installed PulseAudio and fixed the lack of stereo output by commenting out the entire [Mapping analog-mono] section of the file /usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf
> 
> That fixed the problem of the lack of stereo with PulseAudio installed on the Raspberry Pi 4B.
> 
> Then, I uninstalled bluealsa, rebooted the Raspberry Pi 4B, and was able to play audio from Youtube (via Chromium) to the Bluetooth headphones. pnmixer couldn't see the Bluetooth headphones as an audio output; it could only see the on-board analog adapter, but mate-volume-control could see both the Bluetooth headphones and the onboard analog audio adapter, so I uninstalled pnmixer.
> 
> mate-volume-control, which is a utility and a System Tray plugin, comes with the Mate desktop. I've been told that this is a stripped-down version of pavucontrol.
> 
> However, even though there is what appears to be a switcher between the audio output devices in mate-volume-control, I couldn't switch the music playing in Youtube via Chromium from the Bluetooth Headphones to the analog audio jack or vice versa in real time. After I would switch the output device, I had to close the Chromium tab, then reopen the Chromium tab and start playing the music again. Only then, would the music be playing through the device that I selected in mate-volume-control.
> 
> Fortunately, I found another utility called Sound Switcher Indicator, which can switch the audio output in real time. This is also a utility/System Tray plugin, which automatically gets launched at startup once it's installed and the Raspberry Pi 4B is restarted. Now I can switch the audio output in real time without having to close a Chromium tab by switching the audio output device in Sound Switcher Indicator.
> 
> I'm very grateful for all your advice. Even though I had spent about a week researching the lack of stereo with PulseAudio on the Raspberry Pi 4B, I didn't find the thread with the solution that you linked to here. Hence, I had to uninstall PulseAudio just to be able to play music in stereo out of the analog jack in Raspberry Pi 4B. Now, with PulseAudio installed and the fix for the missing stereo implemented, the Raspberry Pi 4B running Raspbian with Mate DE finally has the proper sound via both the analog audio jack and via Bluetooth.
> 
> Thanks again!

I didn't get it, how did you fix it? Should I install PulseAudio now? My Bluetooth headphones are paired and connected but I can't listen to anything until "I select it" as output. I can't do that, the new Alsa volume controller doesn't have the option to select output anymore, and it says Cards, nothing more... it's annoying

borine, 2021-01-19:
@fabianmendes selecting an audio output is not something that bluealsa is responsible for - that is the job of your desktop software. The original poster was using the Mate desktop, which is built around pulseaudio as its audio server, so I recommended that he use pulseaudio.  I would recommend that you stay with whatever your chosen desktop uses. If the desktop is not working as expected, then your first call for help should be with your Linux distribution support.

If you have assembled your own desktop, then you need to provide your own audio management components. The ALSA project does not provide an audio server, and instead relies on static configuration files. Therefore changing system wide defaults while applications are running is exceedingly difficult, maybe even impossible, when using ALSA directly without an audio server to manage dynamic configuration changes. Unfortunately there is not much choice when it comes to linux audio servers, `pulseaudio` and `pipewire` are the only two I am aware of the are now actively maintained.

If you are using the RaspberryPiOS (formerly Raspbian) desktop supplied by the Raspberry Pi foundation, then that recently changed from using its own scripted audio management solution to using pulseaudio. As part of that change, bluealsa was dropped from the image file.

So, in summary, for desktop systems, stay with pulseaudio (or pipewire) for audio unless you are an expert and really know what you are doing. For headless and embedded systems where applications interact directly with ALSA devices, then bluealsa provides the most efficient bluetooth audio solution.

closed 2021-06-06 by borine