arkq/bluez-alsa #288: Difference between Bluez-alsa and bluealsa package in Raspbian stretch

baahver, 2020-01-29:
On my raspberry pi I have installed raspbian and on that the bluealsa package. This is working reasonable well. The package version is: 0.13. 
Has this package anything to do with Arkq/Bluez-alsa on Github?
The version here is 2.10.

rozhasi, 2020-01-29:
I'm pretty sure that bluealsa in Raspbian is bluez-alsa.

baahver, 2020-01-29:
And do you know how these versions compare? 
0.13 from raspbian repo and 2.10 on github.

rozhasi, 2020-01-30:
> And do you know how these versions compare?
> 0.13 from raspbian repo and 2.10 on github.

Maybe I should change my answer. This project is BlueAlsa. BlueZ used to integrate with ALSA directly, but as of version 5, BlueZ removed integration with ALSA. Starting with version 5 of BlueZ, PulseAudio was required for BlueZ to work in Linux for streaming to Bluetooth. 

BlueZ-ALSA is a project that restored direct integration between BlueZ and ALSA. It was later renamed to BlueALSA. 

Let me ask you a question. Are you running PulseAudio in Raspbian? If so, you don’t need BlueALSA. If you are not running PulseAidio, then you need BlueALSA. 

I’m running Raspbian Buster without a PulseAudio. I am having a problem switching from Bluetooth audio to analog audio in Raspbian Buster running Mate desktop.

baahver, 2020-01-30:
I was using pulseaudio earlier, but had a lot of trouble getting the bluetooth speaker to work in a stable waY. Since I'm using bluealsa, the bluetooth connection is stable and setup is almost always working as expected. 
Still do not understand the difference between 0.13 in debian/raspbian buster repo and 2.10 in github.

rozhasi, 2020-01-30:
I don't know the difference myself. Is there a reason you don't want to use version 0.13 in Raspbian Buster? 

I myself am looking for some help here but not getting any responses to my question #287. I am using Raspbian Buster with the Mate desktop environment instead of the standard Raspbian desktop environment. I'm having a problem trying to switch from the Bluetooth audio output to the Raspberry Pi 4B on-board analog audio jack. My .asoundrc file is posted in question #287. 

What desktop environment are you using with Raspbian Buster? Are you having any problems switching from Bluetooth audio to on-board analog audio? Would you mind posting your .asoundrc config here? 

Thank you!

baahver, 2020-01-30:
I switched from 0.13, because I suddenly got problems with bluetooth audio. I started bluealsa with the systemd method. But another config as described in the wiki on this github. 
Now I switched to this version, build from source, and the systemd config as described here. 
maybe the change of systemd config  would have repaired my earlier problems. But for now I am leaving it this way.

I do not use a desktop environment. I have a headless raspberry pi with domoticz. I use bluetooth speaker to ring a bell or give another message.

borine, 2020-01-31:
Raspbian maintain their own source tree for bluez-alsa here:
https://github.com/raspberrypi-ui/bluealsa/

As you can see, that is not a proper "fork" of the original codebase here, so it makes it **very** difficult to see where they are up to in terms of the original development by @Arkq here and what patches (if any) they have applied. The last update in the raspbian source was on the 13th September 2019, and the commit message refers to commit 4c32fc30068a4d6faca462d8e7fbc64e3dec4c0b in this repository. Note that 0.13 is a raspbian version name and tells you nothing about any release version from this repository.

So that's how raspbian 0.13 relates to this project - if is basically an out-of-date clone from some random commit last year with possibly some other changes applied. You would have to contact the maintainer of that project if you need more specific info.

baahver, 2020-01-31:
Thank you Borine, for explaining this to me. I'll stick to Arkq's version then!
It's stable here and working very well.

closed 2020-02-17 by Arkq