arkq/bluez-alsa #289: bluealsa amixer usage

roizcorp, 2020-01-29:
I have a soundcard I am controlling with amixer (volume control, routing etc.)
I'm trying to send line in into BT speaker but I cannot identify bluealsa as soundcard with aplay -l nor amixer

any advice?

rozhasi, 2020-01-30:
You need to create the asoundrc configuration file. Look at my question for a a sample of that file.

roizcorp, 2020-01-30:
hi,
It seems that your configuration include 2 cards bluealsa and onboard something, right?
well I do have asoundrc conf file which works meaning I am streaming music from MPD into speaker,
However though my use is to take a stream from other card e.g. cardA and by using amixer to route it into bluealsa. 
Amixer shows only cardA while my asoundrc file states only bluealsa

borine, 2020-01-31:
> I cannot identify bluealsa as soundcard with aplay -l nor amixer

aplay -l does not show soundcards, it only shows pcm devices; even then it only shows pcm devices that are defined in its static configuration, it does not contact bluealsa or any other sound daemon to enquire about dynamic devices such as bluetooth speakers. So if you add a pcm device definition for a bluetooth device to your .asoundrc file then `aplay -l` will list it, even if it is not connected. 

amixer without arguments shows only controls on the "default" card. To see controls on the bluealsa mixer you need to run amixer as:
`amixer -D bluealsa`

> my use is to take a stream from other card e.g. cardA and by using amixer to route it into bluealsa.

I don't understand how you expect amixer to "route" audio. You need an application that can capture audio samples from a pcm device on cardA and play them back on a bluealsa pcm device. You say you are using MPD, well that application is able to read from one soundcard and playback to another, so you could use that. Another possibility is to use `arecord` for capture and pipe the output to `aplay` for playback. There are many other alternative ways. Consult the manual for your chosen application for advice on how to achieve your requirement.

roizcorp, 2020-02-01:
thank you, `amixer -D bluealsa` comes up with an error.
`amixer -D bluealsa
ALSA lib bluealsa-ctl.c:956:(_snd_ctl_bluealsa_open) Unknown field interface
amixer: Mixer attach bluealsa error: Invalid argument`

My audio card can route audio from input (i.e. mic or line in) into output with the appropriate `amixer`command, it is really neat. you can also make the sound through some filters to clean up noise or to amplify it

The use case is as follows, this is a car music computer solution. Normally in my car (or any decent car) I use the analog out put to play the music in the car's speakers, while the phone sounds (mainly GPS navigation) are streamed via Bluealsa into the computer and then joint stream the audio into the analog output so I can hear both the music and the phone - good thing about BT is when there is no sound it turns into mute so there is no disruption to the sound quality.

However though, I travel for long periods and have long drives so I take the music solution with me. there is a trend in EU cars not to provide solely BT connectivity (which sucks). Bluealsa again here is extremely useful allowing me to stream music directly from the computer into the car (in the past I used the phone as mediator to car audio system but since the "3.5mm jack is out trend" android set BT as realtime priority in the CPU which literally kills the phone).

I think I figured the issue, as long as I am streaming music from the computer (via bluealsa) to the car's BT, I cannot "append" additional stream to it - meaning I have to choose whether it would be music (MPD) or the phone (because if I stop MPD from using bluealsa output it works) - which is the core downfall of ALSA.
What I wanted originally is to route the phone into the bluealsa (via amixer) so the computer "would think" it is a single audio stream

Arkq, 2020-02-01:
> `bluealsa-ctl.c:956:(_snd_ctl_bluealsa_open) Unknown field interface`

If you've upgraded from <2.0.0 to >=2.0.0 you will have to update [ALSA configuration](https://github.com/Arkq/bluez-alsa/blob/master/src/asound/20-bluealsa.conf) file as well. Note, the location of this file depends on the version of ALSA library. Also make sure, that your custom asoundrc files does not contain line `interface "hci0"`, because right now interface is selected by bluealsa server instance, which you have to select via `service "org.bluealsa"` in asoundrc (or keep default if file 20-bluealsa.conf is properly installed).

roizcorp, 2020-02-01:
I haven't upgrade yet...still on 1.4.0
this is my asoundrc below, and I also have the 20-bluealsa.cong installed in /`etc/alsa/conf.d/20-bluealsa.conf`

`defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "AB:CD:EF:12:34:56"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000

pcm.bluealsa {
      type plug
      slave.pcm {
              type bluealsa
              device "AB:CD:EF:12:34:56"
              profile "a2dp"
      }
}

ctl.bluealsa {
        type bluealsa
        interface "hci0"
}
`

borine, 2020-02-01:
> What I wanted originally is to route the phone into the bluealsa (via amixer) so the computer "would think" it is a single audio stream

That's the thing I think amixer cannot do. Your card may be able to combine multiple inputs from its own connectors to one of its outputs, but it cannot directly mix into bluealsa without some application to move the audio data from the card to bluealsa.

It would be nice if alsa's dmix plugin would accept bluealsa as a slave but I think dmix only accepts hardware devices as slaves. So we are missing that middle component that could take input from mpd and the soundcard then combine them into a single stream for delivery to bluetooth for output. I really hate to suggest this on the bluealsa project site, but just maybe what you need is pulseaudio?

borine, 2020-02-02:
Ok, so clearer head this morning, apologies for my last comment and hopefully I can make more sense now.

You've already said you have a hardware mixing card and want to use that to combine the audio streams. So I assume the card has loopback capability so that audio sent to it from the computer can be read back in directly by the computer.

Is that correct? How does the loopback show up as a device - is it just hw:0,2 or is there some other way to identify it in alsa? If you could post the make and model of the sound card I will find time to read up on it and try to understand how this works better.

So assuming the card is hw:0, the line in device is hw:0,1, and the loopback mixer device is hw:0,2, then I guess you configure MPD to output to hw:0,2 (ie directly to the mixer device), and somehow using amixer or alsamixer configure the card to also route audio from hw:0,1 (line-in) to hw:0,2.

Then what's missing is an application to transfer the samples from the card to bluealsa, such as \:

`alsaloop -C hw:0,2 -P bluealsa`

 or

`arecord -D hw:0,2 -r 48000 -c 2 -f S16_LE | aplay -D bluealsa`

Is this any closer to reality?

[
btw in your .asoundrc above, I think you should include an exclamation mark ! in the bluealsa pcm and ctl definitions, since these are overriding the definitions in /etc/alsa/conf.d/20-bluealsa.conf:

```
pcm.!bluealsa {...}
ctl.!bluealsa {...}
```
]

roizcorp, 2020-02-03:
Thank you very much for taking your time and thought into this, I really appreciate it!

> btw in your .asoundrc above, I think you should include an exclamation mark ! in the bluealsa pcm and ctl definitions, since these are overriding the definitions in /etc/alsa/conf.d/20-bluealsa.conf:
> 
> ```
> pcm.!bluealsa {...}
> ctl.!bluealsa {...}
> ```
> 
will do! 

> but just maybe what you need is pulseaudio?
no pulse, this is a HiFi audio card, and songs so it would be such a shame to throw this at pulse. I understand that streaming the music via BT downgrades the audio quality from HiFi but this is only the secondary use of the configuration (main is analog) so I'd rather keep pulse out of the picture entirely.

I will try the loopback proposal, the only thing I need to mention is that computer's BT would be the same one that will receive the audio from the phone and transmit it into the audio system, another challenge is that this BT is 2.1 dongle so I hope it would cope with the load of the bandwidth of both in and out streams (it handles each one them individually quite well to my surprise)

roizcorp, 2020-02-14:
So I did upgrade to 2.1.0 and as expected, nothing works..meaning that I do get the BT and the BT speaker to connect but bluealsa does not register as playback device...so I cannot stream anything to it, the same configuration which worked on 1.4.0 now fails
I did change the .asoundrc file and removed the `interface hci0` as suggested

roizcorp, 2020-02-20:
I'm going to close this issue, I did a bunch of work which not necessarily related soley to bluealsa

 thanks for the help!

closed 2020-02-20 by roizcorp

Asikbct, 2024-03-01:
Hi,
I have a same issue. And my requirement is to redirect the audio from hw:0,1 to bluealsa. From that moment, Audio should not come through hw:0,1. Its like bluealsa needs to act as virtual plugin at runtime.
**bluealsa version :** 4.1.1
For example,
--> If other music sources writes audio on hw:0,1, then if BT got connected it should redirect audio to bluealsa plugin without disturbing (closing pcm and reopen snd_pcm_open() with bluealsa plugin) other music services whoever writing audio.
--> **alsaloop -C hw:0,2 -P bluealsa or arecord -D hw:0,2 -r 48000 -c 2 -f S16_LE | aplay -D bluealsa** 
    I have tried above steps, I could hear audio mixed with noise. What is the issue here ? 
    
  I am new to Alsa and PCM . Any help will be appreciated. Thanks