arkq/bluez-alsa #291: Raspberry PI MPD bluealsa volume control

imsunburnt, 2020-02-04:
I am trying to find out how to control volume for a Bluetooth speaker with the MPD mpc command.  Everything works but I can only control the volume with amixer:

amixer -D bluealsa sset "TP-LINK_Music 1 - A2DP" 70%

I would like to control volume with:     mpc vol 70.

Here is my setup:
Raspberry PI 3b+, buster lite 2019-09-26
bluealsa --version:  1.4.0
bluetoothd --version:   5.50
aplay --version:   1.1.8 by Jaroslav Kysela <perex@perex.cz>
mpd --version:   Music Player Daemon 0.21.5 (0.21.5)

.asoundrc:
pcm.!default plug:tplink1
pcm.!tplink1 {
        type plug
        slave.pcm {
                type bluealsa
                device "18:D6:C7:11:E5:EF"
                profile "a2dp"
        }
}

mpd.config:
audio_output {
       type            "alsa"
       name            "TPLINK 1"
       device          "bluealsa:DEV=18:D6:C7:11:E5:EF,PROFILE=a2dp"
       mixer           "software"
       format          "44100:16:2"
}

I have searched for a solution to no avail.

borine, 2020-02-05:
mpd config has no audio output option called "mixer". see:
https://github.com/MusicPlayerDaemon/MPD/blob/v0.21.x/doc/user.rst#configuring-audio-outputs

imsunburnt, 2020-02-07:
Thank you very much!  Changed the entry in /etc/mpd.conf to the following now allows the mpc vol command to control the Bluetooth volume:

audio_output {
type            "alsa"
name          "TPLINK 1"
device         "bluealsa:DEV=18:D6:C7:11:E5:EF,PROFILE=a2dp"
mixer_type  "software"
format         "44100:16:2"
}

closed 2020-02-07 by imsunburnt