arkq/bluez-alsa #292: aplay works with -r 8000, with other rates speaker just switches off

demetrio108, 2020-02-05:
bluez-alsa-git 2.1.0.r1.gbacbf5d-1

Strange issue I have. Aplay successfully plays sound on speaker with default settings (cat /dev/urandom | aplay -D bluealsa; cat /dev/urandom | aplay -D blualsa -c 2 -r 8000 # This both work). But any other rate - speaker just switches off. All other apps trying to play with 44100 rate, so they don't work as well. How can be this fixed?

Arkq, 2020-02-05:
Are you using sco or a2dp? Please, post here the result of amixer -D bluealsa

demetrio108, 2020-02-05:
A2DP

Simple mixer control 'charge 2+ - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 127 [100%] [on]
  Front Right: Playback 127 [100%] [on]

demetrio108, 2020-02-05:
Also I dumped bluetooth traffic when connecting to speaker from my phone, noticed supported codecs (mSBC and CVSD):
[image]

But in bluetooth.service logs they are different (SBC and MPEG24):
[image]

Or maybe I'm misunderstanding something.

demetrio108, 2020-02-19:
Ok, I've solved the problem by buying another speaker :)

closed 2020-02-19 by demetrio108