arkq/bluez-alsa #293: a2dp source: how to confirm AAC is being used?

macmpi, 2020-02-06:
Hi,
I'm using a PiZeroW as a2dp source to a Bose Speaker.
I compiled bluealsa with AAC support (no debug option), but I'd like to verify/confirm this codec is indeed actually being used in place of SBC.
How can I confirm that?

I did not find an option (in bluealsa parameters or .asoundrc) to enforce it either.

Thanks!

Arkq, 2020-02-06:
See this issue: #283

macmpi, 2020-02-06:
Thanks for the feedback, sorry I missed it in my search.

Here is what I get from dbus command (removed MAC):
`({objectpath '/org/bluealsa/hci0/dev_XX_XX_XX_XX_XX_XX/a2dp': {'Device': <objectpath '/org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX'>, 'Modes': <['source']>, 'Format': <uint16 32784>, 'Channels': <byte 0x02>, 'Sampling': <uint32 48000>, 'Codec': <uint16 0>, 'Delay': <uint16 150>, 'Volume': <uint16 32639>}},)`
So it turns-out that it connects with SBC

I checked Speaker capability with `hcidump avdtp` as explained in related links.
```
HCI sniffer - Bluetooth packet analyzer ver 5.52
device: hci0 snap_len: 1500 filter: 0x400
< AVDTP(s): Discover cmd: transaction 0 nsp 0x00
> AVDTP(s): Discover rsp: transaction 0 nsp 0x00
    ACP SEID 1 - Audio Sink
    ACP SEID 5 - Audio Source
    ACP SEID 6 - Audio Source
< AVDTP(s): Set config cmd: transaction 1 nsp 0x00
    ACP SEID 1 - INT SEID 3
    Media Transport
    Media Codec - SBC
      48kHz
      JointStereo
      16 Blocks
      8 Subbands
      Loudness
      Bitpool Range 2-51
> AVDTP(s): Set config rsp: transaction 1 nsp 0x00
< AVDTP(s): Open cmd: transaction 2 nsp 0x00
    ACP SEID 1
> AVDTP(s): Open rsp: transaction 2 nsp 0x00
```
Which is somehow consistent, but I'm very surprised, as I thought this product (Bose Portable Home Speaker) was supposed to be AAC and AptX compatible, which do not show up here.

Arkq, 2020-02-07:
Maybe you have to enabled it somehow. The same thing is with Sony WH-XM headphones, you have to enable audio quality over connection stability by proprietary app. Otherwise you have SBC only.

macmpi, 2020-02-09:
Thanks. Not possible unfortunately.
More generally, if several advanced codecs are available (AAC, AptX,...) can bluealsa set a priority order of some sort to select one over another?

Arkq, 2020-02-09:
> can bluealsa set a priority order of some sort to select one over another?

Nope, it's not possible I'm afraid. Currently, bluealsa tries to connect best codec at first. If none of AptX, LDAC, AAC can be connected, then SBC will be used. However, this behavior is true only for up to two BT devices connected to bluealsa. If you connect 3rd device, selected codec might be different.... I don't know how to force BlueZ to select codec in a specified order. In the future, it will be possible to change A2DP codec on runtime via BlueALSA D-Bus API :)

macmpi, 2020-02-09:
Thanks, very interesting.

> bluealsa tries to connect best codec at first. 

If all three AptX, LDAC, AAC are available, which is considered best by bluealsa then? (and 2nd best if first best missing?).

Arkq, 2020-02-09:
The order is here: https://github.com/Arkq/bluez-alsa/blob/master/src/bluez-a2dp.c#L412 but it is more or less hack over BlueZ. It seems that BlueZ uses codecs in the order of registration - firstly registered, firstly tried. I have not found any proper way of defining codec priority in BlueZ....

macmpi, 2020-02-10:
Got it, thanks.

closed 2020-02-10 by macmpi