arkq/bluez-alsa #295: No audio from 1 device

gb160, 2020-02-12:
Hi, I have 3 Bluetooth devices, bluealsa works perfectly on 2 of those but on the third I get no audio.

I have:
Fiio BTR-3 BT Amp works fine.
Earfun Wave BT Headphones work fine.
Earfun Free BT ear buds, I get no audio.

Im running bluealsa on a Vero4k, and built it with AAC, Aptx, AptxHD, and LDAC codecs.
Th Fiio BTR-3 uses LDAC, both the earful devices use AAC.

What do u need in the way of logs/information to help diagnose this issue?
Thanks.

graham8, 2020-02-14:
Maybe the issue is this which you shared with me:

```
osmc@osmc:~$ gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
({objectpath '/org/bluealsa/hci0/dev_AC_19_06_01_18_90/a2dpsrc/sink': {'Device': <objectpath '/org/bluez/hci0/dev_AC_19_06_01_18_90'>, 'Transport': <'A2DP-source'>, 'Mode': <'sink'>, 'Format': <uint16 32784>, 'Channels': <byte 0x01>, 'Sampling': <uint32 44100>, 'Codec': <uint16 2>, 'Delay': <uint16 150>, 'Volume': <uint16 32639>}},)

```
The object path looks weird, @Arkq

gb160, 2020-02-15:
Tested the Headphones that won't work on a Pi with bluealsa from the Buster repo
`bluealsa |       0.13 | http://archive.raspberrypi.org/debian buster/main armhf Packages`
... and they work fine, built and installed this one and no output whatsoever.

Arkq, 2020-02-17:
> built and installed this version and no output whatsoever.

Do you mean fresh master branch?

Could you please post here logs from bluealsa server (compile with `--enable-debug`) when connecting to 3rd device and trying to play sound? Alternatively, you can also record BT HCI traffic with `tcpdump -i bluetooth0 -w /tmp/xxx.dump` and attach here.

gb160, 2020-02-18:
> Do you mean fresh master branch?
>
> Could you please post here logs from bluealsa server (compile with `--enable-debug`) when connecting to 3rd device and trying to play sound? Alternatively, you can [...]

Hi mate, to confirm yes,this is the fresh master branch.
Here is the log from bluealsa, pairing, connecting, then attempting to play a couple of different .wav files:

`bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/LDAC/Source/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/LDAC/Source/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/LDAC/Source/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/LDAC/Source/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/APTXHD/Source/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/APTXHD/Source/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/APTXHD/Source/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/APTXHD/Source/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/APTX/Source/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/APTX/Source/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/APTX/Source/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/APTX/Source/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Sink/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Sink/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Sink/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Sink/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/main.c:378: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:383: Starting main dispatching loop
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: D: ../../src/bluez.c:640: A2DP Source (AAC) configured for device 00:00:00:DD:4B:04
bluealsa: D: ../../src/bluez.c:642: Configuration: channels: 2, sampling: 48000
bluealsa: D: ../../src/ba-transport.c:885: State transition: 0 -> 0
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/MPEG24/Source/3
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/3
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: Delay
bluealsa: D: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()
bluealsa: D: ../../src/bluealsa-dbus.c:373: Called: org.bluealsa.PCM1.Open()
bluealsa: D: ../../src/ba-transport.c:1014: New transport: 14 (MTU: R:672 W:672)
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:885: State transition: 0 -> 2
bluealsa: D: ../../src/ba-transport.c:1199: Created new thread [ba-a2dp-aac]: A2DP Source (AAC)
bluealsa: D: ../../src/ba-transport.c:885: State transition: 2 -> 2
bluealsa: D: ../../src/a2dp.c:1375: Starting IO loop: A2DP Source (AAC)
[... cut ...]
bluealsa: D: ../../src/ba-transport.c:1172: Closing PCM: 10
bluealsa: D: ../../src/ba-transport.c:1041: Releasing transport: A2DP Source (AAC)
bluealsa: D: ../../src/ba-transport.c:1070: Closing BT: 16
bluealsa: D: ../../src/ba-transport.c:1221: Exiting IO thread: A2DP Source (AAC)
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:885: State transition: 2 -> 0

`

gb160, 2020-02-18:
Not sure if useful but here is the output from `gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs` with one of my working devices:
({objectpath '/org/bluealsa/hci0/dev_40_ED_98_19_9A_8E/a2dpsrc/sink': {'Device': <objectpath '/org/bluez/hci0/dev_40_ED_98_19_9A_8E'>, 'Transport': <'A2DP-source'>, 'Mode': <'sink'>, 'Format': <uint16 32784>, 'Channels': <byte 0x02>, 'Sampling': <uint32 96000>, 'Codec': <uint16 11775>, 'Delay': <uint16 150>, 'Volume': <uint16 32639>}},)

and this is the output from my non working device:
`({objectpath '/org/bluealsa/hci0/dev_00_00_00_DD_4B_04/a2dpsrc/sink': {'Device': <objectpath '/org/bluez/hci0/dev_00_00_00_DD_4B_04'>, 'Transport': <'A2DP-source'>, 'Mode': <'sink'>, 'Format': <uint16 32784>, 'Channels': <byte 0x02>, 'Sampling': <uint32 48000>, 'Codec': <uint16 2>, 'Delay': <uint16 1800>, 'Volume': <uint16 32639>}},)`

The non working device has a large delay.

Arkq, 2020-02-18:
Compile bluezalsa without AAC support and check. Also (before recompiling) use options --a2dp-force-audio-cd and --aac-vbr-mode=1

gb160, 2020-02-18:
> Compile bluezalsa without AAC support and check. Also (before recompiling) use options --a2dp-force-audio-cd and --aac-vbr-mode=1

using those options (before recompiling) made no difference, still no audio....after compiling without AAC support the earbuds worked fine (using SBC codec).

bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/main.c:378: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:383: Starting main dispatching loop
bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: D: ../../src/bluez.c:1264: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:719: Called: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: D: ../../src/bluez.c:640: A2DP Source (SBC) configured for device 00:00:00:DD:4B:04
bluealsa: D: ../../src/bluez.c:642: Configuration: channels: 2, sampling: 48000
bluealsa: D: ../../src/ba-transport.c:885: State transition: 0 -> 0
bluealsa: D: ../../src/bluez.c:764: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/3
bluealsa: D: ../../src/bluez.c:793: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/3
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: Delay
bluealsa: D: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()
bluealsa: D: ../../src/bluealsa-dbus.c:373: Called: org.bluealsa.PCM1.Open()
bluealsa: D: ../../src/ba-transport.c:1014: New transport: 14 (MTU: R:672 W:672)
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:885: State transition: 0 -> 2
bluealsa: D: ../../src/ba-transport.c:1199: Created new thread [ba-a2dp-sbc]: A2DP Source (SBC)
bluealsa: D: ../../src/a2dp.c:574: Starting IO loop: A2DP Source (SBC)
bluealsa: D: ../../src/ba-transport.c:885: State transition: 2 -> 2
bluealsa: D: ../../src/ba-transport.c:969: PCM drained
bluealsa: D: ../../src/a2dp.c:121: PCM read buffer flushed: 0
bluealsa: D: ../../src/a2dp.c:121: PCM read buffer flushed: 0
bluealsa: D: ../../src/a2dp.c:106: PCM has been closed: 10
bluealsa: D: ../../src/ba-transport.c:1172: Closing PCM: 10
bluealsa: D: ../../src/a2dp.c:276: Keep-alive polling: 0
bluealsa: D: ../../src/a2dp.c:276: Keep-alive polling: 0
bluealsa: D: ../../src/ba-transport.c:1041: Releasing transport: A2DP Source (SBC)
bluealsa: D: ../../src/ba-transport.c:1070: Closing BT: 14
bluealsa: D: ../../src/ba-transport.c:1221: Exiting IO thread: A2DP Source (SBC)
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:885: State transition: 2 -> 0

gb160, 2020-02-20:
@Arkq Hi mate, any ideas with this?

Arkq, 2020-02-20:
Could you confirm with e.g. Android that aac works? Please use bt hci log from developer tools. You will need Android with AAC support, though.

gb160, 2020-02-20:
> Could you confirm with e.g. Android that aac works? Please use bt hci log from developer tools. You will need Android with AAC support, though.

I dont have any android devices apart from a firestick im afraid.
I can confirm that playing from my MacBook Pro to these headphones works with AAC, not sure if thats helpful however.
[image]

Arkq, 2020-02-20:
It is a little bit helpful, but in order to detect issue I would need bluetooth hci logs from working setup... But I'm not sure whether it's available on Mac OS.

gb160, 2020-02-20:
> It is a little bit helpful, but in order to detect issue I would need bluetooth hci logs from working setup... But I'm not sure whether it's available on Mac OS.

I think ive found a way to capture the chi logs in Mac OS using the packet logger app from the Xcode development tools, ive captured connecting the headphones and playing back some music, I can choose saving the log as .pklg or .btsnoop file format...can u view either of those?

Arkq, 2020-02-20:
Whichever is supported by wireshark. But I know that btsnoop is supported, so please use this.

gb160, 2020-02-20:
> Whichever is supported by wireshark. But I know that btsnoop is supported, so please use this.

here u go buddy:
http://s000.tinyupload.com/index.php?file_id=63458408172811469386
I hope it's of use.

Arkq, 2020-02-20:
Format is OK, but you will need to start logging **before** bt connection with earbuts.

gb160, 2020-02-20:
> Format is OK, but you will need to start logging **before** bt connection with earbuts.

Here mate:
http://s000.tinyupload.com/index.php?file_id=91424535311955893581

You dont need the pairing process as well do you?

gb160, 2020-02-20:
Apologies, I think some of the traffic might have been filtered, this app is new to me ;)

Try this one:
http://s000.tinyupload.com/index.php?file_id=08488821378912398639

Arkq, 2020-02-20:
Ok, now it would be nice to have logs from board with bluealsa (where it fails to play) produced with tcpdump.

gb160, 2020-02-20:
> Ok, now it would be nice to have logs from board with bluealsa (where it fails to play) produced with tcpdump.

I used `tcpdump -i bluetooth0 -w ~/xxx.dump` 
https://gofile.io/?c=mUlAsI

gb160, 2020-02-21:
@Arkq I presume this isn't an easy one to fix?

Arkq, 2020-02-21:
I don't know, I hope I will have some time to dig deeper during this weekend. I think that you 've send all required logs/dumps for investigation.

gb160, 2020-02-21:
> I don't know, I hope I will have some time to dig deeper during this weekend. I think that you 've send all required logs/dumps for investigation.

Ok buddy, thanks for looking into this.
Fingers crossed.

gb160, 2020-03-01:
@Arkq Any ideas on this mate?

Arkq, 2020-03-01:
Sorry, but I've got not time for this issue due to tight schedule - I need to refactor bluez-alsa code a little bit. I'll look into this issue when I'll make bluez-alsa codebase ready for FastStream support.

Arkq, 2020-03-14:
I've check the problem with your headset. It seems that it does not suport MPEG-4 for AAC codec. It only supports MPEG-2 (it is mandatory format, though, but bluealsa does not support it....). I thought that one day this issue will pop up, but unfortunately, it's not easy to fix. Bluealsa uses fdk-aac library for AAC encoding/decoding, and this library does not support MPEG-2 format (or I don't know how to configure it). So as for now, this headset will not work with bluealsa.

I will keep this issue open, though. So, maybe one day I will figure out how to support MPEG-2 AAC.

Arkq, 2020-03-14:
Hmmm, maybe I was wrong, it seems that the MPEG-2 support has been restored in fdk-aac-2.0 (it was removed after 0.1.4). @gb160 could you check if removing [this ifdef](https://github.com/Arkq/bluez-alsa/blob/master/src/a2dp.c#L1288) will solve this issue (if compiling with [fdk-aac>=2.0](https://github.com/mstorsjo/fdk-aac))?

gb160, 2020-03-14:
> Hmmm, maybe I was wrong, it seems that the MPEG-2 support has been restored in fdk-aac-2.0 (it was removed after 0.1.4). @gb160 could you check if removing [this [...]

Just tried it buddy and still no output, compiled with fdk-aac 2.0.1

Arkq, 2020-03-14:
@gb160 please try this branch: https://github.com/Arkq/bluez-alsa/tree/mpeg2aac
Also, if you are familiar with IRC, you might go to Freenode #bluealsa, it might be easier to talk over this issue :) e.g. using this client: https://webchat.freenode.net/

closed 2020-03-15 by Arkq

gb160, 2020-03-15:
@Arkq Just built and installed the latest master branch on Vero4k+, and its working perfectly both from aplay, and when routing Kodi audio.
Thanks Buddy!

commit 702e9c42 (2020-03-15): Add MPEG-2 and bump MUX version (fdk-aac >= 2.0)

After fdk-aac version 0.1.4, the audio object type MPEG-2 AAC has been
removed. Unfortunately, this object type is defined as mandatory in the
A2DP specification. Fortunately, it has been restored in the fdk-aac
version >= 2.0, so we can once again fulfill A2DP specification.

Additionally, fdk-aac >= 0.1.5 supports newer LATM syntax version -
ISO-IEC 14496-3 (2005). Since, it might be incompatible with older BT
devices, BlueALSA will allow to downgrade it via command line option.

Fixes #295