arkq/bluez-alsa #296: AAC decode frame error: Transport sync error

graham8, 2020-02-14:
Playing a simple WAV from one device running bluealsa to another running bluealsa-aplay I get the subject error at the receiving end about every 100ms and no sound.  At the sending device everything appears OK:
```
osmc@vero4kp2:~$ aplay -D plug:bluealsa:CC:4B:73:77:7E:0B -v /usr/share/sounds/alsa/Front_Center.wav
2029.287098884: D: ../../../src/asound/bluealsa-pcm.c:624: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Setting constraints
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
2029.289715050: D: ../../../src/asound/bluealsa-pcm.c:344: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing HW
2030.975855843: D: ../../../src/asound/bluealsa-pcm.c:370: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: FIFO buffer size: 4096
2030.975910134: D: ../../../src/asound/bluealsa-pcm.c:376: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Selected HW buffer: 12 periods x 16380 bytes <= 196568 bytes
2030.975939259: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
2030.976013926: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
2030.976325426: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
2030.976360176: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
2030.976384176: D: ../../../src/asound/bluealsa-pcm.c:413: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Prepared
2030.976406218: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
Plug PCM: Rate conversion PCM (96000, sformat=S16_LE)
Converter: libspeex (external)
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24570
  period_size  : 2047
  period_time  : 42666
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 2047
  period_event : 0
  start_threshold  : 24570
  stop_threshold   : 24570
  silence_threshold: 0
  silence_size : 0
  boundary     : 805109760
Slave: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0
    1 <- 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 96000
  exact rate   : 96000 (96000/1)
  msbits       : 16
  buffer_size  : 49142
  period_size  : 4095
  period_time  : 42666
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 4095
  period_event : 0
  start_threshold  : 49140
  stop_threshold   : 49142
  silence_threshold: 0
  silence_size : 0
  boundary     : 1610285056
Slave: BlueALSA PCM: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink
BlueALSA BlueZ device: /org/bluez/hci1/dev_CC_4B_73_77_7E_0B
BlueALSA Bluetooth codec: 0x2
2031.004184301: D: ../../../src/asound/bluealsa-pcm.c:264: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Starting
2031.004859384: D: ../../../src/asound/bluealsa-pcm.c:137: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Starting IO loop: 8
2031.945741760: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
2031.945864760: D: ../../../src/asound/bluealsa-pcm.c:391: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Initializing SW
2032.412612802: D: ../../../src/asound/bluealsa-pcm.c:169: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: IO thread paused: 4
2032.713195677: D: ../../../src/asound/bluealsa-pcm.c:305: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Stopping
2032.713579677: D: ../../../src/asound/bluealsa-pcm.c:105: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: IO thread cleanup
2032.714238469: D: ../../../src/asound/bluealsa-pcm.c:305: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Stopping
2032.714676969: D: ../../../src/asound/bluealsa-pcm.c:383: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Freeing HW
2032.714786927: D: ../../../src/asound/bluealsa-pcm.c:334: /org/bluealsa/hci1/dev_CC_4B_73_77_7E_0B/a2dpsrc/sink: Closing
```
I'll upload some tcpdump logs if that will help explain what's going on - does it matter which device I get them from?

Arkq, 2020-02-17:
Have you checked bluealsa logs on the receiver side? I see that you have bluez-alsa compiled with debug enabled, so some info might be there. Preferably post logs here.

graham8, 2020-02-18:
Thanks for the feedback - here you go. I issued aplay -Dplug:bluealsa:<MAC> /usr/share/sounds/alsa/Front* so there are three files attempting to play.  I cut out most of the AAC error messages
```
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.740173016: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.740268849: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.740300224: D: ../../src/ba-transport.c:885: State transition: 0 -> 1
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.744620308: D: ../../src/ba-transport.c:1014: New transport: 8 (MTU: R:672 W:672)
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.744731599: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.744813391: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.744843849: D: ../../src/ba-transport.c:885: State transition: 1 -> 2
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.745051849: D: ../../src/ba-transport.c:1199: Created new thread [ba-a2dp-aac]: A2DP Sink (AAC)
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.745535558: D: ../../src/a2dp.c:1139: Starting IO loop: A2DP Sink (AAC)
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.797636891: E: AAC decode frame error: Transport sync error
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.800001724: E: AAC decode frame error: Transport sync error
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.802721558: E: AAC decode frame error: Transport sync error
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.836435183: E: AAC decode frame error: Transport sync error
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.838885266: E: AAC decode frame error: Transport sync error
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.845135558: E: AAC decode frame error: Transport sync error
Feb 18 11:58:05 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4753.878878099: E: AAC decode frame error: Transport sync error
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.142628475: E: AAC decode frame error: Transport sync error
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.146353392: E: AAC decode frame error: Transport sync error
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.180079392: E: AAC decode frame error: Transport sync error
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.182534183: E: AAC decode frame error: Transport sync error
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.185015433: E: AAC decode frame error: Transport sync error
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.497420350: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.497531392: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.497562767: D: ../../src/ba-transport.c:885: State transition: 2 -> 0
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.497980517: D: ../../src/ba-transport.c:1041: Releasing transport: A2DP Sink (AAC)
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.498015017: D: ../../src/ba-transport.c:1070: Closing BT: 8
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.498046600: D: ../../src/ba-transport.c:1221: Exiting IO thread: A2DP Sink (AAC)
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.505315767: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.505410017: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.505441059: D: ../../src/ba-transport.c:885: State transition: 0 -> 1
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.508503934: D: ../../src/ba-transport.c:1014: New transport: 8 (MTU: R:672 W:672)
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.508644684: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.508722809: D: ../../src/bluez.c:1380: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.508757600: D: ../../src/ba-transport.c:885: State transition: 1 -> 2
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.508982934: D: ../../src/ba-transport.c:1199: Created new thread [ba-a2dp-aac]: A2DP Sink (AAC)
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.509195309: D: ../../src/a2dp.c:1139: Starting IO loop: A2DP Sink (AAC)
Feb 18 11:58:07 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4755.562552892: E: AAC decode frame error: Transport sync error
[... cut ...]
Feb 18 11:58:11 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4759.173655644: D: ../../src/ba-transport.c:885: State transition: 2 -> 0
Feb 18 11:58:11 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4759.173878519: D: ../../src/ba-transport.c:1041: Releasing transport: A2DP Sink (AAC)
Feb 18 11:58:11 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4759.173911060: D: ../../src/ba-transport.c:1070: Closing BT: 8
Feb 18 11:58:11 vero4kp1 bluealsa[3959]: /usr/bin/bluealsa: 4759.173944685: D: ../../src/ba-transport.c:1221: Exiting IO thread: A2DP Sink (AAC)
```

graham8, 2020-02-19:
@Arqk Any thoughts on this?  I don't really know how to interpret these logs.
I thought of limiting the codec to SBC - am I right there's no way to do that at runtime?
Still it would be nice to know AAC worked between devices that support it.

Arkq, 2020-02-19:
Unfortunately, I don't know why it's not working. Also, runtime codec selection for A2DP is not supported. Could you confirm with e.g. Android, that aac works with this headset (you will need to dump bt hci logs on Android)?

graham8, 2020-02-19:
I can send audio to this sink from android and it works but no idea whether it's AAC and no idea how to grab logs from android.
The other thing I'd like to try is limiting the sample rate to 48k (it's defaulting to 96k).  Is there a way to do that in the bluealsa code or at runtime?
Just to be clear, the sink is not a headset but a media player box running bluealsa-aplay.

Arkq, 2020-02-19:
Sorry... I've responded to your email but I thought about different issue.... the Android logs part is invalid for this one :)

So now the relevant comment. 

Selected coded should be visible in the sink logs (if that's bluealsa as well). If you've got iPhone it will most likely select AAC, Android might select it too, but it has to be one of top models (AAC codec requires license). 

You can lower bitrate with --a2dp-force-audio-cd. I will try to check this on my side, but within a month timeslot.

graham8, 2020-02-19:
I did try force-audio-cd and it had no effect.  I'll check again.

graham8, 2020-02-19:
OK, by setting both source and sink to cd quality, I get AAC at 44.1k :)

closed 2020-02-19 by graham8

Asikbct, 2024-08-28:
Hi @graham8 , @arkq 

I have the same problem with AAC, 96000Hz sample rate. But, audio plays very slowly in the external BT speaker when our device is a2dp source. I could restrict this by applying --a2dp-force-audio-cd. but I also need to have 48000 Hz for our requirement. What do you think could be the solution here?

arkq, 2024-08-28:
@Asikbct you will have to add a patch to the code if you want to remove suport for rates > 48000. With latest release you can do that by adding "if" to this loop: https://github.com/arkq/bluez-alsa/blob/master/src/asound/bluealsa-pcm.c#L1346-L1347 
```c
		if (codec->sampling[i] <= 48000)
			list[n++] = codec->sampling[i];
```

With older release (and the latest as well) you can simply remove support for rates > 48k in the BlueALSA system service by removing samplings that you do not want: https://github.com/arkq/bluez-alsa/blob/master/src/a2dp-aac.c#L750