arkq/bluez-alsa #298: AT command for Mic Mute

hrc-amilcar, 2020-02-18:
What would it take to add support for XHSMICMUTE command?

I have a headset that sends the following when swiveling the mic up and down:
```
./bluealsa: W: Unsupported AT message: SET: command:+ANDROID, value:XHSMICMUTE,1
./bluealsa: W: Unsupported AT message: SET: command:+ANDROID, value:XHSMICMUTE,0
```
Muting works because the device not sending any data, but since BlueAlsa is not interpreting that message, the Alsa mixer does not know that it is actually muted.

Volume Up/Down works just fine.

There are some unsupported battery messages as well:
```
./bluealsa: W: Unsupported AT message: SET: command:+ANDROID, value:XHSTBATSOC,37,T0948,14042019
./bluealsa: W: Unsupported AT message: SET: command:+ANDROID, value:XHSTBATSOH,100,T0948,14042019
```

Arkq, 2020-02-19:
It would be nice to have some spec for these commands. What model of headset it is?

Arkq, 2020-02-29:
I wrote to Zebra technical support about these commands, but I've got no response from them.... I'll way a little bit longer, but in case of no response I'll add support for mic muting only. For XHSTBATSOH and HSTBATSOC it would be required to gather more samples in order to reverse engineer it.

Maybe there is some dedicated Android application for this headset? It would be much easier to decompile apk and check for maybe some other commands and see how they should be interpreted :)

xbr786, 2020-03-06:
Hello  hrc-amilcar  and Arkq

I am Technical Support L3 from Zebra and  Arkq  has opened a new case with use this time and i am trying to help you,  after speaking to engineering have following back:

HS3100 AT commands XHSMICMUTE, HSTBATSOC are private information and proprietary for Zebra.
We would be able to help but Engineering would require an NDA to be in place , that is their current standpoint.

The request is to get more context around why this is needed and why these AT commands are needed to be integrated on this Bluez Alsa  Project

If   hrc-amilcar  can expose your  mail address to Arkq  we can discuss this as part of the current Support case

Arkq, 2020-03-07:
> We would be able to help but Engineering would require an NDA to be in place , that is their current standpoint.

For me, signing NDA is not an issue. However, if such NDA will disallow the integration to be publicly available under the MIT license (the goal is to make opensource bluez-alsa project aware of these proprietary AT commands), I will go with "reverse engineer". Anyway, the usage of "XHSMICMUTE" is kinda obvious.

> The request is to get more context around why this is needed and why these AT commands are needed to be integrated on this Bluez Alsa Project

@xbr786, the context is as follows. This project (bluez-alsa) allows to access BT audio device via ALSA interface. In addition to audio streaming (HFP, A2DP) is allows to control volume. HFP/HSP profile specifies AT commands for controlling speaker/mic volume. It seems that your products (I guess that these commands are used in Zebra HS series) have extra command for controlling/signalling mute state. Adding these commands will enhance user experience (see the very first comment in this thread). Additionally, bluez-alsa exposes (if available) battery level of the device - also better user experience. E.g. Apple products have "XAPL" extension for such signalling. Then, it would be great to add support for Zebra HS battery notification if such information is available.

If you've got some other AT extension related to audio itself (I'm not interested in "shortcuts" for e.g. launching Siri - Sony has such extension), I might be interested in them as well. But that might be discussed under NDA of course.

As for help in engineering, all I need is a simple description, e.g.:

- AT+XHSMICMUTE=1 - sent by headset, possible values: 1 (muted), 0 (unmuted)
- AT+XHSMICMUTE=1 - sent by audio-gateway - controls mute state of headset

something like that :)

xbr786, 2020-03-13:
hello Arkq,  i tried to get this answered internally but out engineering stays put  and  here is the reply;

"we shared the information from your last update with Product Management. 
The HS3100 is positioned as part of the Zebra VDP solution so proprietary operation is not publicly shared. If the end customer would like to access these features they will need to work with a Zebra partner or account team.

Since Blue Alsa opened this request, our response is that we cannot share this information with them without an NDA. Blue Alsa may also work with a Zebra partner or account team to work out an NDA agreement for us to share this information."

That means one way or the other an NDA is required and as a result you will not be able to add this on the bleuz-alsa opensource project.

i stressed this fact 2 times that this request compasses an 'opensource' project  but in vain.

hope this helps 
btw HS3100 is not supporting AD2P  it is BT Class II supporting 
Bluetooth Hands-Free Profile (HFP 1.6)
Headset Profile (HSP 1.2)

Regarding battery can tell you that the HS3100 has an hotswappeable battery and the battery level can be queried for that usuage.

Arkq, 2020-03-13:
@xbr786 many thanks for the explanation of the current standpoint of Zebra Product Management. I totally understand, that from the company point of view, it might seems, that open source contribution is os no value... Also, may thanks for providing some insight into the HS3100 battery features :)

@hrc-amilcar as you can see, we will get no support from Zebra with this task :) Anyway, I think that I can try to "investigate" a little bit available possibilities with this headset. Please, gather some logs from bluealsa daemon. The scenario might be as follows: connect headset, play something, press every available button, long press every possible button, double press every possible button, play something until battery is very low, connect charger, charge to full battery level, if battery is indeed removeable remove it and insert again. You might also try to press every possible button while playing and while not playing audio. After such test, it might be also a good idea to try to send some AT command to the headset, but that might be done later - maybe it is possible to mute mic from the audio-gateway (bluealsa).

hrc-amilcar, 2020-03-13:
@Arkq Many thanks for getting in touch with Zebra! I'll try to get those readouts as soon as I can. Would the interpretation of XHSMICMUTE be trivial to start with, since it is just a 0 or 1? Are there not standard AT commands for Mute? If there are, I wonder why every manufacturer doesn't just adhere to those.

Arkq, 2020-03-14:
> Would the interpretation of XHSMICMUTE be trivial

Yes, it seems to be trivial, but I'd like to see whether there are some other clues about these extra commands. Hence, the "button-pressing" test :)

> Are there not standard AT commands for Mute?

No, there is no standard AT command for mute. HSP/HFP only specifies AT command for gain control, so when you set gain to 0 it means that the headset is "muted", but that's not entirely true. Most sound cards does not mute output when volume level is set to 0. If you listen very carefully, you can hear faint sound :) And I believe that the same is with BT headsets (based on few headsets which I've tested). So, the really nice feature would be the possibility to mute mic/speaker not from the headset itself, but from the bluealsa. But that might be tricky to uncover without Zebra support :/

Arkq, 2021-08-11:
@hrc-amilcar do you still have this HS3100 headset? If yes, could you please gather some debug logs? Testing battery status is not required. Additionally, I'd like to check whether it will be possible to mute/unmute mic from bluealsa side with `bluealsa-rfcomm` tool. Check what will happen if you send something like this: `+ANDROID=XHSMICMUTE,0` and `+ANDROID=XHSMICMUTE,1` or `+ANDROID:XHSMICMUTE,0` and `+ANDROID:XHSMICMUTE,1`

hrc-amilcar, 2021-08-11:
@Arkq , I do still have the HS3100. I'll try to find some time to get some debug logs and sending those AT commands. It would be nice to get the features of this headset recognizing through bluealsa.

hrc-amilcar, 2021-08-16:
Hi @Arkq,

Here are some debug logs for the Zebra HS3100. Let me know if there are some others you'd like to grab.

--- Startup and Connect ---
```
./bluealsa: D: bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
./bluealsa: D: bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
./bluealsa: D: bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
./bluealsa: D: bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
./bluealsa: D: bluez.c:776: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
./bluealsa: D: bluez.c:710: Registering hands-free profile: /org/bluez/HSP/AudioGateway
./bluealsa: D: bluez.c:776: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
./bluealsa: D: bluez.c:710: Registering hands-free profile: /org/bluez/HFP/AudioGateway
./bluealsa: D: main.c:419: Acquiring D-Bus service name: org.bluealsa
./bluealsa: D: main.c:424: Starting main dispatching loop
./bluealsa: D: dbus.c:59: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
./bluealsa: D: ba-rfcomm.c:1289: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway
./bluealsa: D: ba-rfcomm.c:905: Starting RFCOMM loop: HFP Audio Gateway
./bluealsa: D: at.c:161: AT message: SET: command:+BRSF, value:191
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:+BRSF, value:2784
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: sco.c:203: Created SCO dispatcher [ba-sco-dispatch]: hci0
./bluealsa: D: bluez.c:601: HFP Audio Gateway configured for device 00:1B:41:B2:86:9A
./bluealsa: D: ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway state transition: 0 -> 2
./bluealsa: D: sco.c:77: Starting SCO dispatcher loop: hci0
./bluealsa: D: at.c:161: AT message: SET: command:+BAC, value:1,2
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway state transition: 2 -> 3
./bluealsa: D: at.c:161: AT message: TEST: command:+CIND, value:(null)
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway state transition: 3 -> 5
./bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: at.c:161: AT message: GET: command:+CIND, value:(null)
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway state transition: 5 -> 7
./bluealsa: D: at.c:161: AT message: SET: command:+CMER, value:3, 0, 0, 1
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway state transition: 7 -> 8
./bluealsa: D: ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway state transition: 8 -> 9
./bluealsa: D: at.c:161: AT message: SET: command:+VGS, value:10
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: at.c:161: AT message: GET: command:+BTRH, value:(null)
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: at.c:161: AT message: SET: command:+ANDROID, value:XHSTCAP,3.8,3.10,2819M0249,
./bluealsa: W: ba-rfcomm.c:1186: Unsupported AT message: SET: command:+ANDROID, value:XHSTCAP,3.8,3.10,2819M0249,
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
./bluealsa: D: at.c:161: AT message: SET: command:+ANDROID, value:XHSMICMUTE,1
./bluealsa: W: ba-rfcomm.c:1186: Unsupported AT message: SET: command:+ANDROID, value:XHSMICMUTE,1
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
./bluealsa: D: at.c:161: AT message: SET: command:+ANDROID, value:XHSTPOWER,0
./bluealsa: W: ba-rfcomm.c:1186: Unsupported AT message: SET: command:+ANDROID, value:XHSTPOWER,0
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
./bluealsa: D: at.c:161: AT message: SET: command:+ANDROID, value:XHSTBATSOC,96,T0948,14042019
./bluealsa: W: ba-rfcomm.c:1186: Unsupported AT message: SET: command:+ANDROID, value:XHSTBATSOC,96,T0948,14042019
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
./bluealsa: D: at.c:161: AT message: SET: command:+ANDROID, value:XHSTBATSOH,100,T0948,14042019
./bluealsa: W: ba-rfcomm.c:1186: Unsupported AT message: SET: command:+ANDROID, value:XHSTBATSOH,100,T0948,14042019
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
./bluealsa: D: at.c:161: AT message: CMD: command:+CLCC, value:(null)
./bluealsa: W: ba-rfcomm.c:1186: Unsupported AT message: CMD: command:+CLCC, value:(null)
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
./bluealsa: D: at.c:161: AT message: SET: command:+BIA, value:0,1,1,1,0,0,0
./bluealsa: D: ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: at.c:161: AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
[... cut ...]
```

--- Long pressing the Off button on the headset ---
```
./bluealsa: D: ba-rfcomm.c:1237: RFCOMM disconnected: Connection reset by peer
./bluealsa: D: ba-rfcomm.c:868: Closing RFCOMM: 9
./bluealsa: D: ba-rfcomm.c:877: RFCOMM link lost quirk: Destroying SCO transport
./bluealsa: D: ba-transport.c:875: Freeing transport: HFP Audio Gateway (mSBC)
./bluealsa: D: ba-device.c:135: Freeing device: 00:1B:41:B2:86:9A
./bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
```

Arkq, 2021-08-18:
I've created a branch with required changes: https://github.com/Arkq/bluez-alsa/tree/zebra It should support battery level reporting and mic mute reporting.

However, I've got a question regarding mute switch on this headset. Is it stateless switch or on/off switch? Because right now the code allows to update mute state from the headset, but I'm not sure whether later it should be possible to unmute it from bluealsa. If this switch is an on/off switch, I would have to implement something alongside `rfkill`, where there is a software and a hardware on/off.

hrc-amilcar, 2021-08-18:
Excellent! I'll try the branch.

The headset mutes and unmutes by swinging the microphone bar up and down. So it's definitely stateful. I can't imagine that you can unmute it in software and the Mic still work when it's in the Up position.

Arkq, 2021-08-18:
> I can't imagine that you can unmute it in software and the Mic still work when it's in the Up position.

I've updated the zebra branch. So now, there is a software and hardware mute state. The status update from the headset changes the latter. In case of hardware mute it will not be possible to change software mute state. E.g., if you mute with alsamixer and then swing mic bar up it will be not possible to unmute in alsamixer. When you will swing mic bar down, mic will be still muted, but software unmute will be possible now. However, I'm not sure whether such behavior is OK.... That's my invention how to integrate software and hardware mute into one control element in alsamixer. Otherwise, it will be required to add another alsamixer control element which will show hardware mute state.

commit 6e726279 (2021-08-18): Support proprietary AT commands for Zebra HS3100

Fixes #298
commit 407b54fd (2021-08-18): Support proprietary AT commands for Zebra HS3100

Fixes #298
commit 8aed3a77 (2021-08-18): Support proprietary AT commands for Zebra HS3100

Fixes #298
closed 2021-09-16 by Arkq