arkq/bluez-alsa #300: Couldn't get BlueALSA PCM: PCM not found

cerberek, 2020-02-29:
I'm trying to run bluaslsa on my Raspbian Buster. 
Fristly I installed one from debian repository. I got error running aplay that PCM not found.

So i build your bluez-alsa and I have some issue with addition that I see something in logs.

In bluaalsa output I can see that:
/usr/bin/bluealsa: D: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()

My ~/.asoundrc:
```
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "00:12:6F:D2:4B:3B"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000

pcm.minirig {
        type plug
        slave {
            pcm {
                type bluealsa
                device 00:12:6F:D2:4B:3B
                profile "a2dp"
            }
        }
        hint {
            show on
            description "My speaker"
        }
}
```

borine, 2020-02-29:
what is the ouptut of:

`echo "info 00:12:6F:D2:4B:3B" | bluetoothctl | grep Connected`

cerberek, 2020-03-01:
Hi Borine,
I have no clue how this happend but now when I connectd bluetooh device. I see in bluealsaloogs that it register as a2dp sink and all works :D 
I'm not aware of any changes i did. So all good. But nio idea where the problem was

closed 2020-03-01 by cerberek