arkq/bluez-alsa #301: Bluealsa as proxy

cerberek, 2020-03-01:
Hi, I was curious If and how I can setup that bluealsa would work as proxy. 

That I have speaker as sink all the time connected through bluetooth and If I want to play from it from my phone I connect to bluealsa device and it will forward audio to connected speaker.

both devices are paired with bluealsa I just don't know how to route it.

Thanks for advices.

Arkq, 2020-03-01:
You might try something like this: `bluealsa-aplay -d <bluealsa-speaker-pcm> <phone-bt-mac>`.  See `bluealsa-aplay --help` for more info.

cerberek, 2020-03-01:
Hi, thanks for this advise. Looks like good direction and I was able to route sound. But sometime bluealsa-aplay ends up with segfault or SIGABRT. And it can be laggy. Is there way to optimalize it?

Some error outputs I got
```
bluealsa-aplay: D: ../../utils/aplay.c:525: Creating PCM worker 38:30:F9:FC:D3:E5
bluealsa-aplay: D: ../../utils/aplay.c:801: Starting main loop
bluealsa-aplay: D: ../../utils/aplay.c:356: Starting PCM loop
D: ../../../src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Setting constraints
D: ../../../src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: FIFO buffer size: 4096
D: ../../../src/asound/bluealsa-pcm.c:400: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Starting
D: ../../../src/asound/bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: Starting IO loop: 10
ALSA lib ../../../src/asound/bluealsa-pcm.c:244:(io_thread) PCM FIFO write error: Bad addressbluealsa-aplay: malloc.c:2852: mremap_chunk: Assertion `((size + offset) & (GLRO (dl_pagesize) - 1)) == 0' failed.
```
or 
```ALSA lib ../../../src/asound/bluealsa-pcm.c:244:(io_thread) PCM FIFO write error: Bad address
D: ../../../src/asound/bluealsa-pcm.c:111: /org/bluealsa/hci0/dev_00_12_6F_D2_4B_3B/a2dpsrc/sink: IO thread cleanup
Neoprávněný přístup do paměti (SIGSEGV)
```

borine, 2021-06-06:
There have been many improvements to bluealsa pcm plugin and to bluealsa-aplay since this issue was raised. If possible, can you check whether this issue still exists in the latest code? I am unable to reproduce this, so will close this unless it is confirmed to still be a problem.

borine, 2021-06-28:
Closing as I am unable to reproduce this, and suspect the underlying problem (whatever it was) has now been fixed.

closed 2021-06-28 by borine