arkq/bluez-alsa #303: Rasbian Buster HFP-HF

StuartIanNaylor, 2020-03-08:
Struggling slightly prob not an issue prob me.
I have a cheap and cheerful bluetooth speaker/mic
That on default will connect a2dp but trying to force it to use hfp-hf so it is speaker/mic not just speaker

`aplay -D bluealsa:DEV=9B:FD:D5:6E:01:CA,PROFILE=a2dp 1MG.wav`

works fine

`aplay -D bluealsa:DEV=9B:FD:D5:6E:01:CA,PROFILE=sco 1MG.wav`

gives `aplay: main:828: audio open error: No such device`

Seems to always connect as a2dp and will not swap?
`Mar  8 16:50:34 raspberrypi bluetoothd[352]: Endpoint registered: sender=:1.6 path=/org/bluez/hci0/A2DP/SBC/Source/3
Mar  8 16:50:34 raspberrypi kernel: [  428.376318] input: 9B:FD:D5:6E:01:CA as /devices/virtual/input/input4
`

I did try forcing the daemon with just hfp-hf but bluetooth then just tells me the a2dp profile isn't avail.
Currently the service is starting with `ExecStart=/usr/bin/bluealsa -S -p a2dp-source -p a2dp-sink -p hfp-hf` but without profiles, with just hfp-hf

I really don't want to use ofono as installing a desktop framework like qt for wireless audio doesn't compute for me.
Its prob me and not an issue, also I am not using the Pi broadcom module as apparently it has sco routing problems and co-existence with wifi, just a Cambridge Radio usb that on windows by default connects hfp and speaker/mic work.

Anyone got a tutorial or info for a dummy like me as can find much for a2dp which works without problem but hfp-hf is what I need.

Thanks and thanks Arkq as the pulseaudio/ofono only way is such huge bloat.

StuartIanNaylor, 2020-03-09:
Yeah guessing raspbian for some reason has compiled with --enable-ofono so disabling internal HSF/HFP

On arch after compile with mSBC it tries to play but seems to freeze

```
[alarm@alarmpi ~]$ aplay -v -D bluealsa:DEV=9B:FD:D5:6E:01:CA,PROFILE=sco 1MG.wav
D: bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Setting constraints
Playing WAVE '1MG.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
D: bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing HW
D: bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: FIFO buffer size: 4096
D: bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Prepared
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
Plug PCM: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0*0.5 + 1*0.5
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 4000
  period_size  : 1000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 1000
  period_event : 0
  start_threshold  : 4000
  stop_threshold   : 4000
  silence_threshold: 0
  silence_size : 0
  boundary     : 2097152000
Slave: BlueALSA PCM: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink
BlueALSA BlueZ device: /org/bluez/hci0/dev_9B_FD_D5_6E_01_CA
BlueALSA Bluetooth codec: 0x1
D: bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Starting
D: bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Starting IO loop: 8
^CAborted by signal Interrupt...
```

Arkq, 2020-03-10:
The `--enable-ofono` does not "disable" build-in hfp support, it only disable hfp profile during runtime, if ofono service is running. Terminate ofono and you will be able to use build-in HFP.

borine, 2020-03-10:
@Arkq the ofono confusion is probably my fault; the text I wrote on the wiki installation page is misleading. I will update the page later.

@StuartIanNaylor as it happens, the raspbian-ui bluealsa package does not have ofono support enabled - as you can see from https://github.com/raspberrypi-ui/bluealsa/blob/master/debian/rules it uses a completely default configure (i.e. it does not add any options other than the standard Debian ones that are used for all packages). However, the sources it uses are from some time ago now and are missing all the significant improvements to HFP/HSP that @Arkq has made this year.

StuartIanNaylor, 2020-03-10:
Brilliant then an update might be in the offering, I couldn't find any build info even in the .deb.

Also when it comes to blue-asla and raspbian it prob needs to be 2 packages of the same source bluealsa & bluealsa-ofono that conflict but are just comple options.

Its a binary choice really and your going to use one or the other and that would solve that.
Thanks @borine

pali, 2020-03-10:
See my hsphfpd design with prototype implementation which implements that "AT server" and allows easily for dbus audio applications to create HSP or HFP connection:
https://lore.kernel.org/linux-bluetooth/20191201185740.uot7zb2s53p5gu7z@pali/

If you are interested in it, please let me know. It should simplify HSP and HFP implementation in any audio application (including bluez-alsa).

StuartIanNaylor, 2020-03-10:
Anyway just an update with my own tests and swapping back to raspberry pi4 with built in broadcom BT.

` ../configure --enable-debug --enable-aac --enable-mp3lame --enable-mpg123 --enable-ofono --enable-upower`

`aplay -D bluealsa:DEV=9B:FD:D5:6E:01:CA,PROFILE=a2dp file_example_WAV_1MG.wav`
works fine as normal

`aplay -D bluealsa:DEV=9B:FD:D5:6E:01:CA,PROFILE=sco file_example_WAV_1MG.wav`
same as it was on arch but guess its the broadcom BT again?
`pi@raspberrypi:~ $ aplay -D bluealsa:DEV=9B:FD:D5:6E:01:CA,PROFILE=sco file_example_WAV_1MG.wav
D: ../../../src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Setting constraints
Playing WAVE 'file_example_WAV_1MG.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
D: ../../../src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: FIFO buffer size: 4096
D: ../../../src/asound/bluealsa-pcm.c:400: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Starting
D: ../../../src/asound/bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Starting IO loop: 8
^CAborted by signal Interrupt...
aplay: pcm_write:2053: write error: Interrupted system call
D: ../../../src/asound/bluealsa-pcm.c:327: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:111: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:407: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:357: /org/bluealsa/hci0/dev_9B_FD_D5_6E_01_CA/hfpag/sink: Closing
`

syslog
```
Mar 10 20:50:32 raspberrypi /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()
Mar 10 20:50:32 raspberrypi bluealsa[330]: /usr/bin/bluealsa: D: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()
Mar 10 20:50:32 raspberrypi /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:373: Called: org.bluealsa.PCM1.Open()
Mar 10 20:50:32 raspberrypi /usr/bin/bluealsa: Couldn't establish SCO link: Protocol not supported
Mar 10 20:50:32 raspberrypi bluealsa[330]: /usr/bin/bluealsa: D: ../../src/bluealsa-dbus.c:373: Called: org.bluealsa.PCM1.Open()
Mar 10 20:50:32 raspberrypi bluealsa[330]: /usr/bin/bluealsa: E: Couldn't establish SCO link: Protocol not supported
Mar 10 20:50:32 raspberrypi bluealsa[330]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:698: State transition: 2 -> 2
Mar 10 20:50:32 raspberrypi /usr/bin/bluealsa: ../../src/ba-transport.c:698: State transition: 2 -> 2
Mar 10 20:50:39 raspberrypi /usr/bin/bluealsa: ../../src/a2dp.c:123: PCM read buffer flushed: 2000
Mar 10 20:50:39 raspberrypi bluealsa[330]: /usr/bin/bluealsa: D: ../../src/a2dp.c:123: PCM read buffer flushed: 2000
Mar 10 20:50:39 raspberrypi bluealsa[330]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:945: Closing PCM: 17
Mar 10 20:50:39 raspberrypi bluealsa[330]: /usr/bin/bluealsa: D: ../../src/sco.c:337: Releasing SCO due to PCM inactivity
Mar 10 20:50:39 raspberrypi /usr/bin/bluealsa: ../../src/ba-transport.c:945: Closing PCM: 17
Mar 10 20:50:39 raspberrypi /usr/bin/bluealsa: ../../src/sco.c:337: Releasing SCO due to PCM inactivity
```

closed 2020-03-11 by StuartIanNaylor