arkq/bluez-alsa #304: Lack of sound from microphone during call

zadzak, 2020-03-09:
Hi,
I would like build hands-free kit using raspberry pi.
My setup is Rpi4 with USB soundcard with connected wired headset. Rpi4 is paired with Redmi 7A. Both media and voice call profiles are active and connected. I can hear music from Redmi (Spotify) over A2DP profile as well as HFP sound (call waiting signal, remote caller voice). The only issue is with lack of sound from microphone. 
Microphone is ok; with simply command 'arecord | aplay' I can hear my voice in headset speakers.
Compiled bluealsa run in the following way:
sudo ./bluealsa -p a2dp-sink -p hfp-ofono
bluealsa-aplay:
./bluealsa-aplay -vv -profile-sco -d plughw:CARD=Device,DEV=0 00:00:00:00:00:00
./bluealsa-aplay -vv -profile-a2dp -d plughw:CARD=Device,DEV=0 00:00:00:00:00:00
Call starts with ofono:
./dial-number XXXXXXXX
I saw in other issue that amixer (amixer -D bluealsa) returns A2DP and SCO device but in my case returns only A2DP. 
Can you tell me why how to configure bluealso to take microphone sound and send over SCO link?

Thanks
Marcin

Arkq, 2020-03-10:
I guess, that you've enabled msbc. If that's the case, please disable it (do not specify `--enable-msbc` during configuration) - I was never able to stream audio via msbc if BT chip is connected via USB. Also, you might try build-in hfp support (just to confirm that the audio part works correctly). With hfp/ofono mixer should be available, but only after codec connection has been established - basically, after/during first call (if you are using pretty recent bluez-alsa version, preferably master branch).

zadzak, 2020-03-10:
Bluez-alsa master branch configured with the following command:
../configure --enable-debug --enable-debug-time --enable-aac --enable-ofono --enable-rfcomm
I disabled ofono service, then bluealsa run with command:
sudo ./bluealsa -p hfp-hf -p hfp-ag -p a2dp-sink
When call is connected logs show:
Couldn't lookup transport: /org/bluez/hci0/dev_C8_3D_DC_6A_AB_A5/sco
Microphone streaming still does not work.

borine, 2020-04-01:
@zadzak are you certain that the microphone has established a SCO connection? The lack of control element in amixer, and the transport error message suggest that it has not. You can confirm this with:
```
hcitool con
 ```

borine, 2021-10-02:
This issue has been idle for over a year now. Looking at it again, I see no mention of which application is being used to actually capture audio from the local microphone to send to BlueALSA; so I think that is most likely the missing piece from the setup described here. `bluealsa-aplay` does not provide a return audio stream for the local microphone - it only plays the stream from the remote device to the local speaker. I am closing the issue now.

closed 2021-10-02 by borine