arkq/bluez-alsa #306: /usr/share/dbus-1/system-services/org.bluealsa.service doesn't exist

StuartIanNaylor, 2020-03-10:
Dunno if its just the wiki or needed but on a make install that service file isn't created.

Should it?

StuartIanNaylor, 2020-03-11:
PS just thought I would post syslog of boot, nothing to do with above
```
Mar 11 00:18:21 raspberrypi systemd[1]: systemd-rfkill.service: Succeeded.
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/adapter.c:connected_callback() hci0 device 9B:FD:D5:6E:01:CA connected eir_len 10
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/profile.c:ext_confirm() incoming connect from 9B:FD:D5:6E:01:CA
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/service.c:btd_service_ref() 0x29e840: ref=2
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/profile.c:ext_confirm() HFP Audio Gateway authorizing connection from 9B:FD:D5:6E:01:CA
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/profile.c:ext_auth() 9B:FD:D5:6E:01:CA authorized to connect to HFP Audio Gateway
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/profile.c:ext_connect() HFP Audio Gateway connected to 9B:FD:D5:6E:01:CA
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/service.c:change_state() 0x29e840: device 9B:FD:D5:6E:01:CA profile HFP Audio Gateway state changed: disconnected -> connecting (0)
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.523673248: D: ../../src/bluez.c:1104: Called: org.bluez.Profile1.NewConnection()
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.524574785: D: ../../src/ba-rfcomm.c:1261: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.524741063: D: ../../src/ba-rfcomm.c:878: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/bluez.c:1104: Called: org.bluez.Profile1.NewConnection()
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.526055118: D: ../../src/sco.c:206: Created SCO dispatcher [ba-sco-dispatch]: hci0
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.526346933: D: ../../src/bluez.c:1029: HFP Audio Gateway (CVSD) configured for device 9B:FD:D5:6E:01:CA
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.526445248: D: ../../src/ba-transport.c:698: State transition: 0 -> 2
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.526601655: D: ../../src/sco.c:79: Starting SCO dispatcher loop: hci0
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.526921377: D: ../../src/ba-transport.c:972: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.527652044: D: ../../src/sco.c:247: Starting SCO loop: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/ba-rfcomm.c:1261: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/ba-rfcomm.c:878: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/sco.c:206: Created SCO dispatcher [ba-sco-dispatch]: hci0
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.532776933: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:023
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.532886063: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.533083859: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.533250933: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/bluez.c:1029: HFP Audio Gateway (CVSD) configured for device 9B:FD:D5:6E:01:CA
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/ba-transport.c:698: State transition: 0 -> 2
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/sco.c:79: Starting SCO dispatcher loop: hci0
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/ba-transport.c:972: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/sco.c:247: Starting SCO loop: HFP Audio Gateway (CVSD)
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/service.c:change_state() 0x29e840: device 9B:FD:D5:6E:01:CA profile HFP Audio Gateway state changed: connecting -> connected (0)
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/device.c:device_profile_connected() HFP Audio Gateway Success (0)
Mar 11 00:18:22 raspberrypi bluetoothd[467]: src/service.c:btd_service_ref() 0x29e840: ref=3
Mar 11 00:18:22 raspberrypi bluetoothd[467]: plugins/policy.c:service_cb() Added HFP Audio Gateway reconnect 0
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/at.c:161: AT message: SET: command:+BRSF, value:023
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
Mar 11 00:18:22 raspberrypi /usr/bin/bluealsa: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.541930211: D: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
Mar 11 00:18:22 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 18.542033563: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
[... cut ...]
Mar 11 00:23:31 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 316.349976367: D: ../../src/ba-transport.c:945: Closing PCM: 17
Mar 11 00:23:31 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 316.350223978: D: ../../src/a2dp.c:123: PCM read buffer flushed: 2000
Mar 11 00:23:31 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 316.350356181: D: ../../src/sco.c:337: Releasing SCO due to PCM inactivity
Mar 11 00:23:31 raspberrypi /usr/bin/bluealsa: ../../src/a2dp.c:123: PCM read buffer flushed: 2000
Mar 11 00:23:31 raspberrypi /usr/bin/bluealsa: ../../src/sco.c:337: Releasing SCO due to PCM inactivity
```

borine, 2020-03-11:
From the wiki page
> To manage bluealsa with systemd, create the following two files:
**/etc/systemd/system/bluealsa.service**
**/usr/share/dbus-1/system-services/org.bluealsa.service**

So, no, make install does not create them for you.

StuartIanNaylor, 2020-03-11:
Yeah it was /usr/share/dbus-1/system-services/org.bluealsa.service I notice I was missing and just thought I would ask as things seem much the same even when added a2dp works without and wasn't sure if required.
I copied the service from arch which is absent of the /usr/share/dbus-1/system-services/org.bluealsa.service

```
[Unit]
Description=BluezAlsa proxy
Requires=bluetooth.service
After=bluetooth.service

[Service]
Type=simple
EnvironmentFile=/etc/conf.d/bluealsa
#User=bluealsa
#Group=audio
ExecStart=/usr/bin/bluealsa $BLUEALSA_CONF

[Install]
WantedBy=multi-user.target
```
[Unit]
Description=BluezAlsa proxy
Requires=bluetooth.service
After=bluetooth.service

[Service]
Type=simple
EnvironmentFile=/etc/conf.d/bluealsa
#User=bluealsa
#Group=audio
ExecStart=/usr/bin/bluealsa $BLUEALSA_CONF

[Install]
WantedBy=multi-user.target
```

But using the wiki version now but both seem to work.

Apols on the length og the logs but thought will just give you the lot as wouldn't know what is a normal AT operation or error.
My question on the HFP-AG is that bluealsa side or device side? I expected to see HFP-HF in there but I am thinking of the device and guess bluealsa needs to be the HFP-AG device that connects a HFP-HF device but didn't see that in the logs, just HFP-AG?

StuartIanNaylor, 2020-03-11:
Just tried a new clean install of arch this time with the asus bt dongle.

Same but did follow https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration to the T

StuartIanNaylor, 2020-03-11:
Next BT 5 :)

closed 2020-03-11 by StuartIanNaylor