arkq/bluez-alsa #307: Testing speaker/audio sink

ashah7, 2020-03-12:
Is there a way to test if the Bluetooth device is connected and will work using bluealsa?
For example in pulseaudio on can run 
`pactl list sinks short`
to display connected devices, at that point you know its all connected and capable to play.
I'm looking for something similar here and not just Bluetooth commands like
`hcitool con `

graham8, 2020-03-12:
aplay -L | grep bluealsa

graham8, 2020-03-12:
try amixer -D bluealsa

Arkq, 2020-03-15:
Using `amixer` (as @graham8 has suggested) should result in a good estimate whether BT device is connected and will work with BlueALSA. Better approach is to use BlueALSA B-Bus API (which is internally used by `amixer`), because `amixer` might not show all devices. E.g. `amixer` will not show HFP capable device with codec selection support if first call has not been made yet.

To query for all connected BT devices use something as follows:
```
gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
```

closed 2020-03-15 by Arkq