arkq/bluez-alsa #308: How to route SCO over PCM?

adang322, 2020-03-12:
Hi,
I would like to route the SCO over PCM but not HCI, but have no idea how to configure it in bluez-alsa.
Any ones know it?
Thank you!

StuartIanNaylor, 2020-03-12:
Dunno really as been trying to get HFP/HSP profles working that in https://github.com/Arkq/bluez-alsa/issues/306.
Can you get hfp/hsp profiles working over HCI?
I just get
```Mar 11 00:22:54 raspberrypi /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()
Mar 11 00:22:54 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 279.401448574: D: ../../src/bluealsa-dbus.c:172: Called: org.bluealsa.Manager1.GetPCMs()
Mar 11 00:22:54 raspberrypi /usr/bin/bluealsa: ../../src/bluealsa-dbus.c:373: Called: org.bluealsa.PCM1.Open()
Mar 11 00:22:54 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 279.414051407: D: ../../src/bluealsa-dbus.c:373: Called: org.bluealsa.PCM1.Open()
Mar 11 00:22:54 raspberrypi bluealsa[365]: /usr/bin/bluealsa: 279.415505463: E: Couldn't establish SCO link: Protocol not supported
M```

Guess the old bluez details might tell you as not sure as there is also the old bluetooth-alsa that might give some ideas as it used to be a kernel module bt_sco but don't even know where its handled
https://www.kernel.org/doc/ols/2008/ols2008v1-pages-193-196.pdf
https://sourceforge.net/projects/bluetooth-alsa/

Anyway enough of my ignorance, but so want sco for HFP/HSP profiles

adang322, 2020-03-13:
Yes. i can get hfp/hsp profiles working over HCI.
I would like to route SCO over PCM(TDM/I2S), but I don't know where to configure it in bluez-alsa.
I found that in bluez4.101, there is an option "SCORouting" for SCO can be set to PCM or HCI in audio.conf, but it seems that it was removed in new version.

StuartIanNaylor, 2020-03-13:
So its almost the oppisite of the pcm fix for the Pi
https://github.com/raspberrypi/linux/issues/2229#issuecomment-359233111
I know sudo hcitool cmd 0x3F 0x01C 0x01 0x02 0x00 0x01 0x01 swaps from pcm to hci for the broadcom BT.

I dunno but presuming the routing now is a hardware thing? As above.

I have a PI4 but using a USB dongle as think on raspberry even if you fix the routing it still has a MTU fault so you get pops and hissing as its fixed on the pi @ 64 but is usually setup as 48.

PS what distro/hardware you using?

adang322, 2020-03-16:
My hardware: dspg(soc) +  Realtek(USB BT).

adang322, 2020-03-25:
I disabled below codes in function "io_thread_sco(..)" for testing.
`/*if (release) {
				t->release(t);
				asrs.frames = 0;
			}
			else */`

and then run "dbus-send --system --type=method_call --print-reply --dest=org.bluealsa /org/bluealsa/hci0/dev_XX_XX_XX_XX_XX_XX/sco org.bluealsa.PCM1.Open string:sink (or source)" to open SCO connect only. No need spk_pcm and mic_pcm. then play audio through PCM(TDM) interface, it will be ok.

closed 2020-03-25 by adang322