arkq/bluez-alsa #309: Simultaneous audio output for bluetooth device and audio jack

supersoaker, 2020-03-22:
Hi,

I would like to have a simultaneous audio output to a connected bluetooth speaker and the normal audio jack at the same time.
The audio output via the bluetooth works already, but I can't get it running for the audio jack.
Here is my `/etc/asound.conf`:
`
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.slave.pcm "jack"

pcm.rawjack {
    type jack
    playback_ports {
        0 system:playback_1
        1 system:playback_2
    }
    capture_ports {
        0 system:capture_1
        1 system:capture_2
    }
}

pcm.jack {
    type plug
    slave { pcm "rawjack" }
    hint {
        description "JACK Audio Connection Kit"
    }
}

`

Do you have an idea?

Btw: I'm using a raspberry pi gen. 3

supersoaker, 2020-03-30:
do you need more information for that or is that kind of use case not yet supported?

borine, 2020-03-30:
Its not clear what you are asking. 

Looking at your asound.conf, you appear to be using the JACK audio connection kit; and that is not working with the RPi 3.5mm socket. If that's right, then should you not be asking for help on a JACK forum or a RPi forum?

Or is there some  issue with bluealsa that you need help with?

supersoaker, 2020-04-05:
ah yeah, I reworked the configuration, but somehow it does not work either:
```
defaults.bluealsa {
  type plug
  slave.pcm {
    type bluealsa
    device "C0:28:8D:XX:XX:XX"
    profile "a2dp"
  }
}
```
I just want to use the analog audio plug of the raspberry pi with the bluetooth device as a slave together.

Btw here are the error that I get when I execute `aplay -D bluealsa /usr/share/sounds/alsa/Noise.wav`:
```
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
aplay: main:828: audio open error: No such file or directory
```

borine, 2020-04-05:
I think you have misunderstood the use of "default" in alsa configurations. The only argument to pcm.bluealsa whose default you need to define here is `device`.

In a plain alsa system, the way to  get a single audio stream to be directed to 2 pcms is  the "multi" plugin. Be aware though, that this plugin does not *synchronize" the streams. It is extemely likely that the bluetooth speaker will have a much higher latency than the jack plug. So to synchronize, you would need to add a delay to the jack output. I don't know how to do that with plain alsa - you would have to seek advice from ALSA experts somewhere.

An example .asoundrc that would achieve simultaneous, but not synchronized, output is:
```
defaults.bluealsa.device "C0:28:8D:XX:XX:XX"
pcm.jackplug "hw:0,0"

pcm.Duplicate {
	type plug
	slave.pcm {
		type multi
		slaves {
			a { pcm "jackplug" channels 2 }
			b { pcm "bluealsa" channels 2 }
		}
		bindings [
			{ slave a channel 0 }
			{ slave a channel 1 }
			{ slave b channel 0 }
			{ slave b channel 1 }
		]
		master 1
	}
	ttable [
		[ 1 0 1 0 ]
		[ 0 1 0 1 ]
	]
	hint {
		show on
		description "Duplicate output to jack plug and bluealsa"
	}
}
```

redefine "pcm.jackplug" if hw:0,0 is wrong for your setup.

Now , for example,
```aplay -D Duplicate testfile.wav```

will play to both outputs, most likely with an uncomfortable "echo" effect because the two outputs are not synchronized.

graham8, 2020-04-05:
You can use an upmix plugin which has a delay parameter (if the difference in latency is predictable!)
```
pcm.bothdevices {
	type upmix
	slave "pairup"
	channels 4
	delay 170  ## adjust here as necessary
	hint {
		show {
			@func refer
			name defaults.namehint.extended
		}
		description "device 1 and device 2"
	}
}

pcm_slave.pairup.pcm {
	type plug
	slave.pcm {
		type multi
			slaves {
				a { pcm "device1" channels 2 }
				b { pcm "device2" channels 2 }
			}
		bindings [
		{ slave a channel 0 }
		{ slave a channel 1 }
		{ slave b channel 0 }
		{ slave b channel 1 }
		]
	}
} 
```

borine, 2021-06-21:
This issue appears to be resolved.

closed 2021-06-21 by borine