arkq/bluez-alsa #310: Soundcard is paired an connect but not found otherwise.

roberth1990, 2020-03-28:
Hello

I can't get this to work. AFAIK I paired and connected correctly:

```
[bluetooth]# pair 7C:96:D2:30:DD:E7
Attempting to pair with 7C:96:D2:30:DD:E7
[CHG] Device 7C:96:D2:30:DD:E7 Connected: yes
[CHG] Device 7C:96:D2:30:DD:E7 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 7C:96:D2:30:DD:E7 ServicesResolved: yes
[CHG] Device 7C:96:D2:30:DD:E7 Paired: yes
Pairing successful
[CHG] Device 7C:96:D2:30:DD:E7 ServicesResolved: no
[CHG] Device 7C:96:D2:30:DD:E7 Connected: no
[bluetooth]# connect 7C:96:D2:30:DD:E7
Attempting to connect to 7C:96:D2:30:DD:E7
[CHG] Device 7C:96:D2:30:DD:E7 Connected: yes
Connection successful
[CHG] Device 7C:96:D2:30:DD:E7 ServicesResolved: yes
```

```
[roberth@Gerd ~]$ cat ~/.asoundrc
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "7C:96:D2:30:DD:E7"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000
```

```
[roberth@Gerd ~]$ aplay -D bluealsa:SRV=org.bluealsa,DEV=7C:96:D2:30:DD:E7,PROFILE=a2dp tasmanian-devil-daniel_simon.wav
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM bluealsa:SRV=org.bluealsa,DEV=7C:96:D2:30:DD:E7,PROFILE=a2dp
aplay: main:830: audio open error: No such file or directory
```

```
[roberth@Gerd ~]$ systemctl status bluealsa
● bluealsa.service - BluezAlsa proxy
     Loaded: loaded (/usr/lib/systemd/system/bluealsa.service; enabled; vendor preset: disabled)
     Active: active (running) since Sat 2020-03-28 14:25:46 CET; 13min ago
   Main PID: 1521 (bluealsa)
      Tasks: 3 (limit: 16692)
     Memory: 1.2M
     CGroup: /system.slice/bluealsa.service
             └─1521 /usr/bin/bluealsa

mars 28 14:25:46 Gerd systemd[1]: Started BluezAlsa proxy.
```

```
[roberth@Gerd ~]$ cat /proc/asound/cards
 0 [Generic        ]: HDA-Intel - HD-Audio Generic
                      HD-Audio Generic at 0xfcd88000 irq 54
 1 [Generic_1      ]: HDA-Intel - HD-Audio Generic
                      HD-Audio Generic at 0xfcd80000 irq 55
```

None of those is the soundcard of course.

`bluealsa-aplay 7C:96:D2:30:DD:E7`

Gives no sound or any output in the shell.

Any help would be appreciated.

borine, 2020-03-28:
Are there any error messages from the bluealsa daemon?
```
journalctl -xeu bluealsa.service
```
what version of bluealsa are you running (I assume it is from some distribution repository?)
```
bluealsa --version
```
For bluealsa-aplay, you need to start bluealsa  with at least `-p a2dp-sink` in order to be able to receive a2dp audio streams; and  `-p a2dp-sink -p a2dp-source` to use both bluealsa-aplay and aplay; see `bluealsa --help`

borine, 2020-03-28:
update: the libasound error message
> `ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM bluealsa:SRV=org.bluealsa,DEV=7C:96:D2:30:DD:E7,PROFILE=a2dp`

means that alsa cannot find a definition for a pcm called "bluealsa" in its configuration. This is usually caused by the bluealsa conf file (20-bluealsa.conf) being in the wrong directory, or perhaps being corrupted. The correct directory depends on what version of alsa you are using and what options it was built with by the distribution.

roberth1990, 2020-03-28:
> Are there any error messages from the bluealsa daemon?
>
> ```
> journalctl -xeu bluealsa.service
> ```
>
> what version of bluealsa are you running (I assume it is from some distribution repository?)
>
> ``` [...]

```
mars 28 20:02:00 Gerd systemd[1]: Started BluezAlsa proxy.
-- Subject: A start job for unit bluealsa.service has finished successfully
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A start job for unit bluealsa.service has finished successfully.
-- 
-- The job identifier is 1154.
```

```
[roberth@Gerd ~]$ bluealsa --version
v2.1.0-26-g14046ec
```

> update: the libasound error message
>
>
> means that alsa cannot find a definition for a pcm called "bluealsa" in its configuration. This is usually caused by the bluealsa conf file (20-bluealsa.conf) [...]

Looks like I had to manually copy that file yes, but now bluez can't connect to device at all, just pair.

borine, 2020-03-28:
You connected OK before, but not now, so something else has changed (the alsa config file plays no part in bluez connection). 
- Is the bluealsa service still running?
- What command line arguments are you starting it with?
- Are there any error messages now from bluealsa or bluetooth (journalctl) ?

roberth1990, 2020-03-28:
The service is still running.
I don't use any commanline arguments.
No error messages in journalctl.

borine, 2020-03-29:
> No error messages in journalctl

Hmm - the bluetooth service typically logs a message when a connection attempt fails or is rejected. In addition to the journalctl command above for bluealsa, did you also try for bluetooth?
```
journalctl -xeu bluetooth.service
```

roberth1990, 2020-03-29:
`mars 29 13:31:13 Gerd bluetoothd[521]: connect error: Permission denied (13)
`

borine, 2020-03-29:
That connect error might mean that the remote bluetooth device is rejecting the connection request. If so, then maybe its already connected to some other system, or maybe it has lost its pairing info with the bluealsa system. If the latter, then deleting it in bluetoothctl then pairing again should fix it.

roberth1990, 2020-03-29:
Okay, I unpaired the device and paired it again and connected to it, and added the 20-bluealsa.conf to a directory where it can used.

`aplay -D bluealsa:SRV=org.bluealsa,DEV=7C:96:D2:30:DD:E7,PROFILE=a2dp tasmanian-devil-daniel_simon.wav`

Works fine, but otherwise the soundcard isn't detected with `cat /proc/asound/cards` or `aplay -l`, kmix or alsamixer.

borine, 2020-03-30:
> Works fine

Excellent news !

> soundcard isn't detected with cat /proc/asound/cards or aplay -l

That is correct and as expected.  /proc/asound/ is created by the linux kernel and holds information about all the **hardware sound cards** plugged into the system.

The kernel sees your bluetooth adapter as a **network** device, not a sound card.

The bluealsa pcm and ctl objects are "software-defined" artifacts that exist only in user space. They are created by alsa when an alsa application reads the file `20-bluealsa.conf` as you discovered earlier in this thread. There is no **card** involved.

alsa can list the bluealsa ctl objects, for example with:
```
alsamixer -D bluealsa
```
but there is no equivalent alsa utility that will list the pcm objects.

[ @Arkq : just a thought - maybe bluealsa-aplay could be extended with to list connected bluealsa pcms as `bluealsa-aplay -l` ? Maybe I'll think about that more sometime ]

You can list them using bluealsa's D-Bus interface, for example:
```
gbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
```
Going back to the original issue with the alsa config file, I would expect the bluealsa version you are using
> v2.1.0-26-g14046ec

to put 20-bluealsa.conf in the correct directory with `make install`

If you built bluealsa yourself from source, can you say what configure options you used and how you installed the build result onto your system? If configure got this wrong then you may have uncovered a bug.

Arkq, 2020-03-30:
> maybe bluealsa-aplay could be extended with to list connected bluealsa pcms as bluealsa-aplay -l ?

@borine excelent idea, I'll add that. Also I will add note to readme that software PCM is not visible in kernel, because this question pops out from time to time :)

roberth1990, 2020-05-20:
> > Works fine
> 
> Excellent news !
> 
> > soundcard isn't detected with cat /proc/asound/cards or aplay -l
> 
> That is correct and as expected. /proc/asound/ is created by the linux kernel and holds information about all the **hardware sound cards** plugged into the system.
> 
> The kernel sees your bluetooth adapter as a **network** device, not a sound card.
> 
> The bluealsa pcm and ctl objects are "software-defined" artifacts that exist only in user space. They are created by alsa when an alsa application reads the file `20-bluealsa.conf` as you discovered earlier in this thread. There is no **card** involved.
> 
> alsa can list the bluealsa ctl objects, for example with:
> 
> ```
> alsamixer -D bluealsa
> ```
> 
> but there is no equivalent alsa utility that will list the pcm objects.
> 
> [ @Arkq : just a thought - maybe bluealsa-aplay could be extended with to list connected bluealsa pcms as `bluealsa-aplay -l` ? Maybe I'll think about that more sometime ]
> 
> You can list them using bluealsa's D-Bus interface, for example:
> 
> ```
> gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
> ```
> 
> Going back to the original issue with the alsa config file, I would expect the bluealsa version you are using
> 
> > v2.1.0-26-g14046ec
> 
> to put 20-bluealsa.conf in the correct directory with `make install`
> 
> If you built bluealsa yourself from source, can you say what configure options you used and how you installed the build result onto your system? If configure got this wrong then you may have uncovered a bug.

Sorry for the late reply, haven't been able to use my linux machine for a while.

So the device appears in when running:

`alsamixer -D bluealsa`

It's unmuted and I am still not getting any sound, with for example firefox. Kmix doesn't detect the card at all.

borine, 2020-05-20:
> with for example firefox

As far as I know, firefox supports only pulseaudio sound on linux - I think you cannot use it with *any* alsa device; you have to go through pulse audio.

roberth1990, 2020-05-20:
Well that is correct, but when using firefox with pulseaudio emulation for firefox, there is still no sound.

And yet kmix isn't finding the soundcard either.

borine, 2020-05-20:
>  right now I how specify for every application to use pcm device.

In your ~/.asoundrc you can define the default pcm (and ctl) to be an alias for your bluetooth device, so that alsa applications will just use it: for example
```
pcm.!default "bluealsa:DEV=xx:xx:xx:xx:xx:xx,PROFILE=a2dp"
ctl.!default "bluealsa"
```
Then 
```
aplay myfile.wav
```
will play to the bluealsa device. The downside there is that it is not easy to switch between bluetooth and on-board sound card.

> But how do I set up an audio mixing solution? 

Well, I suppose the simple, and unhelpful, answer is to use pulseaudio - that's what most linux desktop distributions do.

If you are determined to connect all your applications directly to alsa, then alsa's own solution is the "dmix" plugin, normally wrapped by the "plug" plugin. Setting all that up is really out of scope for this list though. The downside is that alsa's dmix can only mix streams into a hardware (ie kernel driver) device - so we are back to the discussion above about bluealsa not being a hardware sound card - for that reason you cannot use dmix with it. So right now there is no alsa stream mixing solution available for bluealsa, or any other software pcm.

There is a possible, "clunky" workaround to use dmix indirectly, and I am investigating that at the moment. It will be several more weeks yet before I know whether I can produce a working solution. The trick is to use alsa's "Loopback" driver (snd-aloop), to create a virtual card that is provided by a kernel driver so can be accessed via dmix: then use the alsaloop application to pass the mixed audio stream from the virtual card to a bluealsa device. I said it was "clunky" !! :smile:  

If you have a look at #48 you will see a suggested "patch" (actually more of a "hack" as it breaks so many other things) that enables bluealsa to work with alsaloop and therefore indirectly with dmix. I am currently looking into why alsaloop does not behave well with bluealsa in the hope of producing a more robust solution.

Even if I can get it to work reliably, this whole setup is painfully complicated; and so if your interest is just to be able to use bluetooth headphones within a linux desktop environment, then maybe pulseaudio is not such a bad solution?

Bluealsa is a very excellent bluetooth audio service; but that's all it does; the old Unix application mantra of "do one thing and do it well".  For a desktop you also need stream routing to permit use of different output devices and stream mixing to permit use of multiple applications. Those things require external applications, and unfortunately I don't know of any that have been written yet for use with bluealsa. Pulseaudio has a different mantra: "Try to do everything and don't worry about the compromises that creates". Only you can decide which is best for your particular use case.

roberth1990, 2020-05-20:
Oh well, I can live without dmix anyway untill I get the bluetooth speakers replaced. I just thought that since kmix couldn't detect the card meant something was very wrong.

borine, 2020-05-20:
> kmix couldn't detect the card

well, yes. bluealsa is not a card. Seek advice on a kmix / kde site - perhaps kmix can be configured to see software PCMs as well as cards. I don't know kmix, but some other applications will list software pcms provided they are explicitly named in .asoundrc. for example:
```
pcm.mybtspeaker "bluealsa:DEV=xx:xx:xx:xx:xx:xx,PROFILE=a2dp"
```
for some other applications you need a more complete definition, eg:
```
pcm.mybtspeaker {
        type plug
        slave.pcm "bluealsa:DEV=xx:xx:xx:xx:xx:xx,PROFILE=a2dp"
        hint.show on
        hint.description "bluetooth speaker"
}
```
with the second one, now `aplay -L` will give
```
mybtspeaker
    bluetooth speaker
```

hboetes, 2020-05-30:
> > with for example firefox
> 
> As far as I know, firefox supports only pulseaudio sound on linux - I think you cannot use it with _any_ alsa device; you have to go through pulse audio.

I searched a bit around and now I compile Firefox with jack support. I hope they enable that in the normal builds.

borine, 2021-06-21:
I'm closing this as I think there is not much more that can be said. Bluealsa is not a soundcard, so applications that look only for soundcards will not find it. That is the applications fault. Bluealsa does not currently implement PCM stream mixing, and ALSA's dmix plugin works only with soundcards.

closed 2021-06-21 by borine