arkq/bluez-alsa #312: bluepulse not working

DasNaCl, 2020-04-02:
So I'm trying to use bluez-alsa with pulseaudio 13.0.0
For that, I use the bluepulse script as written down in the wiki.
Unfortunately, it can't connect to the running (!) pulseaudio server instance. Upon connecting my headset via bluetooth it prints:

```
λ ~/ sudo bluepulse.bash
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
```

Manually running `pactl` also just prints `Failure: Module initialization failed`.

How can I debug this properly?

borine, 2020-04-03:
Hi DasNaCl
I posted the pulseaudio wiki page so I'm afraid any errors in it are entirely down to me :(

There have been several revisions over the last few weeks, and i know at least one of them had a typo in the script that caused errors, so to make sure we are looking at the same page, could I ask you to get the latest version of the script from the wiki which I updated yesterday (April 2)

First up:
> λ ~/ sudo bluepulse.bash

I had not intended that the script should be run as root; in fact if you are using pulseaudio in a normal desktop session you most definitely should **not** use sudo.

Next:
 > Manually running pactl also just prints Failure: Module initialization failed.

We need to debug that before attempting to run the script, since the script depends on `pactl load-module` to function.

Can you please:
- make sure bluepulse.bash is **not** running
- connect your bluetooth headset
- run this command (substitute your headset bt address and change a2dp to sco if necessary): 
```
pactl load-module module-alsa-sink device=bluealsa_raw:DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp fragments=1 fragment_size=960 sink_name=btheadset sink_properties=device.description=btheadset
```
Please report the output here

DasNaCl, 2020-04-03:
Hey borine,
Thank you for your quick reply! :)
The output suggests that everything works fine:

```
λ ~/ pactl load-module module-alsa-sink device=bluealsa_raw:DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp fragments=1 fragment_size=960 sink_name=btheadset sink_properties=device.description=btheadset
20
λ ~/ 
```

The device also appears in pavucontrol.

borine, 2020-04-03:
> The device also appears in pavucontrol.

And can you play sound to your headset using pulse ?

borine, 2020-04-03:
I've just realized that the script requires gnu awk (gawk) because it uses a specific gnu extension. I'll change the text and script to highlight this. You can check your version of awk with
```
awk -W version
```
If it says "mawk" or anything other than "GNU Awk" then the bluepulse.bash script will fail.

DasNaCl, 2020-04-03:
Unfortunately, I cannot play any sound. I'm running gawk, so there should be no issue on that side.
Running the dbus-monitor command from the script manually revealed that it needs additional permissions:

```
λ ~/ dbus-monitor --system type='signal',sender='org.bluealsa',interface='org.bluealsa.Manager1' sender='org.freedesktop.DBus',member='NameOwnerChanged',arg0='org.bluealsa',arg2=''
dbus-monitor: unable to enable new-style monitoring: org.freedesktop.DBus.Error.AccessDenied: "Rejected send message, 1 matched rules; type="method_call", sender=":1.62" (uid=1000 pid=14413 comm="dbus-monitor --system type=signal,sender=org.bluea") interface="org.freedesktop.DBus.Monitoring" member="BecomeMonitor" error name="(unset)" requested_reply="0" destination="org.freedesktop.DBus" (bus)". Falling back to eavesdropping.
signal time=1585910562.844908 sender=org.freedesktop.DBus -> destination=:1.62 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.62"
^C
λ ~/ sudo dbus-monitor --system type='signal',sender='org.bluealsa',interface='org.bluealsa.Manager1' sender='org.freedesktop.DBus',member='NameOwnerChanged',arg0='org.bluealsa',arg2=''
[sudo] password for dasnacl:
signal time=1585910592.181565 sender=org.freedesktop.DBus -> destination=:1.63 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.63"
signal time=1585910592.181618 sender=org.freedesktop.DBus -> destination=:1.63 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.63"
^C
λ ~/
```

I tested the bluepulse script again with `sudo` in front of the dbus-monitor command, this does seem to work. Is there a group I need to add my user to in order to not have the sudo command in the script?

borine, 2020-04-03:
Ignore the dbus-monitor "error" message - in fact its just a warning that it can't use "new-style monitoring" and is instead using eavesdropping mode. That is expected and the script just ignores that message.

Can you now please:
- stop bluepulse.bash
- disconnect the bluetooth headset
- run this command:
dbus-monitor --system "type='signal',sender='org.bluealsa',interface='org.bluealsa.Manager1'"
- connect the bluetooth headset (and wait a few seconds for some output from dbus-monitor)
- Ctrl-C to stop the dbus-monitor
- post the dbus-monitor output here

DasNaCl, 2020-04-03:
```
λ ~/ dbus-monitor --system "type='signal',sender='org.bluealsa',interface='org.bluealsa.Manager1'"
dbus-monitor: unable to enable new-style monitoring: org.freedesktop.DBus.Error.AccessDenied: "Rejected
r='org.bl") interface="org.freedesktop.DBus.Monitoring" member="BecomeMonitor" error name="(unset)" req
signal time=1585914581.888972 sender=org.freedesktop.DBus -> destination=:1.58 serial=2 path=/org/freed
   string ":1.58"
signal time=1585914594.907748 sender=:1.1 -> destination=(null destination) serial=66 path=/org/bluealsa; interface=org.bluealsa.Manager1; member=PCMAdded
   object path "/org/bluealsa/hci0/dev_XX_XX_XX_XX_XX_XX/a2dpsrc/sink"
   array [
      dict entry(
         string "Device"
         variant             object path "/org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX"
      )
      dict entry(
         string "Transport"
         variant             string "A2DP-source"
      )
      dict entry(
         string "Mode"
         variant             string "sink"
      )
      dict entry(
         string "Format"
         variant             uint16 32784
      )
      dict entry(
         string "Channels"
         variant             byte 2
      )
      dict entry(
         string "Sampling"
         variant             uint32 48000
      )
      dict entry(
         string "Codec"
         variant             uint16 2
      )
      dict entry(
         string "Delay"
         variant             uint16 0
      )
      dict entry(
         string "SoftVolume"
         variant             boolean true
      )
      dict entry(
         string "Volume"
         variant             uint16 32639
      )
   ]
```

borine, 2020-04-03:
So, no permissions problem then. I think we now need to rewind to start afresh and try the script from the command line. To ensure we have a fresh start, please can you:

- disconnect the bluetooth headset
- stop bluepulse.bash
- restore the bluepulse.bash script to exactly as it is on the wiki page
- if any bluetooth devices are showing in either the "Output Devices" or "Input Devices" tabs of pavucontrol they must be manually unloaded. For outputs, type `pactl list sinks`, look for each bluetooth sink, and note its "Owner Module" number. Then type `pactl unload-module N` where N is the Owner Module number. Repeat for any Inputs, using `pactl list sources`. Check that all bluetooth devices are gone from pavucontrol.
- restart pulseaudio:
```
pulseaudio --kill
pulseaudio --start
```
- start the bluepulse.bash script
- connect the bluetooth headset
- check pavucontrol to see if now has an Output for your headset called "Bluetooth: _alias_ (a2dp)"
- try to play some sound to the bluetooth Output
- try switching the saudio tream to the  on-board sound card then back to bluetooth

Thanks

DasNaCl, 2020-04-03:
Only the output device is added. Sound is good.
After switching to the soundcard output and back to the headphones, the sound is not there anymore.

borine, 2020-04-03:
OK, thanks for your feedback, it has helped me improve the wiki page. Hope you get this working correctly soon.

DasNaCl, 2020-04-03:
Thank you for your kind words.

I've managed to get it to work after disabling a few profiles for the running bluealsa instance. Unfortunately, the headset is now always in hfp mode. I can get it to a2dp if I disable hfp-ag.

What is more, the icons in the script are not used really, since everything is just `ICON_DEFAULT`. On my system, I had to remove the "bluetooth" suffixes from the icon names to get a valid icon instead of a box.

borine, 2020-04-09:
@DasNaCl if you are still using this, it may help to know that I've discovered that I was unwittingly relying on the pulse "suspend-on-idle" module to be able to switch from a2dp to sco on the same device. I've updated the wiki to make this explicit. Perhaps that explains why you needed to disable a2dp. There may still be other obstacles to overcome, but I would be interested to know if adding the suspend-on-idle module as described in the wiki helps in your use case.

DasNaCl, 2020-04-09:
Thank you for giving some additional thoughts to this!

The suspend-on-idle module was loaded with a default of 5 seconds. I could notice the white background noise to stop after that time.
Now, the setting is set to two seconds, allow me to get rid of the noise earlier.
I guess this counts as working.

What I think is interesting: Whether I just use audio output (i.e. no mic, but hfp-ag is loaded) or not, the noise is present. In my setup before, were I was not using pulseaudio, I did not have any noticeable noise.

borine, 2020-04-09:
> One caveat: Make sure that pavucontrol is closed, otherwise, the device won't idle

oops! I obviously haven't tested this with pavucontrol running. There does not appear to be any obvious workaround other than not to use pavucontrol. I'll add this to the wiki page too.
 
>  In my setup before, were I was not using pulseaudio, I did not have any noticeable noise.

I'll have to think about that one. Is the noise related to the use of pavucontrol too? I haven't noticed any such noise here.

> So, there is no a2dp if I run bluealsa with hfp-ag.

Was that true before you introduced pulseaudio? If so, then I guess its just the way your headset works?

Thanks again for all this excellent feedback

closed 2021-06-23 by borine