arkq/bluez-alsa #313: Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files

b23prodtm, 2020-04-03:
Hi,
 following the script at [Wiki Systemd integration](https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration), the system can't start on 

    exec /usr/bin/bluealsa-aplay --pcm-buffer-time=$PCM_BUFFER_TIME -d $BLUE_SPEAKERS,0 00:00:00:00:00:00

 Also alsa play with bluealsa as output speaker device 

    aplay -D bluealsa sounds/connect.wav
    bluealsa-pcm.c:802:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files

b23prodtm, 2020-04-03:
It's a docker image built on Debian stretch. How would I reboot dbus service ? 
I did
    
     systemctl reload dbus.service

borine, 2020-04-04:
> It's a docker image

Ok, so that's important information. I need to admit here that I know nothing of container technology - I was already retired from IT when that became a "thing".

In case its helpful to you, here's a quick sketch of the bluealsa architecture that you need to map onto your container layout.

- aplay needs to communicate with an instance of the bluealsa daemon. It finds the daemon by exchanging messages with the system d-bus. So aplay and bluealsa both need to "see" the same instance of d-bus.
- the bluealsa daemon needs to communicate with the bluetooth daemon, again brokered by d-bus. So the bluetooth daemon also needs to "see" that same instance of d-bus.
- both the bluealsa and bluetooth daemons need to communicate with the bluetooth adapter by sending messages across the hci. The hci is a virtual device created by the kernel, and only one instance of bluetooth and one instance of bluealsa can access it at a time.

It seems to me that the most likely cause of your issue is that the container boundaries are blocking one or more of these communication channels. How to fix that, I have no idea. Hopefully someone else can advise. If this can be solved, it might make a useful wiki article in its own right.

b23prodtm, 2020-04-05:
I see now, talking about D-Bus, it was to exchange messages with the system instance. 
Looking backwards to my Dockerfile configuration and printing the exact D-Bus configuration, 

     dbus-daemon --config-file=/usr/share/dbus-1/system.conf --print-address

Shows off that d-bus wasn't mounted into the container and unreachable. It's the environment variable setting that failed on startup. So I corrected it, accordingly using quotes "" 🥇 

I 'm about to retest and feed back in if anything's wrong.
Source:    
         - https://georgik.rocks/how-to-start-d-bus-in-docker-container/
         - https://www.balena.io/docs/learn/develop/runtime/#dbus-communication-with-host-os

b23prodtm, 2020-04-05:
I've been using the wrong DBus path until I reset the environement variable associated to DBUS_SYSTEM_BUS_ADDRESS. It was fixed now, and bluealsa gets access to the host devices bus. Thank you @borine I'm closing this issue !

closed 2020-04-05 by b23prodtm