arkq/bluez-alsa #315: sco acquistion failure not handled

borine, 2020-04-08:
Sometimes, when a client opens a sco pcm for playback on an AG instance, transport acquisition can fail. This failure is ignored by the sco io thread, so the client becomes blocked writing to a full pipe that the server never reads from. At the very least, the client pcm should be closed when no transport can be acquired (the a2dp code does this).

I can trigger this by first starting an a2dp playback client, then starting a sco playback client to the same device (using `bluealsa -p a2dp-source -p hfp-ag`). Once this has happened with my device, all future sco attempts fail, even after the a2dp transport has been released. Disconnecting the device then reconnecting restores normal function.

borine, 2020-04-09:
I'm unable to reproduce the acquisition failure today. The connect() call was failing with ECONNABORTED which bluez maps from HCI error 0x16 "Connection terminated by local host", but whatever was causing it has corrected itself now :(
I think the issue is still valid, but just very hard to deliberately trigger.

Arkq, 2020-04-09:
I think that this issue is real. Sometimes, I've also experienced such behavior.... It was when I was mostly focused on HFP - I've run a lot of "tests". And I remember that from time to time bluealsa server got stuck until BT device reconnection or what worst bluealsa server restart.... I will look at this, but rather in a distant future :|

commit 49ad3488 (2020-04-15): Synchronously acquire SCO link in D-Bus handler

SCO transport acquisition for Audio Gateway was done asynchronously in
the SCO thread after PCM_OPEN signal delivering. In case of failure, PCM
client was not notified, which might have caused an unexpected behavior.
In order to simplify client notification, we are going to acquire SCO
link in the D-Bus method call handler - like A2DP Source transport.

Also, fix PCM denial-of-service in case of transport acquisition failure.
Previously, when the acquire() has failed, the pcm->fd was not cleared,
which has caused PCM DoS until BT device reconnec [... cut]
closed 2020-04-15 by Arkq