arkq/bluez-alsa #316: Audio speaker is connected but not playing music

davinci-klod, 2020-04-10:
Hi, 

I'm facing an issue which sounds recurrent but I'm not able to fix it despite the fact I've searched a solution for hours.

I'm trying to connect my bluetooth audio speaker to my RPI Zero. Pairing and connecting it works very well, using bluetoothctl. Here what's the info command returns to me : 

```
Device FC:58:FA:19:E9:21 (public)
        Name: Punker
        Alias: Punker
        Class: 0x00240414
        Icon: audio-card
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
```

To test it, I use `aplay` but nothing comes out of the speaker. So I've used many commands I've seen in similar threads, like  `aplay -D bluealsa jmr.wav` that returns to me :  

```
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
aplay: main:828: audio open error: No such file or directory
```

And amixer -D bluealsa : 

```
ALSA lib control.c:1375:(snd_ctl_open_noupdate) Invalid CTL bluealsa
amixer: Mixer attach bluealsa error: No such file or directory
```

It looks like amixer is not able to recognize bluealsa.
In alsamixer, sound is enabled at maximum. By the way, I only have the sound card `bcm2835 ALSA`.

`bluealsa-aplay` doesn't work neither.

What's service bluetooth status returns : 

bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-04-10 06:17:19 UTC; 27min ago
     Docs: man:bluetoothd(8)
 Main PID: 316 (bluetoothd)
   Status: "Running"
   Memory: 2.8M
   CGroup: /system.slice/bluetooth.service
           └─316 /usr/lib/bluetooth/bluetoothd

```
Apr 10 06:30:10 CUISINE bluetoothd[316]: Authentication attempt without agent
Apr 10 06:30:10 CUISINE bluetoothd[316]: Hands-Free Voice gateway rejected FC:58:FA:19:E9:21: org.bluez.Error.Rejected
Apr 10 06:30:11 CUISINE bluetoothd[316]: Unable to get connect data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 10 06:30:11 CUISINE bluetoothd[316]: Endpoint registered: sender=:1.9 path=/A2DP/SBC/Source/2
Apr 10 06:31:43 CUISINE bluetoothd[316]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource
Apr 10 06:31:43 CUISINE bluetoothd[316]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink
Apr 10 06:31:43 CUISINE bluetoothd[316]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)
Apr 10 06:31:43 CUISINE bluetoothd[316]: Unable to get connect data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 10 06:32:11 CUISINE bluetoothd[316]: Endpoint unregistered: sender=:1.28 path=/MediaEndpoint/A2DPSource
Apr 10 06:32:11 CUISINE bluetoothd[316]: Endpoint unregistered: sender=:1.28 path=/MediaEndpoint/A2DPSink
```

I'm not sure It's necessary but I have also run pulseaudio using `pulseaudio --start` after running theses two commands to enable ALSA : 

```
sudo modprobe snd-bcm2835                      # load module for single boot
echo "snd-bcm2835" | sudo tee -a /etc/modules  # load module for persistance
```

I am a little lost. The most frustrating thing is that I succeed to get a sound and the problem occurred after a reboot. I don't know what happened to make it stop working. I restarted the RPI and paired / connected the audio speaker several times but nothing helped.

Did you have a solution for me ?
Thanks in advance :)

borine, 2020-04-10:
You should normally choose to run _either_ bluealsa _or_ pulseaudio when using bluetooth audio. If you really need both services, then make sure to disable bluetooth modules in pulseaudio.

My first guess at your issue is that you are actually using pulseaudio for bluetooth, maybe even bluealsa is not running at all?

davinci-klod, 2020-04-10:
Ok. I didn't understand I don't have to use the two !

Bluealsa seems to run : 

```
root@CUISINE:/home/pi# service bluealsa status
● bluealsa.service - BluezALSA proxy
   Loaded: loaded (/lib/systemd/system/bluealsa.service; static; vendor preset: enabled)
   Active: active (running) since Fri 2020-04-10 06:17:19 UTC; 1h 52min ago
 Main PID: 320 (bluealsa)
   Memory: 2.5M
   CGroup: /system.slice/bluealsa.service
           └─320 /usr/bin/bluealsa

Apr 10 06:17:19 CUISINE systemd[1]: Started BluezALSA proxy.
Apr 10 06:30:11 CUISINE bluealsa[320]: /usr/bin/bluealsa: Unsupported AT message: CMD: command:+CLCC, value:(null)
Apr 10 06:30:11 CUISINE bluealsa[320]: /usr/bin/bluealsa: Unsupported AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
```

In fact, I need Pulseaudio to use my RPI Zero as a Multiroom client (like suggested in this tutorial : https://github.com/skalavala/Multi-Room-Audio-Centralized-Audio-for-Home/blob/master/Setup%20Pulseaudio.md). 

Is there is a way to make it work without disabling pulseaudio ?

borine, 2020-04-10:
> Is there is a way to make it work without disabling pulseaudio ?

Yes, just disable the pulseaudio bluetooth modules, the rest of pulseaudio function is not affected by this.

I've written some notes on the wiki [here](https://github.com/Arkq/bluez-alsa/wiki/PulseAudio-integration) on how I use bluealsa and pulseaudio together. It works for me, but is not well tested on different systems - not at all on a Pi Zero. It does explain how to disable pulseaudio bluetooth, so that part may be helpful for you; whether the rest of it is useful depends on exactly what you want to achieve.

davinci-klod, 2020-04-10:
I followed the part concerning the deactivation of the module but still no luck. 
For the moment, I'm just trying to be able to play a sound without using my multiroom system which requires Pulseaudio. It would be a good start!I started again from a clean installation without having installed pulseaudio and still nothing. Same problem so I don't think it comes from Pulseaudio!

`alsamixer -D bluealsa` returns this to me : 

```
ALSA lib control.c:1375:(snd_ctl_open_noupdate) Invalid CTL bluelalsa
cannot open mixer: No such file or directory
```

Did you have an idea ?

borine, 2020-04-10:
>ALSA lib control.c:1375:(snd_ctl_open_noupdate) Invalid CTL _**bluelalsa**_
cannot open mixer: No such file or directory

where did "_**bluelalsa**_" come from?

davinci-klod, 2020-04-11:
> > ALSA lib control.c:1375:(snd_ctl_open_noupdate) Invalid CTL _**bluelalsa**_
> > cannot open mixer: No such file or directory
> 
> where did "_**bluelalsa**_" come from?

In fact, It's works better using `bluealsa` ! Now, I'm able to play a sound using aplay -D bluealsa:DEV=(mac address). But It's only working using this command. When I want to play a song through my multiroom solution (the project mentionned upper in the thread), It doesn't work. Pulseaudio is started to achieve this but pulseaudio-module-bluetooth is not installed. How can I make the audio bluetooth speaker works globally with any sound played by the RPI ?

borine, 2020-04-11:
so, we're back to choosing either pulseaudio or bluealsa to provide bluetooth audio, but not both. You said why you need pulseaudio, but never said why you need bluealsa. 

If you don't need any specific features provided by bluealsa, then uninstall bluealsa, re-enable the pulseaudio bluetooth modules and use that.

If you are sure you need bluealsa, then try the method described on the wiki page. That page assumes that pulseaudio is running in a desktop gui session, which I guess is not true for you. So you have a lot of work to do to (for example, if you use the example bash script, it has to run in the same user session as pulseaudio. The script is untested as a system service and I guess very risky to run as root from a security perspective, so if you are running pulseaudio as root then you may have quite some difficulty in setting it all up securely. Then you have to learn how to control pulseaudio without a desktop. The script just loads the bluealsa devices as sinks for pulseaudio - you will have to work out how to tell pulseaudio which sink to use. None of this is trivial. Are you quite sure you need bluealsa _and_ pulseaudio together given the amount of effort that will require?

closed 2021-06-21 by borine