arkq/bluez-alsa #319: No audio on A2DP profile despite data being sent

Amperaa, 2020-04-12:
Hello!

I'm trying to get my aptX capable bluetooth headphones working here. It's confirmed to work with aptX both by the manufacturer (Sennheiser) and via using [https://github.com/EHfive/pulseaudio-modules-bt](url) to get these headphones working with PulseAudio, so I know it works on this computer, with my version of Linux, on PulseAudio, but I need direct ALSA support.

I am running Arch Linux, I tend to keep most packages as recent as possible, and bluealsa was installed via a pkgbuild for it up on the AUR that I have modified to enabled aptX (and for this issue debugging). I have also substituted asound.conf and systemd service files for modified examples I've found here. The code compiled is, as of this time of writing, cloned directly from /this/ repository, and is most recent. I do also have openaptx installed from the AUR, but this issue is still a problem on SBC anyways.

I'm capable of connecting my headphones, and I can see them in alsamixer under the blueaudio device, and even adjust their volume. If I launch bluealsa using the hfp-ag or hsp-ag profiles, I am able to get audio via playing a PCM16LE wav file in aplay, but if I try to use the a2dp-source profile, with aptX or SBC (SBC utilized via recompiling bluealsa without aptX enabled) I will hear a faint pop when the file starts playing, and another when I stop it, but nothing more. I did make sure the volume was up in alsamixer, but no amount of changing it up or down did anything.

blueman-applet (my GUI for connecting bluetooth devices) shows that data is being sent to the headphones at rates appropriate for audio, but it's almost as if bogus data is being sent somehow that my headphones are unable to decode. If there's some information I can provide via some tool or something that would help get an idea of what's going on, just let me know.

FYI, as I know this might be a common issue PulseAudio is most certainly not running while I'm testing this, and there are no other audio servers or programs that might be blocking aplay from playing audio.

I'm also going to post some pertinent configuration and log files here as they might help shed light as to what my problem is:

**/usr/lib/systemd/system/bluealsa.service** (yes I did daemon-reload when modifying)

```
[Unit]
Description=Bluealsa daemon
Requires=dbus-org.bluez.service
After=dbus-org.bluez.service

[Service]
Type=dbus
BusName=org.bluealsa
EnvironmentFile=/etc/conf.d/bluealsa
ExecStart=/usr/bin/bluealsa -p a2dp-source -i hci0
Restart=on-failure
ProtectSystem=strict
ProtectHome=true
PrivateTmp=true
PrivateDevices=true
RemoveIPC=true
RestrictAddressFamilies=AF_UNIX AF_BLUETOOTH

[Install]
WantedBy=bluetooth.target
```
**/etc/asound.rc** (I rebooted whenever modifying just to be safe)
```
# Use PulseAudio by default
pcm.!default {
  type pulse
  fallback "sysdefault"
  hint {
    show on
    description "Default ALSA Output (currently PulseAudio Sound Server)"
  }
}

ctl.!default {
  type pulse
  fallback "sysdefault"
}

# vim:set ft=alsaconf:

defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "00:1B:66:AC:72:C5"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 20000
defaults.bluealsa.battery "yes"

ctl.bluealsa {
        @args [ SRV BAT ]
        @args.SRV {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.service
                }
        }
        @args.BAT {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.battery
                }
        }
        type bluealsa
        service $SRV
        battery $BAT
}

pcm.bluealsa {
        @args [ SRV DEV PROFILE DELAY ]
        @args.SRV {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.service
                }
        }
        @args.DEV {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.device
                }
        }
        @args.PROFILE {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.profile
                }
        }
        @args.DELAY {
                type integer
                default {
                        @func refer
                        name defaults.bluealsa.delay
                }
        }
        type plug
        slave.pcm {
                type bluealsa
                service $SRV
                device $DEV
                profile $PROFILE
                delay $DELAY
        }
        hint {
                show {
                        @func refer
                        name defaults.namehint.extended
                }
                description "Bluetooth Audio Hub"
        }
}

```
**aplay command used and subsequent debug information** (yes the wav does have audio in it)
```
$ aplay -D bluealsa:SRV=org.bluealsa,DEV=00:1B:66:AC:72:C5,PROFILE=a2dp test.wav 
D: bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Setting constraints
Playing WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
D: bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Initializing HW
D: bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: FIFO buffer size: 4096
D: bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Initializing SW
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Initializing SW
D: bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Prepared
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Initializing SW
D: bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Starting
D: bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Starting IO loop: 9
^CAborted by signal Interrupt...
aplay: pcm_write:2061: write error: Interrupted system call
D: bluealsa-pcm.c:327: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Stopping
D: bluealsa-pcm.c:111: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: IO thread cleanup
D: bluealsa-pcm.c:407: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Freeing HW
D: bluealsa-pcm.c:357: /org/bluealsa/hci0/dev_00_1B_66_AC_72_C5/a2dpsrc/sink: Closing
```
**bluealsa output/log as seen by systemctl status bluealsa** (my hostname redacted)
```
bluealsa.service - Bluealsa daemon
     Loaded: loaded (/usr/lib/systemd/system/bluealsa.service; disabled; vendor preset: disabled)
     Active: active (running) since Sat 2020-04-11 22:40:01 EDT; 17s ago
   Main PID: 75149 (bluealsa)
      Tasks: 4 (limit: 77100)
     Memory: 1.4M
     CGroup: /system.slice/bluealsa.service
             └─75149 /usr/bin/bluealsa -p a2dp-source -i hci0

Apr 11 22:40:09 MyHostname bluealsa[75149]: /usr/bin/bluealsa: D: bluez.c:1407: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
Apr 11 22:40:09 MyHostname bluealsa[75149]: /usr/bin/bluealsa: D: bluez.c:1441: Signal: PropertiesChanged: org.bluez.MediaTransport1: State
Apr 11 22:40:09 MyHostname bluealsa[75149]: /usr/bin/bluealsa: D: ba-transport.c:701: State transition: 0 -> 2
Apr 11 22:40:09 MyHostname bluealsa[75149]: /usr/bin/bluealsa: D: ba-transport.c:975: Created new thread [ba-a2dp-aptx]: A2DP Source (aptX)
Apr 11 22:40:09 MyHostname bluealsa[75149]: WARNING! Initializing apt-X encoder stub library. This library does NOT
Apr 11 22:40:09 MyHostname bluealsa[75149]: perform any encoding at all. The sole reason for this library to exist,
Apr 11 22:40:09 MyHostname bluealsa[75149]: is to simplify integration with the proprietary apt-X encoder library
Apr 11 22:40:09 MyHostname bluealsa[75149]: for open-source projects.
Apr 11 22:40:09 MyHostname bluealsa[75149]: /usr/bin/bluealsa: D: a2dp.c:1536: Starting IO loop: A2DP Source (aptX)
Apr 11 22:40:09 MyHostname bluealsa[75149]: /usr/bin/bluealsa: D: ba-transport.c:701: State transition: 2 -> 2
```
Any help would be immensely appreciated, as PulseAudio drives me nuts at the best of times, adding bluetooth to the problem makes it way worse.

Amperaa, 2020-04-23:
Any help here?

duhow, 2021-01-11:
Your last logs specify the error:

> This library does NOT perform any encoding at all.

Arkq, 2021-01-11:
I forgot about this issue, but since I've got a ping, I can answer it now :)

Like @duhow has said:

> WARNING! Initializing apt-X encoder stub library. This library does NOT perform any encoding at all.

The openaptx library is only an API bridge for proprietary Qualcomm library (it allows to compile application with proper library API). So, it is possible to have apt-X support in open source project. However, right now it is possible to compile openaptx with FFMpeg as an encoding backend: https://github.com/Arkq/openaptx#configure-options By doing so you will be able to use bluez-alsa with open-source (FFMpeg-based) apt-X encoder.

closed 2021-01-11 by Arkq

Arkq, 2021-01-11:
Ooo, I can see right now that you've got no sound even with SBC codec.... Please verify this with current master branch, and if this issue still occurs, please reopen it. Also it might be helpful to collect hci dump from host. Run `tcpdump -i bluetooth0 -w /tmp/xxx.dump` BEFORE connecting headset. Play sound for few seconds (1-2 seconds) SBC, and apt-X, then post dump files here.