arkq/bluez-alsa #320: Disable auto connect from Windows

biva, 2020-04-12:
Hello,

I'm using bluez-alsa on Moode and I love your work, thank you so much, it's working so well :)) Sorry if it's the wrong place to ask this question :/

I'm experiencing something boring with my laptop under Windows 10: every time I switch it on, it automatically connects to my RPI3 bluetooth, and stops the music I'm playing with Moode. Is there a way to disable auto-connect for this specific device? Apparently, it's not possible on the Windows side (I'm not the only one to have this issue, with various bluetooth devices and Microsoft doesn't provide any solution, except unpairing the device: https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/disable-bluetooth-auto-connect/17e708eb-6f98-4d2d-bc2b-bdd3a5145b98)

I can't disable Bluetooth on my laptop, because I still want that my bluetooth mouse does autoconnect. Also, I want that my bluetooth headset autoconnects as well.

For the moment, I'm not experiencing the same kind of issue with my Android phone, but it might happen at some point I guess, as my phone is auto-connecting to my bluetooth headset.

borine, 2020-04-12:
bluetooth auto connection takes place when a device is set to "trust" another. So I assume your rpi "trusts" your Windows PC.

I'm not familiar with Moode, but I guess that the script it uses for pairing just blindly adds trust to every paired device. If there is no way from the GUI to change that behaviour it can be done using the command-line tool bluetoothctl.

I would recommend you raise this as an issue with the Moode developers.

biva, 2020-04-12:
Thanks! I also posted the issue here: http://moodeaudio.org/forum/showthread.php?tid=2356

So if I understand you well, the "trust" status is something different from "Paired" and from "Connected", right? What's the "technical" name for this "trust" status?

borine, 2020-04-12:
> So if I understand you well, the "trust" status is something different from "Paired" and from "Connected", right?

yes

> What's the "technical" name for this "trust" status

trust

> And if I set a device to "untrusted", how should I connect it?

When a paired device is trusted by the rpi, then connection requests from that device are accepted by the rpi; when it is not trusted then you have to initiate the connect request from the rpi.

> I just click on "Connect" manually in the bluetooth setting page of Windows 10?

I know nothing of windows - is that like an operating system or something?

biva, 2020-04-12:
Hahaha, thanks :) Yes, Windows is kind of an OS. I heard that the founder made some money out of it (and might have bought this same website where we chat now, Github).

Thanks, I'll go back to Moode devs and we'll try to figure this out!

borine, 2021-06-21:
Not a bluez-alsa issue.

closed 2021-06-21 by borine