arkq/bluez-alsa #322: device_bonding_failed status 14

delleceste, 2020-04-14:
Hello.
I am trying to send audio from my huawei smartphone to a PC running archlinux + bluez + alsa only (no pulseaudio).
I manually succeeded  ONCE (with bluetoothctl + systemctl start bluealsa + systemctl start aplay, where  /etc/systemd/system/aplay.service is like
----------------------
[Unit]
Description=BlueALSA aplay service
After=bluetooth.service
Requires=bluetooth.service
 
[Service]
ExecStart=/usr/bin/bluealsa-aplay 00:00:00:00:00:00
 
[Install]
WantedBy=multi-user.target
-----------------------

After that, I enabled bluetooth auto power on throug /etc/bluetooth/main.conf, systemctl enabled the two aforementioned services but NEVER been able to connect again.

journalctls says the following:

_Apr 14 11:26:45 arki bluetoothd[397]: src/adapter.c:remove_temp_devices() /org/bluez/hci0
Apr 14 11:28:39 arki bluetoothd[397]: src/adapter.c:connected_callback() hci0 device EC:8C:9A:B1:94:DC connected eir_len 21
Apr 14 11:28:44 arki bluetoothd[397]: src/adapter.c:dev_disconnected() Device EC:8C:9A:B1:94:DC disconnected, reason 3
Apr 14 11:28:44 arki bluetoothd[397]: src/adapter.c:adapter_remove_connection()
Apr 14 11:28:44 arki bluetoothd[397]: plugins/policy.c:disconnect_cb() reason 3
Apr 14 11:28:44 arki bluetoothd[397]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr EC:8C:9A:B1:94:DC type 0 status 0xe
Apr 14 11:28:44 arki bluetoothd[397]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Apr 14 11:28:44 arki bluetoothd[397]: src/device.c:device_bonding_failed() status 14
Apr 14 11:28:44 arki bluetoothd[397]: src/adapter.c:resume_discovery()_

Scrolling up the logs, I see also this

_Apr 14 10:51:16 arki bluetoothd[396]: profiles/audio/avrcp.c:avrcp_controller_server_probe() path /org/bluez/hci0
Apr 14 10:51:16 arki bluetoothd[396]: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 14 10:51:16 arki bluetoothd[396]: avrcp-controller: Protocol not supported (93)
Apr 14 10:51:16 arki bluetoothd[396]: profiles/audio/avrcp.c:avrcp_target_server_probe() path /org/bluez/hci0
Apr 14 10:51:16 arki bluetoothd[396]: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 14 10:51:16 arki bluetoothd[396]: audio-avrcp-target: Protocol not supported (93)
Apr 14 10:51:16 arki bluetoothd[396]: profiles/audio/a2dp.c:a2dp_sink_server_probe() path /org/bluez/hci0
Apr 14 10:51:16 arki bluetoothd[396]: profiles/audio/a2dp.c:a2dp_source_server_probe() path /org/bluez/hci0_

MORE INFO
bluez Version         : 5.54-1

KERNEL 5.6.3

[root@arki ~]# pacman -Q --info bluez-alsa-git
Name            : bluez-alsa-git
Version         : 2.1.0.r23.g4a4dcd8-1
Description     : Bluetooth Audio ALSA Backend
Architecture    : x86_64
URL             : https://github.com/Arkq/bluez-alsa
Depends On      : bluez-libs  glib2  libfdk-aac  sbc  alsa-lib

HARDWARE Intel nuc

Thanks for any support

delleceste, 2020-04-14:
Some more details:
1. I tried on another machine where the same arch installation and pulseaudio, it works.
2. On the Intel Nuc mentioned in the issue, there is no pulseaudio installation.
3. On the intel Nuc, I **disable** bluealsa and the "aplay" service, and I am not able to connect to send files either. The error message on the log is the same. In other words, it looks like without pulseaudio you cannot even make file transfer work.
1 b. On the machine with pulseaudio, if I kill pulseaudio and rename /usr/bin/pulseaudio to something else in order not to have it spawned, bluetooth fails with the same debug messages.

There's something weird.

delleceste, 2020-04-14:
The fix seems to be this:

Since by default all output profiles are enabled, I just specified -p a2dp-sink, and it works
# BLUEALSA_CONF="-p a2dp-source -p a2dp-sink -p hfp-hf -p hfp-ag -p hsp-hs -p hsp-ag"
BLUEALSA_CONF="-p a2dp-sink"

Thanks for any explanation

delleceste, 2020-04-14:
OK, I didn't solve actually... it just worked for a while, now I connect, but no audio can be heard

borine, 2021-06-23:
There have been many changes in bluez-alsa since this issue was raised. I am unable to reproduce this error, so am closing.

If this problem still exists with the latest bluez-alsa code, please re-open, and please also build bluez-alsa with debug enabled and post the debug logs from the bluealsa daemon.

closed 2021-06-23 by borine