arkq/bluez-alsa #323: double free problem

powerpun, 2020-04-14:
when disconnect bluetooth by switch the phone's bluetooth off or away from the valid range, aplay sometimes will crash.

*** Error in `aplay/bluealsa-aplay': free(): invalid next size (normal): 0xb6100470 ***
Aborted

borine, 2020-04-16:
what version of bluealsa are you using? if 1.4.0 or earlier, you could be hitting this bug: #210 

if so, the fix appears in release 2.0.0.

powerpun, 2020-04-17:
> what version of bluealsa are you using? if 1.4.0 or earlier, you could be hitting this bug: #210
> 
> if so, the fix appears in release 2.0.0.

no, I'm using the latest one. aplay.c is in aplay directory

borine, 2020-04-17:
a backtrace from gdb when bluealsa-aplay crashes would help to locate the cause of the error

powerpun, 2020-04-17:
> a backtrace from gdb when bluealsa-aplay crashes would help to locate the cause of the error

did not use bt to check. cause I'm busy modify the pipe output and making a mixer to mix shairport output with bt

powerpun, 2020-04-17:
found a very strange thing for data comes in
printf("m: done:%d-%d, leftBT=%d, leftAP=%d ", doneBT, doneAP, leftBT, leftAP);
					printf("%"PRId64" %"PRId64" \n", ttlBT, ttlAP);

m: done:0-1, leftBT=2960, leftAP=0 10416128 13354304

don't know why BT data comes in slower than airplay... 
that's why I try one day could not solve the problem cause BT single play ok, airplay single play ok but try to mix 2 signals not ok... cause BT data rate is not good enough

Arkq, 2020-07-05:
@powerpun please check the latest master branch. I think I've found the cause of occasional segfaults in bluealsa-aplay - https://github.com/Arkq/bluez-alsa/commit/f80cd489b50fdfbe032c4a29fa9de99c420a3f11 Also this bug has caused audio distortions - not initialized memory might have been written to PCM device.