arkq/bluez-alsa #324: Can no longer run bluealsa-aplay - Open PCM: No such file or directory

gearhead, 2020-04-19:
I was running this last year and got sidetracked and am back at it. I had this all working last year, but there must've been a change because the same commands do not work.
What I'm trying to do: Use an RPi as an a2dp sink. I have a sound card on my RPi and want to send BT audio to it.
If I connect then type:
`/usr/bin/bluealsa-aplay  00:00:00:00:00:00`
I get sound from the phone through the speakers! Yea. 
When I try to do this programmatically according to the bluealsa wiki, I get this:

```
# systemctl status bluealsa-aplay
● bluealsa-aplay.service - Bluealsa audio player
     Loaded: loaded (/etc/systemd/system/bluealsa-aplay.service; disabled; vendor preset: disabled)
     Active: active (running) since Sun 2020-04-19 10:58:53 CDT; 21s ago
       Docs: https://github.com/Arkq/bluez-alsa/
   Main PID: 4947 (bluealsa-aplay)
      Tasks: 2 (limit: 2287)
     Memory: 560.0K
     CGroup: /system.slice/bluealsa-aplay.service
             └─4947 /usr/bin/bluealsa-aplay 00:00:00:00:00:00

Apr 19 10:58:53 livingrune systemd[1]: Started Bluealsa audio player.
Apr 19 10:59:10 livingrune bluealsa-aplay[4947]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 10:59:10 livingrune bluealsa-aplay[4947]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
Apr 19 10:59:11 livingrune bluealsa-aplay[4947]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 10:59:11 livingrune bluealsa-aplay[4947]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
Apr 19 10:59:12 livingrune bluealsa-aplay[4947]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 10:59:12 livingrune bluealsa-aplay[4947]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
Apr 19 10:59:13 livingrune bluealsa-aplay[4947]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 10:59:13 livingrune bluealsa-aplay[4947]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
```
In /etc/default/bluealsa-aplay all I have is:
`BT_ADDR="00:00:00:00:00:00"`

I know this has to be  some change as to how to tell bluealsa where to send the audio but shouldn't it work this way? Regardless, how do I set it up so it works, again. 
My alsa setup is:
```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0 [HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
speex
    Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0
    Default Audio Device
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ Pro HiFi pcm512x-hifi-0
    Default Audio Device
usbstream:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus
    USB Stream Output
```
I note that alsa mentions PulseAudio sound server. Pulse is not installed nor running, though libpulse is installed. I think it was installed when alsa was installed. 

Previously, my bluealsa-aplay command was:
`/usr/bin/bluealsa-aplay %I -dhw:0,0 00:00:00:00:00:00`

but that doesn't work any more:
```
# /usr/bin/bluealsa-aplay %I -dhw:0,0 00:00:00:00:00:00
/usr/bin/bluealsa-aplay: invalid option -- 'd'
Try '/usr/bin/bluealsa-aplay --help' for more information.
```
How can I get this working again?

Arkq, 2020-04-19:
> `Try '/usr/bin/bluealsa-aplay --help' for more information.`

Have you tried `--help`? In master branch there is a change which breaks backward compatibility. As the message says, there is no `-d` option - right now it's called `-D` (I've changed this option to be the same as in ALSA's `aplay`).

For "stable" functionality, please do not use master branch, because it a development branch and non-backward-compatible changes might be introduced. This project (`bluez-alsa`) uses semantic versioning, so as long as the major version number is not bumped you should be fine with seamless upgrade.

gearhead, 2020-04-19:
I just ran across that and tried it again... Still no joy. The current settings are:
```
OPTIONS="-Dhw:0,0"
BT_ADDR="00:00:00:00:00:00"
```
The response is still:
```
# systemctl status bluealsa-aplay
● bluealsa-aplay.service - Bluealsa audio player
     Loaded: loaded (/etc/systemd/system/bluealsa-aplay.service; disabled; vendor preset: disabled)
     Active: active (running) since Sun 2020-04-19 11:37:05 CDT; 17s ago
       Docs: https://github.com/Arkq/bluez-alsa/
   Main PID: 5591 (bluealsa-aplay)
      Tasks: 2 (limit: 2287)
     Memory: 576.0K
     CGroup: /system.slice/bluealsa-aplay.service
             └─5591 /usr/bin/bluealsa-aplay -Dhw:0,0 00:00:00:00:00:00

Apr 19 11:37:05 livingrune systemd[1]: Started Bluealsa audio player.
Apr 19 11:37:20 livingrune bluealsa-aplay[5591]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 11:37:20 livingrune bluealsa-aplay[5591]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
Apr 19 11:37:21 livingrune bluealsa-aplay[5591]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 11:37:21 livingrune bluealsa-aplay[5591]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
Apr 19 11:37:22 livingrune bluealsa-aplay[5591]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 11:37:22 livingrune bluealsa-aplay[5591]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
```
I am using the git version. Maybe I need to downgrade, but still what are the settings?

gearhead, 2020-04-19:
So, I tried building the latest v-2.1.0.tar.gz and get the same thing, basically.
```
 /etc/default/bluealsa 
OPTIONS="--profile a2dp-sink"
BT_ADDR="00:00:00:00:00:00"
/etc/default/bluealsa-aplay
OPTIONS="-dhw:0,0"
BT_ADDR="00:00:00:00:00:00"
```
The .service files are what is shown in the wiki section of this repository

bluealsa-aplay loads, but when I try to start a stream over BT, I get the error I saw previously
```
Apr 19 12:08:55 livingrune bluealsa-aplay[6172]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 12:08:55 livingrune bluealsa-aplay[6172]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
```
It is looking for a PCM and cannot find it. 
```
# systemctl status bluealsa-aplay
● bluealsa-aplay.service - Bluealsa audio player
     Loaded: loaded (/etc/systemd/system/bluealsa-aplay.service; disabled; vendor preset: disabled)
     Active: active (running) since Sun 2020-04-19 12:14:13 CDT; 17s ago
       Docs: https://github.com/Arkq/bluez-alsa/
   Main PID: 6318 (bluealsa-aplay)
      Tasks: 2 (limit: 2287)
     Memory: 688.0K
     CGroup: /system.slice/bluealsa-aplay.service
             └─6318 /usr/bin/bluealsa-aplay -dhw:0,0 00:00:00:00:00:00

Apr 19 12:14:27 livingrune bluealsa-aplay[6318]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 12:14:27 livingrune bluealsa-aplay[6318]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
Apr 19 12:14:28 livingrune bluealsa-aplay[6318]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 12:14:28 livingrune bluealsa-aplay[6318]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
Apr 19 12:14:29 livingrune bluealsa-aplay[6318]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 12:14:29 livingrune bluealsa-aplay[6318]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
Apr 19 12:14:30 livingrune bluealsa-aplay[6318]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 12:14:30 livingrune bluealsa-aplay[6318]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
Apr 19 12:14:31 livingrune bluealsa-aplay[6318]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 12:14:31 livingrune bluealsa-aplay[6318]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
```

gearhead, 2020-04-19:
So, in summary, BlueAlsa works fine. I pair and connect and start some file playing on my phone when I run it from the cli as:
`/usr/bin/bluealsa-aplay 00:00:00:00:00:00`
All is as expected. Then I try to have it 'auto start' by using the bluealsa-aplay.service file from the wiki, I get the error:
```
Apr 19 15:12:56 livingrune bluealsa-aplay[8743]: ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
Apr 19 15:12:56 livingrune bluealsa-aplay[8743]: /usr/bin/bluealsa-aplay: W: Couldn't open PCM: No such file or directory
```
This is running it as 'root' from the cli.
I am at a loss as to what it is asking for.  This worked back in spring of 2019. I can try to build older versions of bluealsa to see if it starts working, but feel that this should 'just work' as this is not really a corner case, is it?

Arkq, 2020-04-19:
> ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card

If `-D` option is not specified, bluealsa-aplay uses `default` (similar to `aplay`). Does `aplay somefile.wav` work for you (without `-D` option) for the same user?

borine, 2020-04-20:
> Invalid value for card

That error is caused by a mistake in the wiki guide. In the file `bluealsa-aplay.service` delete the line

`PrivateDevices=true`

That line hides all devices from the bluealsa-aplay process, including all sound cards. D'oh !!
I'll update the wiki very soon. My apologies for the time this has wasted.

> Previously, my bluealsa-aplay command was:
/usr/bin/bluealsa-aplay %I -dhw:0,0 00:00:00:00:00:00

`%I` implies that previously you were launching bluealsa-aplay using udev - in which case systemd expands `%I` to `hci0`. The above command still looks wrong though (old versions of bluealsa used `-i hci0` but you are missing `-i `). For newer bluealsa that whole syntax is now invalid so do not use %I anymore.

gearhead, 2020-04-20:
Woo Hoo! That works, Thanks for the help! I am running Bluez-alsa 2.1.0. With that change to toe bluealsa-aplay.service file and my /etc/default/bluealsa-aplay file looks like this:
```
~# cat /etc/default/bluealsa-aplay
OPTIONS="-dhw:0,0"
BT_ADDR="00:00:00:00:00:00"
```
If I am paired and connect and start playing, sound comes out the speakers. Seems to work as expected. Thanks for the hard work!

borine, 2020-04-21:
just for completeness, two comments on your /etc/default/bluealsa-aplay file:

- as you have already learned, the `-d` parameter will change to `-D` in the next major release. You can instead use the long form:
```--pcm=hw:0,0```
which will work in all versions of bluealsa-aplay
- the bluealsa-aplay systemd unit file already sets `BT_ADDR="00:00:00:00:00:00"` as a default, so there is no need to include this in the environment file unless you want to override it with something else. Does no harm to include it though if you prefer.

and one comment on your /etc/default/bluealsa file:
- the bluealsa systemd unit file does not use the environment variable `BT_ADDR` so there is no point setting it in the environment file.

graham8, 2020-04-21:
> the bluealsa-aplay systemd unit file already sets BT_ADDR="00:00:00:00:00:00" as a default

Can I suggest this default is baked into bluealsa-aplay itself?  Save us some typing ;)

borine, 2021-06-04:
> Can I suggest this default is baked into bluealsa-aplay itself

done in release 3.0.0

closed 2021-06-04 by borine