arkq/bluez-alsa #326: Using BlueALSA with alsa_out/alsa_in to bridge BT audio to JACK

unfa, 2020-04-23:
Is it possible to use BlueALSA to bridge Bluetooth audio devices to the JACK Audio Connection Kit?

The alsa_out / alsa_in programs should be able to do it, but I wasn't able to get it to work.

Here's results:

```
$ alsa_out -d bluealsa
D: ../../../src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Setting constraints
WARNING: buffer size does not match: (requested 2048, got 9600)
WARNING: period size does not match: (requested 1024, got 1066)
D: ../../../src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: FIFO buffer size: 4096
D: ../../../src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Selected HW buffer: 9 periods x 4264 bytes <= 38400 bytes
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_00_0A_45_0E_90_F7/a2dpsrc/sink: Initializing SW
*** Segmentation fault
Register dump:

 RAX: 000055593964e768   RBX: 00007f05bc086740   RCX: 000000000000026d
 RDX: 00007ffea9e04000   RSI: 0000000000000004   RDI: 0000000000000008
 RBP: 0000000000000000   R8 : 000000007fffffff   R9 : 00005559396286d0
 R10: 00007f05bc086740   R11: 00007ffea9dfd310   R12: 0000000000001000
 R13: 00007f05bc0866e0   R14: 000055593964a6b0   R15: 00007ffea9dfd260
 RSP: 00007ffea9dfd1a8

 RIP: 00007f05bc08674b   EFLAGS: 00010213

 CS: 0033   FS: 0000   GS: 0000

 Trap: 0000000e   Error: 00000004   OldMask: 00001000   CR2: a9e04000

 FPUCW: 0000037f   FPUSW: 00000000   TAG: 00000000
 RIP: 00000000   RDP: 00000000

 ST(0) 0000 0000000000000000   ST(1) 0000 0000000000000000
 ST(2) 0000 0000000000000000   ST(3) 0000 0000000000000000
 ST(4) 0000 0000000000000000   ST(5) 0000 0000000000000000
 ST(6) 0000 0000000000000000   ST(7) 0000 0000000000000000
 mxcsr: 1fa3
 XMM0:  0000000000000000000000003f800000 XMM1:  0000000000000000000000003f800000
 XMM2:  0000000000000000000000003f800000 XMM3:  0000000000000000000000003f800000
 XMM4:  0000000000000000000000003f800000 XMM5:  0000000000000000000000003f800000
 XMM6:  0000000000000000000000003f800000 XMM7:  0000000000000000000000003f800000
 XMM8:  0000000000000000000000003f800000 XMM9:  0000000000000000000000003f800000
 XMM10: 0000000000000000000000003f800000 XMM11: 0000000000000000000000003f800000
 XMM12: 0000000000000000000000003f800000 XMM13: 0000000000000000000000003f800000
 XMM14: 0000000000000000000000003f800000 XMM15: 0000000000000000000000003f800000

Backtrace:
/usr/lib/libasound.so.2(snd_pcm_lfloat_convert_float_integer+0x3fb)[0x7f05bc08674b]
/usr/lib/libasound.so.2(+0x79f12)[0x7f05bc085f12]
/usr/lib/libasound.so.2(+0x5d472)[0x7f05bc069472]
/usr/lib/libasound.so.2(+0x50bc4)[0x7f05bc05cbc4]
/usr/lib/libasound.so.2(+0x5d974)[0x7f05bc069974]
alsa_out(+0x370e)[0x5559381cc70e]
alsa_out(+0x2339)[0x5559381cb339]
/usr/lib/libc.so.6(__libc_start_main+0xf3)[0x7f05bbbb9153]
alsa_out(+0x29ae)[0x5559381cb9ae]

Memory map:

5559381c9000-5559381cb000 r--p 00000000 00:18 4925057                    /usr/bin/alsa_out
5559381cb000-5559381cf000 r-xp 00002000 00:18 4925057                    /usr/bin/alsa_out
5559381cf000-5559381d1000 r--p 00006000 00:18 4925057                    /usr/bin/alsa_out
5559381d1000-5559381d2000 r--p 00007000 00:18 4925057                    /usr/bin/alsa_out
5559381d2000-5559381d3000 rw-p 00008000 00:18 4925057                    /usr/bin/alsa_out
[... 122 lines cut ...]
7f05bc584000-7f05bc59a000 r--p 00034000 00:18 4925130                    /usr/lib/libjack.so.0.1.0
7f05bc59a000-7f05bc59d000 r--p 00049000 00:18 4925130                    /usr/lib/libjack.so.0.1.0
7f05bc59d000-7f05bc59e000 rw-p 0004c000 00:18 4925130                    /usr/lib/libjack.so.0.1.0
7f05bc59e000-7f05bc5a0000 rw-p 00000000 00:00 0 
7f05bc5a1000-7f05bc5a2000 rw-s 00000000 00:25 1111                       /dev/shm/jack_sem.1000_default_alsa_out
7f05bc5a2000-7f05bc5a3000 rw-s 00000000 00:25 1196                       /dev/shm/jack-1000-3 (deleted)
7f05bc5a3000-7f05bc5a4000 rw-s 00000000 00:25 142                        /dev/shm/jack-1000-1
7f05bc5a4000-7f05bc5b6000 rw-s 00000000 00:25 21                         /dev/shm/jack-shm-registry
7f05bc5b6000-7f05bc5b7000 rw-s 00000000 00:25 582                        /dev/shm/jack_sem.1000_default_a2j
7f05bc5b7000-7f05bc5b8000 rw-s 00000000 00:25 499                        /dev/shm/jack_sem.1000_default_dbusapi
7f05bc5b8000-7f05bc5b9000 rw-s 00000000 00:25 482                        /dev/shm/jack_sem.1000_default_freewheel
7f05bc5b9000-7f05bc5f7000 rw-s 00000000 00:25 272                        /dev/shm/jack_db/__db.002
7f05bc5f7000-7f05bc5f9000 r--p 00000000 00:18 4888106                    /usr/lib/ld-2.30.so
7f05bc5f9000-7f05bc618000 r-xp 00002000 00:18 4888106                    /usr/lib/ld-2.30.so
7f05bc618000-7f05bc620000 r--p 00021000 00:18 4888106                    /usr/lib/ld-2.30.so
7f05bc620000-7f05bc621000 rw-s 00000000 00:25 387                        /dev/shm/jack_sem.1000_default_system
7f05bc621000-7f05bc622000 r--p 00029000 00:18 4888106                    /usr/lib/ld-2.30.so
7f05bc622000-7f05bc623000 rw-p 0002a000 00:18 4888106                    /usr/lib/ld-2.30.so
7f05bc623000-7f05bc624000 rw-p 00000000 00:00 0 
7ffea9de3000-7ffea9e04000 rw-p 00000000 00:00 0                          [stack]
7ffea9fd2000-7ffea9fd5000 r--p 00000000 00:00 0                          [vvar]
7ffea9fd5000-7ffea9fd6000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]
Segmentation fault (core dumped)
```

borine, 2020-04-24:
see my comment to a previous issue here : https://github.com/Arkq/bluez-alsa/issues/250#issuecomment-607094063

Arkq, 2021-01-16:
Duplicate of #250.

closed 2021-01-16 by Arkq