arkq/bluez-alsa #328: How to control speaker volume?

graham8, 2020-04-27:
I've been following https://github.com/Arkq/bluez-alsa/pull/321 with interest. Can someone clarify what this proposal affects? It appears to be all about a source sending **to** bluealsa-aplay and that source's controller adjusting the volume of the sink device.  Am I right @joerg-krause?

If there is also some effect when bluealsa is the source, what is that?  At the moment, if I pass --a2dp-volume I can't control the hw volume of my BT speaker at all.  A volume control shows up in amixer but has no effect.  If I omit --a2dp-volume then I do get a usable volume control for each of a2dp and sco.  These seem to be software controls, ie independent of the buttons on the speaker.  If 'native' means the speaker's hw volume then it's not working as it should.  Is that something PR321 is going to fix or do I have to look elsewhere for a fix?

joerg-krause, 2020-04-27:
bluez-alsa is scaling the volume in software, but does not use the ALSA mixer control interface. #321 is about adding the latter, but it does not work with all ALSA configurations yet. See: https://github.com/Arkq/bluez-alsa/pull/321#discussion_r408158149. I will rework #321, but it may take some time.

graham8, 2020-04-27:
Thanks, but I'm still not clear.  I'm sort of expecting to be able to use amixer to set the (hw) volume on a BT speaker or headphones and have a volume level which has been set using the buttons on the speaker/headphones feed that value back to ALSA.  I thought since there is the --a2dp-volume option that functionality was already implemented (otherwise what does that option do?)

If not and your work will make all that happen then that's great :D

Arkq, 2020-04-27:
@graham8 `--a2dp-volume` enables volume control forwarding to BlueZ, which should forward this to remote BT device. Unfortunately, that not happens always/sometimes, I don't know. On my local host I've got patched BlueZ version which "fixes" this problem: d567b84

Arkq, 2020-10-29:
With bluez-5.55 (it might be [this](https://github.com/bluez/bluez/commit/df7d3fa5002373f7f955219defc7d0bc794a5a6c) commit) there should be no need to patch bluez for volume to work (at least, the issue was fixed for my headset). Also, with current bluez-alsa master audio scaling should work in a more user friendly way. Please note, that you still need to add `--a2dp-volume` to command line options, though.

closed 2020-10-29 by Arkq