arkq/bluez-alsa #329: HIDP connection issues borking sound

macmpi, 2020-04-30:
Hi,
I'm building a small Pi audio player project relying on MPD/bluez-alsa, sending audio to bluetooth speaker. Player may be controlled by speaker keys or HID bluetooth remote.

I'm facing an issue where at some random times, some bluetooth errors have a side effect on audio-decoding, hence producing garbled output. Here is what is always the log at those times:
```
Apr 29 12:54:11 player daemon.err bluetoothd[2325]: Can't get HIDP connection info
Apr 29 12:54:15 player daemon.warn libsmbclient[2747]: alsa_output: Decoder is too slow; playing silence to avoid xrun
Apr 29 12:54:17 player daemon.err bluetoothd[2325]: connect error: Host is down (112)
Apr 29 12:54:41 player daemon.err bluetoothd[2325]: Can't get HIDP connection info
Apr 29 12:54:47 player daemon.err bluetoothd[2325]: connect error: Host is down (112)
```
Can't tell if its `bluez` or `bluez-alsa` or both acting together, and I'm happy to provide more detailed logs if directed to do so.
After some searches it seems many similar HIDP errors have started popping-out from kernel 4.19.x  Obviously blacklisting `hidp` driver solves the issue (but then I have no remote).
I'm on Alpine 3.11, kernel 5.4.27, bluez 5.54, bluez-alsa 2.1.0

Any hint appreciated (fix or workaround).
Thanks.

borine, 2020-05-03:
Which Pi model are you using? Are you using the on-board bluetooth adapter?

I'm not familiar with Alpine Linux, but from other posts I've seen here I know that some distributions struggle to correctly configure the UART interface to the BT adapter on Pis. So I would recommend looking into that first if you are using a Pi with on-board BT.

macmpi, 2020-05-03:
Thanks for feedback. I'm on a PiZeroW and yes, I did take much care in configuring UART properly for that device (it has HW flow control and decent speed).
In most cases audio and BT remote work just fine. Issue pop-up only when those HID error message pop-up. 

Could be related to [that](https://bugzilla.redhat.com/show_bug.cgi?id=1289863#c40), but I'm not capable to trace further without directions.
Maybe the stack upper in the foodchain gets locked by this problem?

borine, 2021-07-26:
As HIDP is a Bluez / kernel issue, there is nothing that bluealsa can do about this. I'm guessing that as there have been no further posts on this subject in over a year this is a rare occurrence. So I'm closing this issue now.

closed 2021-07-26 by borine