arkq/bluez-alsa #330: bluealsa starting before bluetooth

graham8, 2020-05-04:
We had some strange behaviour with other BT devices not connecting and found that bluealsa is sometimes starting before bluetoothd.

In fact if we disable bluealsa.service and bluealsa-aplay.service, something is still starting bluealsa.

Only if we delete the two module files does bluealsa not show up in journalctl.  Any clues what's happening here?

borine, 2020-05-05:
This is an interesting one. If you are using the latest systemd unit files from the wiki, then the behaviour you are seeing should not occur. The bluealsa.service file includes the line
```After=dbus-org.bluez.service``` which should enforce the ordering you require.

bluealsa itself does not need to be started after bluetooth, it will discover the bluetooth service via d-bus whenever it starts, so the After= directive is optional depending on how you want to start bluealsa (see below).

similarly, bluealsa-aplay can start before bluealsa. I put no ordering constraint there because I could see no circumstance that it would be necessary.

> In fact if we disable bluealsa.service and bluealsa-aplay.service, something is still starting bluealsa.

systemd will start bluealsa if any other service is configured to "Want" or "Require" or "BindTo" it. You can list all units that have such a dependency on bluealsa with:
```systemctl --list-dependencies --reverse bluealsa```

In addition, if you installed the example d-bus service file given in the wiki under "**start on demand by a D-Bus client**", then bluealsa will also be started by any d-bus client that sends a message to it - for example bluealsa-aplay and any application that tries to open a bluetooth audio device. If you do not want alsa clients to start bluealsa, simply remove the file `/usr/share/dbus-1/system-services/org.bluealsa.service` - it will still be started automatically by bluealsa-aplay, because the systemd dependency ensures that.

> Only if we delete the two module files does bluealsa not show up in journalctl. 

Disabling a service merely removes any dependencies listed in the [Install] section of its own unit file.  It does not prevent it being started either manually or by a dependency in some other unit file. To achieve that you need to use `systemctl mask bluealsa.service`, which can later be undone by `systemctl unmask bluealsa.service`

I suspect the effect you are seeing is something like:
- some alsa client tries to open a bluealsa device
- d-bus starts bluealsa by sending a request to systemd
- systemd sees that it requires to be started after bluetooth, so starts that first.
- bluetooth is configured as `Type=simple` so systemd immediately starts bluealsa, not waiting for bluetooth to become ready.
- bluealsa completes its startup before bluetooth, processes the client request, which fails because bluetooth is not ready yet.
- bluetooth completes its startup, but is too late !!

A possible solution would be to change bluetooth to `Type=dbus` so that systemd waits until bluetooth has registered its d-bus interface before starting bluealsa.

 Now that you have raised this as an issue, perhaps I need to add some more text to the D-Bus client section, warning that this can happen. I need to think how best to explain this.

Could you try out changing the bluetooth service to `Type=dbus` to see if that fixes this issue?
Thanks

graham8, 2020-05-05:
Thanks for that. We think we've found the problem and will let you know.

DBMandrake, 2020-05-05:
Hi,

Thanks for the detailed reply. I managed to figure out what the cause was shortly after @graham8 posted here.

>bluealsa itself does not need to be started after bluetooth, it will discover the bluetooth service via d-bus whenever it starts, so the After= directive is optional depending on how you want to start bluealsa (see below).

>similarly, bluealsa-aplay can start before bluealsa. I put no ordering constraint there because I could see no circumstance that it would be necessary.

On my system bluealsa.service was starting about 500ms before bluetooth.service, and bluealsa-aplay.service was starting in the middle of bluetooth.service startup. (as reported by systemd-analyze plot) I don't understand why, but this was causing some really nasty side effects on the entire bluetooth stack unrelated to audio.

For example I have a bluetooth keyboard which pairs (with bluetoothctl) without requiring a pin code and which auto-reconnects when it is turned back on - as you would expect. This has been working fine for me for months.

After installing bluez-alsa using the two services (and with the effective startup order above) my keyboard would no longer auto connect when turned on and in fact I couldn't even force a connection with bluetoothctl - the connection would fail. If I deleted and re-paired the device it would ask for a 6 digit pin during pairing - on a keyboard which has never needed a pin before. (?!)

Once paired it would work, but as soon as it was turned off and on again it would not reconnect and I would have to go through the whole delete and repair process. I eventually tracked this down to the service startup order, as if I delayed startup of bluealsa.service and bluealsa-aplay.service the problem didn't occur. I have no idea why the timing of the service startup would have such a detrimental effect on the entire bluetooth stack, perhaps you would have an idea. As well as keyboards I was having a lot of difficulty connecting to speakers.

As to why the startup ordering was wrong that turned out to be /usr/share/dbus-1/system-services/org.bluealsa.service as you suggest above - something during boot is opening the alsa modules, they presumably try to connect to the org.bluealsa dbus name, this then causes dbus to explicitly start the service, (shown in the system journal) and in doing so completely bypasses all the service dependencies in the systemd unit.

So I don't really see the point of this redundant method of startup when the systemd service is already configured to start with bluetooth.target. If there is no bluetooth.target (bluetooth disabled) there is no reason to start the service anyway, as you can't send bluetooth audio without a bluetooth stack. 😄 

Another minor problem I found is that the example bluealsa-aplay.service installs as a dependency for multi-user.target. This means this service will always start during bootup even if bluetooth is disabled on the system, this results in bluealsa.service and bluetooth.service starting as well through a dependency chain, all on a system where bluetooth is disabled.

I'd suggest that the Install section of bluealsa-aplay.service be changed to bluetooth.target, same as the bluealsa service. That way if bluetooth is disabled on the system none of bluetooth.service, bluealsa.service and bluealsa-aplay.service will start. But if bluetooth is enabled they will all start in the order bluetooth.service -> bluealsa.service -> bluealsa-aplay.service.

Anyway with these two changes (delete /usr/share/dbus-1/system-services/org.bluealsa.service and change the bluealsa-aplay.service install target to bluetooth.target) everything seems to be working well now. 😄 

Thanks for your rapid and detailed response.

borine, 2020-05-05:
>I don't understand why, but this was causing some really nasty side effects on the entire bluetooth stack unrelated to audio.

That is unexpected, perhaps some bug in the bluetoothd? I'll look into that sometime, too busy at the moment. Incidentally, are you using a Raspberry Pi with on-board BT ? If so, it takes ages for the Broadcom firmware to be uploaded to the adapter so there is a big delay in the boot sequence before the hci0 interface comes up. Maybe that might have an influence here if services are starting before the hci - just a speculation.

>  this then causes dbus to explicitly start the service, (shown in the system journal) and in doing so completely bypasses all the service dependencies in the systemd unit.

I don't understand that; the d-bus org.bluealsa.service file explicitly tells d-dus to use systemd to launch bluealsa. The file does not contain any path to the bluealsa executable, so d-bus cannot possibly start it without systemd. Systemd will then apply all directives, just as if `systemctl start` had been used.

> I'd suggest that the Install section of bluealsa-aplay.service be changed to bluetooth.target

I agree, will do.

graham8, 2020-05-05:
We''re well aware of the fw upload issue.  @DBMandrake has spent a lot of time on that recently for our platform (Vero 4k +).  I'll let him respond on the other points.

DBMandrake, 2020-05-05:
> That is unexpected, perhaps some bug in the bluetoothd? I'll look into that sometime, too busy at the moment. Incidentally, are you using a Raspberry Pi with on-board BT ? If so, it takes ages for [...]

No it's a Vero4K however it has a similar 5 second delay for firmware upload to the bluetooth chip.

> I don't understand that; the d-bus org.bluealsa.service file explicitly tells d-dus to use systemd to launch bluealsa. The file does not contain any path to the bluealsa executable, so d-bus cannot [...]

Your scepticism over this explanation lead me to retrace my steps carefully and verify which things of the multiple things I've been changing actually made a difference and it turns out that in this particular instance you're right - adding or removing this file didn't affect the outcome.

The real culprit when I recheck carefully turns out to be bluealsa-aplay.service being set to install for multi-user.target instead of bluetooth.target, and I think I've worked out why this causes it.

multi-user.target is available earlier in the boot sequence than bluetooth.target, (which is a dynamic target that is added by udev rules when active bluetooth hardware is detected or powers on) so enabling it using this target allows bluealsa-aplay.service to start sooner bluetooth.target.

It has After and Requires dependencies on bluealsa.service so triggers this service to start first. bluealsa.service has After and Requires dependencies on dbus-org.bluez.service which is an alias for bluetooth.service so on the surface it appears that this should force bluetooth.service to start first out of the three - but it doesn't.

However it turns out there is an exception for Requires, as documented in the systemd docs:

"Note that this dependency type does not imply that the other unit always has to be in active state when this unit is running. **Specifically: failing condition checks (such as ConditionPathExists=, ConditionPathIsSymbolicLink=, … — see below)** do not cause the start job of a unit with a Requires= dependency on it to fail. Also, some unit types may deactivate on their own (for example, a service process may decide to exit cleanly, or a device may be unplugged by the user), which is not propagated to units having a Requires= dependency. Use the BindsTo= dependency type together with After= to ensure that a unit may never be in active state without a specific other unit also in active state (see below)."

bluetooth.service has a condition "ConditionPathIsDirectory=/sys/class/bluetooth", and this /sys path probably doesn't exist until the bluetooth hardware is powered on (which doesn't happen until connman changes rfkill state) which means if an attempt to start bluetooth.service is made before that condition can be satisfied it fails to start, however bluealsa.service which is trying to cause it to start via it's Requires dependency will not fail!

Then later on when bluetooth.target is created bluetooth.service will attempt to start a second time and this time ConditionPathIsDirectory will succeed and the service will start. Hence you end up with bluealsa.service starting first, followed by bluealsa-aplay.service, and completely asyncronously bluetooth.service is being started by bluetooth.target after bluealsa.service, and that matches exactly what systemd-analyze shows.

So my particular case it is the install target of multi-user.target that is trying to cause bluealsa.service to start too soon triggering the issue, however the dbus system-services file could cause the exact same thing - if something very early in boot tried to access org.bluealsa dbus would try to start bluealsa.service before bluetooth.service could pass its ConditionPathIsDirectory and create the same issue.

So to avoid all possibility of this out of order service startup occuring it seems that the dbus system-services file must be removed, and bluealsa-aplay.service set to install for bluetooth.target.

If both of those are true, the only way any of these services can be activated is when bluetooth.target is created, and this will schedule all three services to start (since they all target this target) and they will then follow the ordering specified in the After dependencies.

Phew! 😄

borine, 2020-05-05:
I'm glad to see you have your bluetooth stack working as required now :relieved: 

Just to correct an error in my first comment, for the record, bluealsa *does* require the bluetooth daemon to be running first - otherwise it terminates with `GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown`. That's why the "Requires=" and "After=" directives are included in the wiki example systemd unit file.

I will try to improve the text around the bluealsa dbus service file to make it clearer that it is an alternative method of starting bluealsa if you do not want to enable the systemd service for any reason. Its worth noting that there may be other dbus service files that result in the automatic startup of bluetooth, (for example raspbian includes `/usr/share/dbus-1/system-services/org.bluez.service` that permits any bluetooth client to start the daemon on demand). So bluealsa is not the only risk in that respect. 

What still needs to be explained is why the bluetooth stack is not initializing properly, and I still strongly suspect that it is the UART setup that is at the centre of it, but I cannot justify that suspicion,I guess its just a consequence of the pain of trying to make sense of hciattach/btattach and the bluez kernel modules btbcm and hci_uart :smile:

closed 2020-05-06 by graham8