arkq/bluez-alsa #333: profile=sco: is simultaneous playback  and capture possible?

BobGraton, 2020-05-13:
I manage to capture and playback from bluealsa:SRV=org.bluealsa,DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco
separately,  with arecord and aplay respectively, but not simultaneously. Is this a known limitation?

graham8, 2020-05-14:
If I talk on my phone through BT I don't think I can hear what the other side is saying while I'm talking but obviously it can switch rapidly.

BobGraton, 2020-05-14:
Thanks! I will pay more attention during my next conversation with my GF :-)

If I **"playback" to the bluealsa device** what my USB card microphone captures (1),
I don't hear anything anymore **"captured" from the bluealsa device** and played back on my USB card speaker (2). To hear something from the "capture" from the bluealsa device (2), I have to kill the playback to the bluealsa device (1).
There is no error when running these commands simultaneously. I can capture and play from and to my USB card simultaneously, so I don't expect the problem coming from there.
Running only one command (1) or (2) at a time works. I ran each command in a separate shell.

(1) $ arecord -D dsnoop:CARD=S3,DEV=0 -c 2 -r 48000 -f S16_LE | aplay -D bluealsa:SRV=org.bluealsa,DEV=F8:E6:1A:B0:71:44,PROFILE=sco -

(2) $ arecord -D bluealsa:SRV=org.bluealsa,DEV=F8:E6:1A:B0:71:44,PROFILE=sco -c 1 -r 8000 -f S16_LE | aplay -D plughw:CARD=S3,DEV=0 -

$  bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: F8:E6:1A:B0:71:44 [Samsung Galaxy S7], trusted phone
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: F8:E6:1A:B0:71:44 [Samsung Galaxy S7], trusted phone
  A2DP (AAC): S16_LE 2 channels 44100 Hz
  SCO (CVSD): S16_LE 1 channel 8000 Hz

Do you suggest bluetooth does not support full duplex? Does something trigger the rapid switch you suggest?

graham8, 2020-05-14:
Dunno.  I'd have to Google it.

BobGraton, 2020-05-15:
graham8, thanks for your thoughts:
Looking rapidly at the code, it seems that duplex is supported by bluealsa when using SCO profile, (unless I'm wrong), so I guess that the limitation is on the phone side. Nevertheless, using SOX, I made a soft "VOX", like the radio amateurs say (Voice Activated Transmission), with the following lines:

#!/bin/bash
while :
do
        arecord -D dsnoop:CARD=S3,DEV=0 -c 2 -r 48000 -f S16_LE | sox -t raw -b 16 -e signed -c 2 -v 3 -r 48000 - -t raw - silence 1 0.3 1% 1 0.3 1% | aplay -c 2 -r 48000 -f S16_LE -D bluealsa:SRV=org.bluealsa,DEV=F8:E6:1A:B0:71:44,PROFILE=sco -
done

and it works like a charm! With sox, one can also add a band pass filter and tune amplification easily.

BobGraton, 2020-05-16:
Nevertheless, it seems it's still a limitation or a bug, not a "feature", that has been around:
SCO simultaneous aplay arecord not working #114 (not addressed yet, but closed)

borine, 2021-10-02:
This issue has been idle for over a year now. I've tested the latest BlueALSA code with HSP and HFP, in both AG and HF/HS roles, and am able to play and record simultaneously in all cases without any issues. So this looks like it was possibly a limitation of the device being tested, or an old bug in BlueALSA that has since been fixed. So I am closing the issue now. If anyone experiences this problem with the latest BlueALSA code please raise a new issue, giving sufficient information so that  others can try to reproduce it and hopefully debug it.

closed 2021-10-02 by borine