arkq/bluez-alsa #334: Occasional segmentation fault on rpi when bluetooth speaker disconnects

e-nando, 2020-05-14:
With the latest build (and bluez 5.54, kernel 4.19.66-v7+) I'm seeing occasional segmentation faults on an rpi when bluetooth speaker disconnects. Here's the output from journalctl...

[seg fault.txt](https://github.com/Arkq/bluez-alsa/files/4628635/seg.fault.txt)

Arkq, 2020-05-14:
It is definitely caused by some race condition during disconnection:
```
../../src/ba-transport.c:948: Closing PCM: 10
../../src/ba-device.c:131: Freeing device: F8:DF:15:E4:4D:92
../../src/bluez.c:1435: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
../../src/ba-transport.c:948: Closing PCM: 779318594
```

The second "Closing PCM" should not have been called, because memory was already freed during the first one "Closing PCM". So on second call there is a garbage 779318594.

I will look at this, but not right away. Right now, I'm in the middle of yet another "big" change in the bluez-alsa codebase. I think I will finish my current work in 1-3 weeks, so then I will look at this one, or maaaaybe current modifications will fix this issue (I'm working on threading improvements).

e-nando, 2020-05-14:
Arkadiusz,

Many thanks; once your current work is published I’ll recompile bluez-alsa anyway and see if it fixes the problem.

All the best,

Chris

Arkq, 2020-05-30:
@e-nando please, compile bluez-alsa from [dbus-threaded](https://github.com/Arkq/bluez-alsa/tree/dbus-threaded) branch and check whether this issue still occurs.

e-nando, 2020-05-30:
Will do; will find time over the weekend...

e-nando, 2020-06-05:
Have tested it since the weekend without any segmentation fault. Seems to connect BT quicker, and no issues seen to date!

Many thanks,

Chris

Arkq, 2020-06-05:
Great to hear that there was no segfault so far :) I will push dbus-threaded into the master then.

> Seems to connect BT quicker

It might be placebo, because I've made no changes which would result in a quicker BT connection :D

commit 86633a65 (2020-06-05): Keep transport ref in on-going PCM controller

It is required to keep reference to PCM structure (transport) until
PCM controller is removed from the mail loop context. Otherwise, we
might use already freed transport structure.

Fixes #334
closed 2020-06-05 by Arkq