arkq/bluez-alsa #335: Couldn't acquire D-Bus name: org.bluealsa from alpine container

tdesaules, 2020-05-15:
Hi there,

I have a classic dbus issue using bluealsa.
```
bash-5.0# bluealsa
bluealsa: E: Couldn't acquire D-Bus name: org.bluealsa
```
I take a look from the other case, think I'm right but would like to clarify

I run an alpine linux 3.11 container with alsa / bluez / bluealsa installed on it
This container live on top of BalenaOS on a Raspberry Pi 3 A+
(I have a topic on it here : https://forums.balena.io/t/bluetooth-configuration-with-alpine-linux/113733)
 
I think my bluetooth conf is good. From bluetoothctl I'm able to pair a device : 
```
bash-5.0# bluetoothctl 
Agent registered
[bluetooth]# show
Controller B8:27:EB:14:A6:97 (public)
        Name: 5fc958d
        Alias: chambre
        Class: 0x00000000
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: no
Advertising Features:
        ActiveInstances: 0x00
        SupportedInstances: 0x05
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[NEW] Device 6C:29:95:7C:B3:01 Chromebook_3036
[CHG] Device 6C:29:95:7C:B3:01 Connected: no
[CHG] Device 6C:29:95:7C:B3:01 Paired: yes
[bluetooth]# info 6C:29:95:7C:B3:01 
Device 6C:29:95:7C:B3:01 (public)
        Name: Chromebook_3036
        Alias: Chromebook_3036
        Class: 0x00480104
        Icon: computer
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
```
dbus daemon is runing on the host and from my container it is working I have a reply when I try this : 
```
bash-5.0# dbus-send --system --print-reply --reply-timeout=2000 --type=method_call --dest=org.freedesktop.timedate1 /org/freedesktop/timedate1 org.freedesktop.DBus.Properties.GetAll string:"org.freedesktop.timedate1"
method return time=1589529332.400402 sender=:1.280 -> destination=:1.279 serial=4 reply_serial=2
   array [
      dict entry(
         string "Timezone"
         variant             string ""
      )
      dict entry(
         string "LocalRTC"
         variant             boolean false
      )
      dict entry(
         string "CanNTP"
         variant             boolean false
      )
      dict entry(
         string "NTP"
         variant             boolean false
      )
      dict entry(
         string "NTPSynchronized"
         variant             boolean true
      )
      dict entry(
         string "TimeUSec"
         variant             uint64 1589529332399850
      )
      dict entry(
         string "RTCTimeUSec"
         variant             uint64 0
      )
   ]
```
But when I launch bluealsa I get the `Couldn't acquire D-Bus name: org.bluealsa` error
I think the `/etc/dbus-1/system.d/bluealsa.conf` should be in the host BalenaOS. 
It is not of course and the filesystem is lock so I can't write it there.

The strange part is when I try from another container (debian strech) it is working but the bluealsa version are different

alpine linux : 
```
bash-5.0# bluealsa --version
2.1.0
```
debian : 
```
root@5fc958d:/# bluealsa --version
1.3.0
```
I also try to grep some info from dbus-monitor on the BalenaOS host : 
```
root@5fc958d:~# dbus-monitor --system | grep org.bluealsa
   string "org.bluealsa"
   string "Connection ":1.343" is not allowed to own the service "org.bluealsa" due to security policies in the configuration file"
```
Look like obvious that I can't use org.bluealsa bus
But when I lauch bluealsa from the debian container (bluealsa 1.3.0) I don't have any issue on dbus-monitor. In fact I can't fing any log

Do you have any explanation ? maybe using another bus address ?
I don't know very well dbus. Can I use another one with bluealsa ?

borine, 2020-05-15:
There was a recent issue regarding use of bluealsa with Docker, which the user resolved. There may be something useful in the links that were provided there:
#313

tdesaules, 2020-05-15:
Hi !

I will take a look soon. I don't start to play with bluealsa-aplay yet

I confirm that org.bluealsa wasn't authorized from the BalenaOS host

From the Balena forum I figured out how to enable RW on the filesystem and add the `/etc/dbus-1/system.d/bluealsa.conf`

I'm able to start `/usr/bin/bluealsa -i hci0 -p a2dp-sink &`

Next step alsa :)

tdesaules, 2020-05-16:
finaly everything work perfectly :)

/usr/bin/bluealsa-aplay --profile-a2dp --pcm-buffer-time=1000000 00:00:00:00:00:00 &

sound perfectly ^^

closed 2020-05-16 by tdesaules