arkq/bluez-alsa #336: Couldn't get BlueALSA PCM list: Rejected send message

antizo, 2020-05-25:
$sudo bluealsa
bluealsa: D: ../../src/ofono.c:623: Checking oFono service presence
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:1154: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:1182: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:1154: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:1182: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/main.c:397: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:402: Starting main dispatching loop

$ amixer -D bluealsa
ALSA lib ../../../src/asound/bluealsa-ctl.c:901:(_snd_ctl_bluealsa_open) Couldn't get BlueALSA PCM list: Rejected send message, 1 matched rules; type="method_call", sender=":1.149" (uid=1000 pid=23258 comm="amixer -D bluealsa") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=0 pid=13069 comm="bluealsa")
amixer: Mixer attach bluealsa error: No such device

borine, 2020-05-25:
It looks like your `MI BT 18` device is not advertising support for the A2DP profile - to be sure, can you start bluealsa as above then post the output from:
```
echo -e 'show\ninfo 8C:5A:F8:29:63:08' | bluetoothctl
```
That should print the BT services supported by your computer, and the services provided by your BT device.

For A2DP playback, the Controller needs to have:
 `UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)` 

and the Device needs to have:
 `UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)`

antizo, 2020-05-25:
> echo -e 'show\ninfo 8C:5A:F8:29:63:08' | bluetoothctl

$ echo -e 'show\ninfo 8C:5A:F8:29:63:08' | bluetoothctl
Agent registered
[MI BT18]# show
Controller 00:22:68:EF:D2:EF (public)
        Name: localhost
        Alias: localhost
        Class: 0x001c010c
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0536
        Discovering: yes
[MI BT18]# info 8C:5A:F8:29:63:08
Device 8C:5A:F8:29:63:08 (public)
        Name: MI BT18
        Alias: MI BT18
        Class: 0x00340404
        Icon: audio-card
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: yes
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v05D6p000Ad0240
        RSSI: -55

antizo, 2020-05-25:
> It looks like your `MI BT 18` device is not advertising support for the A2DP profile - to be sure, can you start bluealsa as above then post the output from:
>
> ```
> echo -e 'show\ninfo [...]

$ echo -e 'show\ninfo 8C:5A:F8:29:63:08' | bluetoothctl
Agent registered
[MI BT18]# show
Controller 00:22:68:EF:D2:EF (public)
        Name: localhost
        Alias: localhost
        Class: 0x005c010c
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0536
        Discovering: yes
[MI BT18]# info 8C:5A:F8:29:63:08
Device 8C:5A:F8:29:63:08 (public)
        Name: MI BT18
        Alias: MI BT18
        Class: 0x00340404
        Icon: audio-card
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: yes
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v05D6p000Ad0240
        RSSI: -55

antizo, 2020-05-25:
$sudo bluealsa
bluealsa: D: ../../src/ofono.c:623: Checking oFono service presence
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:1154: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:1182: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:1154: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:1182: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/main.c:397: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:402: Starting main dispatching loop
bluealsa: D: ../../src/bluez.c:1121: Called: org.bluez.Profile1.NewConnection()
bluealsa: D: ../../src/ba-rfcomm.c:880: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-rfcomm.c:1262: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:168: Checking Broadcom internal SCO routing
bluealsa: D: ../../src/sco.c:174: Current SCO interface setup: 1 2 0 1 1
bluealsa: D: ../../src/sco.c:202: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: ../../src/sco.c:79: Starting SCO dispatcher loop: hci0
bluealsa: D: ../../src/bluez.c:1044: HFP Audio Gateway (CVSD) configured for device 8C:5A:F8:29:63:08
bluealsa: D: ../../src/ba-transport.c:701: State transition: 0 -> 2
bluealsa: D: ../../src/sco.c:243: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-transport.c:975: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+BRSF, value:031
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+BRSF, value:2272
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
bluealsa: D: ../../src/at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
bluealsa: D: ../../src/at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 5 -> 7
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CMER, value:3,0,0,1
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 7 -> 8
bluealsa: D: ../../src/ba-rfcomm.c:126: RFCOMM: HFP Audio Gateway (CVSD) state transition: 8 -> 9
bluealsa: D: ../../src/at.c:161: AT message: TEST: command:+CHLD, value:(null)
bluealsa: W: Unsupported AT message: TEST: command:+CHLD, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CMEE, value:1
bluealsa: W: Unsupported AT message: SET: command:+CMEE, value:1
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CLIP, value:1
bluealsa: W: Unsupported AT message: SET: command:+CLIP, value:1
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CCWA, value:1
bluealsa: W: Unsupported AT message: SET: command:+CCWA, value:1
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+NREC, value:0
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+VGS, value:15
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+VGM, value:15
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+XAPL, value:ABCD-1234-0100,10
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,6
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+IPHONEACCEV, value:1,1,9
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK

borine, 2020-05-25:
Controller `UUID: Audio Sink (0000110b-0000-1000-8000-00805f9b34fb)`

That looks suspicious - you have started bluealsa without a2dp sink support, but your controller is still advertising the service. That suggests to me that perhaps you have pulseaudio running with its own bluetooth modules enabled? Something is preventing bluealsa from using A2DP - pusleaudio is the most likely culprit.

antizo, 2020-05-25:
> pulseaudio

$ sudo pulseaudio --kill
E: [pulseaudio] main.c: Failed to kill daemon: No such process
$ echo -e 'show\ninfo 8C:5A:F8:29:63:08' | bluetoothctl
Agent registered
[MI BT18]# show
Controller 00:22:68:EF:D2:EF (public)
        Name: localhost
        Alias: localhost
        Class: 0x0058010c
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0536
        Discovering: yes
[MI BT18]# info 8C:5A:F8:29:63:08
Device 8C:5A:F8:29:63:08 (public)
        Name: MI BT18
        Alias: MI BT18
        Class: 0x00340404
        Icon: audio-card
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: yes
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v05D6p000Ad0240
        RSSI: -58

borine, 2020-05-25:
The  `Audio Sink` UUID has now disappeared from the Controller (hurrah!)
double-check that pulseaudio is not running with
```
ps aux | grep pulseaudio
```
then restart bluealsa, connect your device, and try aplay again.

antizo, 2020-05-25:
> The `Audio Sink` UUID has now disappeared from the Controller (hurrah!)
> double-check that pulseaudio is not running with
> 
> ```
> ps aux | grep pulseaudio
> ```
> 
> then restart bluealsa, connect your device, and try aplay again.

$ ps aux | grep pulseaudio
lee        10003  0.0  0.0  30384   632 pts/0    S+   21:59   0:00 grep --color pulseaudio

$ echo -e 'show\ninfo 8C:5A:F8:29:63:08' | bluetoothctl
Agent registered
[MI BT18]# show
Controller 00:22:68:EF:D2:EF (public)
        Name: localhost
        Alias: localhost
        Class: 0x0058010c
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x00000000
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0536
        Discovering: yes
[MI BT18]# info 8C:5A:F8:29:63:08
Device 8C:5A:F8:29:63:08 (public)
        Name: MI BT18
        Alias: MI BT18
        Class: 0x00340404
        Icon: audio-card
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: yes
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v05D6p000Ad0240
        RSSI: -44

$ sudo bluealsa
bluealsa: D: ../../src/ofono.c:623: Checking oFono service presence
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/1
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Source/2
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
bluealsa: D: ../../src/bluez.c:1154: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:1182: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:1154: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
[... 41 lines cut ...]
bluealsa: D: ../../src/bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:796: Called: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: D: ../../src/bluez.c:565: Configuration: Selected A2DP SBC bit-pool range: [2, 53]
bluealsa: D: ../../src/bluez.c:715: A2DP Source (SBC) configured for device 8C:5A:F8:29:63:08
bluealsa: D: ../../src/bluez.c:718: Configuration: channels: 2, sampling: 48000
bluealsa: D: ../../src/ba-transport.c:701: State transition: 0 -> 0
bluealsa: D: ../../src/bluez.c:842: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/3
bluealsa: D: ../../src/bluez.c:871: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/3
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CLIP, value:1
bluealsa: W: Unsupported AT message: SET: command:+CLIP, value:1
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+CCWA, value:1
bluealsa: W: Unsupported AT message: SET: command:+CCWA, value:1
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+NREC, value:0
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+VGS, value:15
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+VGM, value:15
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+XAPL, value:ABCD-1234-0100,10
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,6
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/at.c:161: AT message: SET: command:+IPHONEACCEV, value:1,1,9
bluealsa: D: ../../src/ba-rfcomm.c:107: Sending AT message: RESP: command:(null), value:OK

borine, 2020-05-25:
> ... and try aplay again.

No output from aplay ?

antizo, 2020-05-25:
> > ... and try aplay again.
> 
> No output from aplay ?

$ sudo aplay -D bluealsa:SRV=org.bluealsa,DEV=8C:5A:F8:29:63:08,PROFILE=a2dp 2-20.wav 
D: ../../../src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Setting constraints
Playing WAVE '2-20.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
D: ../../../src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: FIFO buffer size: 4096
D: ../../../src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Selected HW buffer: 6 periods x 16380 bytes <= 98292 bytes
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Starting
D: ../../../src/asound/bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: Starting IO loop: 8
D: ../../../src/asound/bluealsa-pcm.c:111: /org/bluealsa/hci0/dev_8C_5A_F8_29_63_08/a2dpsrc/sink: IO thread cleanup
aplay: pcm_write:2061: write error: No such device

borine, 2020-05-25:
An A2DP connection is successfully established, playback begins, then it appears the device disconnects. There is no evidence in the logs to suggest why. Perhaps its related to the HFP link that is also created. You could test that by running bluealsa as:
```
sudo bluealsa -p a2dp-source
```
which would disable the HFP gateway. If that still fails, then I am out of ideas, I'm sorry to say.

borine, 2021-06-21:
This thread appears to have gone stale. Please feel free to re-open if any new information to help resolve it becomes available.

closed 2021-06-21 by borine