arkq/bluez-alsa #337: Failure: Module initialization failed

tyzbit, 2020-05-26:
This is almost a duplicate of https://github.com/Arkq/bluez-alsa/issues/312 as I have the same symptoms and behavior, but trying to load the module fails for me instead of succeeds:

```
sample|master⚡ ⇒  pactl load-module module-alsa-sink device=bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=a2dp fragments=1 fragment_size=960 sink_name=btheadset sink_properties=device.description=sennheiser
Failure: Module initialization failed
```
I've tried the a2dp profile, the sco profile and the hfp-ag profile (the one I actually want to use)
I see my device in DBus
```
sample|master⚡ ⇒  sudo gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
({objectpath '/org/bluealsa/hci0/dev_00_16_94_47_B6_F4/hfpag/sink': {'Device': <objectpath '/org/bluez/hci0/dev_00_16_94_47_B6_F4'>, 'Transport': <'HFP-AG'>, 'Mode': <'sink'>, 'Format': <uint16 32784>, 'Channels': <byte 0x01>, 'Sampling': <uint32 16000>, 'Codec': <'mSBC'>, 'Delay': <uint16 10>, 'SoftVolume': <true>, 'Volume': <uint16 3840>}, '/org/bluealsa/hci0/dev_00_16_94_47_B6_F4/hfpag/source': {'Device': <objectpath '/org/bluez/hci0/dev_00_16_94_47_B6_F4'>, 'Transport': <'HFP-AG'>, 'Mode': <'source'>, 'Format': <uint16 32784>, 'Channels': <byte 0x01>, 'Sampling': <uint32 16000>, 'Codec': <'mSBC'>, 'Delay': <uint16 10>, 'SoftVolume': <true>, 'Volume': <uint16 3840>}},)
```
I'm on commit 49ad348, and I'm running bluealsa manually specifying profiles.
```
sudo bluealsa -p a2dp-source -p hfp-hf -p hfp-ag
```
I don't have the pulseaudio bluetooth module even installed.

borine, 2020-05-26:
Hi @tyzbit 
The pulseaudio integration is very experimental, and has had very little testing, so may suffer many faults. If you have time to help faultfinding , that would be much appreciated :smile: 

Please can you try running pulseaudio with the --verbose flag set, hopefully the output will then give some clue as to why the alsa-sink module initialization is failing.

tyzbit, 2020-05-26:
Here's what pulseaudio says in the logs:
```
May 26 14:04:06 sen pulseaudio[4882]: E: [pulseaudio] ltdl-bind-now.c: Failed to open module --verbose.so: --verbose.so: cannot open shared object file: No such file or directory
May 26 14:04:06 sen pulseaudio[4882]: E: [pulseaudio] module.c: Failed to open module "--verbose".
```

borine, 2020-05-26:
>` I: [pulseaudio] alsa-util.c: Error opening PCM device bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=a2dp: No such device`

The above tells me that the a2dp profile is not connected for your headphones

> `I: [pulseaudio] (alsa-lib)bluealsa-pcm.c: Invalid BT profile [a2dp, sco]: hfp-ag
I: [pulseaudio] alsa-util.c: Error opening PCM device bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=hfp-ag: Invalid argument`

That is because bluealsa accepts only a2dp or sco as values for the `PROFILE=` argument. To use hfp you must select `PROFILE=sco`

To keep things simple until you get the basics working, I recommend you start bluealsa as:
```
sudo bluealsa -p hfp-ag
```
That will ensure that only hfp profile can be used, and prevent any issues with a2dp confusing things for now.

Then try loading the pulseaudio alsa-sink module with:
```
pactl load-module module-alsa-sink device=bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=sco fragments=1 fragment_size=960 sink_name=btheadset sink_properties=device.description=sennheiser
```
Please report back with the result, and any error from pulseaudio if it fails. Once that is working, we can try adding a2dp if required.

tyzbit, 2020-05-26:
I might be one of the few people that wants HFP since I'm trying to use my headset as a communications device.

Here's the output of trying to load the pulseaudio alsa-sink module:
```
I: [pulseaudio] client.c: Created 33 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Got credentials: uid=1001 gid=1001 success=1
I: [pulseaudio] (alsa-lib)bluealsa-pcm.c: Couldn't get BlueALSA PCM: Rejected send message, 1 matched rules; type="method_call", sender=":1.203" (uid=1001 pid=6579 comm="pulseaudio -v " label="unconfined") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=0 pid=11458 comm="bluealsa -p hfp-ag " label="unconfined")
I: [pulseaudio] alsa-util.c: Error opening PCM device bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=sco: No such device
E: [pulseaudio] module.c: Failed to load module "module-alsa-sink" (argument: "device=bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=sco fragments=1 fragment_size=960 sink_name=btheadset sink_properties=device.description=sennheiser"): initialization failed.
I: [pulseaudio] client.c: Freed 33 "pactl"
I: [pulseaudio] protocol-native.c: Connection died.
```
Rerunning the DBus command, I see the headset present.

borine, 2020-05-26:
> `[pulseaudio] alsa-util.c: Error opening PCM device bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=sco: No such device`

that suggests the headset is not connected (at least not for hfp), but bluez (via dbus) says that it is. Maybe its actually the `bluealsa_raw` part that is missing. Where have you installed the `21-bluealsa-raw.conf` file, and what version of alsa are you using (`aplay --version`) ?

I think we need to make sure that bluealsa is configured correctly, and is working, before trying to use it through pulseaudio.

tyzbit, 2020-05-26:
> Where have you installed the 21-bluealsa-raw.conf file

/usr/share/alsa/alsa.conf.d/21-bluealsa-raw.conf

> what version of alsa are you using (aplay --version) 
```
~|⇒  aplay --version
aplay: version 1.1.3 by Jaroslav Kysela <perex@perex.cz>
```

tyzbit, 2020-05-27:
I was able to make some progress.  If I run bluealsa with no profiles specified, this command now works (I have to use `sudo`):
```
~|⇒  sudo aplay -D bluealsa:SRV=org.bluealsa,DEV=00:16:94:47:B6:F4,PROFILE=sco english.wav
Playing WAVE 'english.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Mono
```

tyzbit, 2020-05-27:
I have made even more progress - I added my user to the `audio` group and rebooted and now the module is loading.  I will report back when I have everything actually working.

tyzbit, 2020-05-27:
I was able to get the device working like this:
```
~|⇒  MODULE=$(pactl load-module module-alsa-sink device=bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=sco fragments=1 fragment_size=1500 sink_name=btheadset sink_properties="device.description='Sennheiser'device.icon_name='audio-headset-bluetooth'")
~|⇒  SOURCEMODULE=$(pactl load-module module-alsa-source device=bluealsa_raw:DEV=00:16:94:47:B6:F4,PROFILE=sco fragments=1 fragment_size=1500 source_properties="device.description='Sennheiser'device.icon_name='audio-headset-bluetooth'")
```
I do notice that after a dozen seconds or more, the audio stream gets choppy.  This happens in all applications but is very noticeable in Discord.  Should I open up a new issue?

borine, 2020-05-27:
Do you get the "choppy" playback when using
 `aplay -D bluealsa:DEV=00:16:94:47:B6:F4,PROFILE=sco  english.wav` ?

If so, then pulseaudio will only make things worse.

If the above works correctly, then try different values for the fragment_size pulseaudio module parameter. For use with hfp, try a low value such as 80, then increase it until you find the best compromise between cpu load and audio stability. a fragment size of 80 for hfp works out at 5ms of audio, which is the value which has worked best for me.

tyzbit, 2020-05-27:
Using `aplay`, the audio is stable.  I have tried a number of different fragment combinations and can get it to work for about 90 seconds at the longest and then it breaks up.  My best setting was 1 fragment, 154 fragment size.

tyzbit, 2020-05-27:
Doing some more testing, I found different applications trigger the choppy playback differently.  Discord triggers it earliest and it's the worst, but playing audio through Firefox allows me to go minutes without an issue and when the issue does occur, it's less severe.

To explain, when the audio gets choppy:

### Discord

Audio will pause, then play back for 30-80ms, then pause again for longer, then play the next 30-80ms, then pause for even longer, then usually skip a few seconds and play a semi-recent slice of 30-80ms.  During this time, although my system's CPU is fine, it becomes less responsive and stutters

### Firefox
Audio will briefly stutter for 10ms or so, then play again for 3-5 seconds, then another brief stutter for 10ms.

I've gotten over 3 minutes without any issues on Firefox with an fragment_size of 90.

I did notice the following in my syslog:
<details><summary>Expand</summary>
<pre>

## unloading existing pactl module
May 27 19:45:45 sen gsd-media-keys[2702]: Unable to get default sink
May 27 19:45:45 sen kernel: [78358.316997] Bluetooth: hci0: SCO packet for unknown connection handle 284
May 27 19:45:45 sen kernel: [78358.317002] Bluetooth: hci0: SCO packet for unknown connection handle 284
May 27 19:45:45 sen kernel: [78358.317005] Bluetooth: hci0: SCO packet for unknown connection handle 284

## loading new pactl module
May 27 19:46:01 sen rtkit-daemon[1517]: Supervising 0 threads of 0 processes of 0 users.
May 27 19:46:01 sen rtkit-daemon[1517]: Successfully made thread 9804 of process 2338 (n/a) owned by '1001' RT at priority 5.
May 27 19:46:01 sen rtkit-daemon[1517]: Supervising 1 threads of 1 processes of 1 users.
May 27 19:46:01 sen kernel: [78374.225824] Bluetooth: hci0: SCO packet for unknown connection handle 0
May 27 19:46:01 sen kernel: [78374.225825] Bluetooth: hci0: SCO packet for unknown connection handle 0
May 27 19:46:01 sen kernel: [78374.235857] Bluetooth: hci0: SCO packet for unknown connection handle 0
May 27 19:46:01 sen kernel: [78374.235858] Bluetooth: hci0: SCO packet for unknown connection handle 0
</pre>
</details>

tyzbit, 2020-05-28:
Doing some more testing, I found different applications trigger the choppy playback differently.  Discord triggers it earliest and it's the worst, but playing audio through Firefox allows me to go minutes without an issue and when the issue does occur, it's less severe.

To explain, when the audio gets choppy:

### Discord

Audio will pause, then play back for 30-80ms, then pause again for longer, then play the next 30-80ms, then pause for even longer, then usually skip a few seconds and play a semi-recent slice of 30-80ms.  During this time, although my system's CPU is fine, it becomes less responsive and stutters

### Firefox
Audio will briefly stutter for 10ms or so, then play again for 3-5 seconds, then another brief stutter for 10ms.

I've gotten over 3 minutes without any issues on Firefox with an fragment_size of 90.

I did notice the following in my syslog:
<details><summary>Expand</summary>
<pre>

## unloading existing pactl module
May 27 19:45:45 sen gsd-media-keys[2702]: Unable to get default sink
May 27 19:45:45 sen kernel: [78358.316997] Bluetooth: hci0: SCO packet for unknown connection handle 284
May 27 19:45:45 sen kernel: [78358.317002] Bluetooth: hci0: SCO packet for unknown connection handle 284
May 27 19:45:45 sen kernel: [78358.317005] Bluetooth: hci0: SCO packet for unknown connection handle 284

## loading new pactl module
May 27 19:46:01 sen rtkit-daemon[1517]: Supervising 0 threads of 0 processes of 0 users.
May 27 19:46:01 sen rtkit-daemon[1517]: Successfully made thread 9804 of process 2338 (n/a) owned by '1001' RT at priority 5.
May 27 19:46:01 sen rtkit-daemon[1517]: Supervising 1 threads of 1 processes of 1 users.
May 27 19:46:01 sen kernel: [78374.225824] Bluetooth: hci0: SCO packet for unknown connection handle 0
May 27 19:46:01 sen kernel: [78374.225825] Bluetooth: hci0: SCO packet for unknown connection handle 0
May 27 19:46:01 sen kernel: [78374.235857] Bluetooth: hci0: SCO packet for unknown connection handle 0
May 27 19:46:01 sen kernel: [78374.235858] Bluetooth: hci0: SCO packet for unknown connection handle 0
</pre>
</details>

When it happens via Discord, this gets logged in syslog:
<details><summary>Expand</summary>
<pre>
May 27 20:38:25 sen discord.desktop[19709]: [022:440] [19818] (audio_device_pulse_linux.cc:1855): Can't query latency
May 27 20:38:25 sen discord.desktop[19709]: [022:510] [19818] (audio_device_pulse_linux.cc:1855): Can't query latency
May 27 20:38:25 sen discord.desktop[19709]: message repeated 2 times: [ [022:510] [19818] (audio_device_pulse_linux.cc:1855): Can't query latency]
May 27 20:38:25 sen discord.desktop[19709]: [022:510] [19814] (audio_send_stream.cc:285): Failed to set up send codec state.
May 27 20:38:25 sen discord.desktop[19709]: [022:512] [19818] (audio_device_pulse_linux.cc:1855): Can't query latency
May 27 20:38:25 sen discord.desktop[19709]: message repeated 3 times: [ [022:512] [19818] (audio_device_pulse_linux.cc:1855): Can't query latency]
May 27 20:38:26 sen discord.desktop[19709]: [023:000] [19818] (audio_device_pulse_linux.cc:1855): Can't query latency
</pre>
</details>

tyzbit, 2020-05-28:
PulseAudio doesn't appear to log any messages when the issue starts, but it does log messages when Discord tries to use the device:

<details><summary>Expand</summary>
<pre>
I: [pulseaudio] client.c: Created 98 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Got credentials: uid=1001 gid=1001 success=1
I: [pulseaudio] client.c: Created 99 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Got credentials: uid=1001 gid=1001 success=1
I: [pulseaudio] client.c: Created 100 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Got credentials: uid=1001 gid=1001 success=1
I: [pulseaudio] module-stream-restore.c: Restoring device for stream sink-input-by-application-name:Chromium.
I: [pulseaudio] sink-input.c: Trying to change sample rate
I: [pulseaudio] sink.c: Cannot update rate, SINK_IS_RUNNING, will keep using 8000 Hz
I: [pulseaudio] module-stream-restore.c: Restoring mute state for sink input sink-input-by-application-name:Chromium.
I: [pulseaudio] remap.c: Using stereo to mono remapping
I: [pulseaudio] speex.c: Choosing speex quality setting 1.
I: [pulseaudio] sink-input.c: Created input 38 "Playback" on btheadset with sample spec float32le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] sink-input.c:     application.icon_name = "chromium-browser"
I: [pulseaudio] sink-input.c:     media.name = "Playback"
I: [pulseaudio] sink-input.c:     application.name = "Chromium"
I: [pulseaudio] sink-input.c:     native-protocol.peer = "UNIX socket client"
I: [pulseaudio] sink-input.c:     native-protocol.version = "32"
I: [pulseaudio] sink-input.c:     application.process.id = "7139"
I: [pulseaudio] sink-input.c:     application.process.user = "tyzbit"
I: [pulseaudio] sink-input.c:     application.process.host = "sen"
I: [pulseaudio] sink-input.c:     application.process.binary = "Discord"
I: [pulseaudio] sink-input.c:     application.language = "en_US.UTF-8"
I: [pulseaudio] sink-input.c:     window.x11.display = ":0"
I: [pulseaudio] sink-input.c:     application.process.machine_id = "92ac2dd12f9045f3961d5c0df7002221"
I: [pulseaudio] sink-input.c:     application.process.session_id = "3"
I: [pulseaudio] sink-input.c:     module-stream-restore.id = "sink-input-by-application-name:Chromium"
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Final latency 414.96 ms = 200.00 ms + 2*7.48 ms + 200.00 ms
I: [pulseaudio] module-stream-restore.c: Restoring mute state for source output source-output-by-application-name:WEBRTC VoiceEngine.
I: [pulseaudio] source-output.c: Created output 34 "recStream" on alsa_input.pci-0000_00_1f.3.analog-stereo with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] source-output.c:     media.name = "recStream"
I: [pulseaudio] source-output.c:     application.name = "WEBRTC VoiceEngine"
I: [pulseaudio] source-output.c:     native-protocol.peer = "UNIX socket client"
I: [pulseaudio] source-output.c:     native-protocol.version = "32"
I: [pulseaudio] source-output.c:     application.process.id = "7107"
I: [pulseaudio] source-output.c:     application.process.user = "tyzbit"
I: [pulseaudio] source-output.c:     application.process.host = "sen"
I: [pulseaudio] source-output.c:     application.process.binary = "Discord"
I: [pulseaudio] source-output.c:     application.language = "en_US.UTF-8"
I: [pulseaudio] source-output.c:     window.x11.display = ":0"
I: [pulseaudio] source-output.c:     application.process.machine_id = "92ac2dd12f9045f3961d5c0df7002221"
I: [pulseaudio] source-output.c:     application.process.session_id = "3"
I: [pulseaudio] source-output.c:     module-stream-restore.id = "source-output-by-application-name:WEBRTC VoiceEngine"
I: [pulseaudio] protocol-native.c: Final latency 60.00 ms = 30.00 ms + 30.00 ms
I: [pulseaudio] sink-input.c: Trying to change sample rate
I: [pulseaudio] sink.c: Cannot update rate, SINK_IS_RUNNING, will keep using 8000 Hz
I: [pulseaudio] module-stream-restore.c: Restoring mute state for sink input sink-input-by-application-name:WEBRTC VoiceEngine.
I: [pulseaudio] speex.c: Choosing speex quality setting 1.
I: [pulseaudio] sink-input.c: Created input 39 "playStream" on btheadset with sample spec s16le 1ch 44100Hz and channel map mono
I: [pulseaudio] sink-input.c:     media.name = "playStream"
I: [pulseaudio] sink-input.c:     application.name = "WEBRTC VoiceEngine"
I: [pulseaudio] sink-input.c:     native-protocol.peer = "UNIX socket client"
I: [pulseaudio] sink-input.c:     native-protocol.version = "32"
I: [pulseaudio] sink-input.c:     application.process.id = "7107"
I: [pulseaudio] sink-input.c:     application.process.user = "tyzbit"
I: [pulseaudio] sink-input.c:     application.process.host = "sen"
I: [pulseaudio] sink-input.c:     application.process.binary = "Discord"
I: [pulseaudio] sink-input.c:     application.language = "en_US.UTF-8"
I: [pulseaudio] sink-input.c:     window.x11.display = ":0"
I: [pulseaudio] sink-input.c:     application.process.machine_id = "92ac2dd12f9045f3961d5c0df7002221"
I: [pulseaudio] sink-input.c:     application.process.session_id = "3"
I: [pulseaudio] sink-input.c:     module-stream-restore.id = "sink-input-by-application-name:WEBRTC VoiceEngine"
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Final latency 414.96 ms = 200.00 ms + 2*7.48 ms + 200.00 ms
[... cut ...]
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] protocol-native.c: Requested tlength=30.00 ms, minreq=7.48 ms
I: [pulseaudio] sink-input.c: Freeing input 39 "playStream"
</pre>
</details>

borine, 2020-05-29:
Unfortunately, I don't have free time to work on this atm, but will look at it again when I can.

I would suggest that you try simpler applications first, before Discord etc, just to discover if the basics are working.  Pulseaudio clients use one of two methods to interface with the pulse daemon: most modern Linux desktop applications use pulse's own native API, whereas many others use the alsa API through alsa's pulse plugin. The simplest available application for the pulse API is paplay, which is in the pulseaudio-utils package and included by default in Ubuntu desktop. The simplest alsa application is of course aplay. 

To test the pulseaudio API, try:
```
paplay --verbose --device=btheadset testfile.wav
```
And to test the alsa API, first make sure the headset is selected as the default sink in pulse, then try:
```
aplay --verbose -D pulse testfile.wav
```

If you cannot solve the choppy playback with either of these, then please post  the outputs here and I will eventually try to understand where the issue is.

tyzbit, 2020-05-29:
Great, thank you for your help so far, it has been tremendously helpful and I feel as though I'm extremely close to getting it working.  As a troubleshooting step I upgraded from 18.04 to 20.04, which has changed things a bit. I will take some time to continue investigating including using `paplay` as a simpler tool to test the PulseAudio integration.

tyzbit, 2020-07-13:
I'm unable to get my system back to the state it was in to continue troubleshooting, so I'm closing this issue.

closed 2020-07-13 by tyzbit