arkq/bluez-alsa #341: PCM information

juliandc1969, 2020-06-03:
Hi,

I'm just after some general information on how bluealsa-aplay gets the audio data. By this I mean from somewhere bluealsa gets audio data ( filling buffers etc ) to feed the ALSA's PCM. On various WiFi/BT modules there's a PCM output, is this what Bluealsa-aplay uses as an input via the SOC? 

Any information would be appreciated.
Cheers.

borine, 2020-06-03:
bluealsa gets all audio data from the HCI connection via the Bluez bluetooth kernel drivers. It is unable to access the PCM pins on BT module. If your module routes SCO data to the PCM output by default, then you will need to configure it to redirect to HCI instead. A2DP data is always routed through the HCI anyway.

juliandc1969, 2020-06-03:
Thanks! Suspected that was the case. I specifically need A2DP so it's HCI. Last question, how/where does Bluealsa-aplay find the audio stream once Bluealsa sets everything up? You mentioned A2DP is routed over HCI and obviously Bluealsa-alsa picks this up, but how?

borine, 2020-06-03:
bluealsa-aplay receives the audio stream over a pipe from the bluealsa daemon. The API by which this is negotiated uses D-Bus, and is documented here: https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-api.txt

closed 2020-06-04 by juliandc1969