arkq/bluez-alsa #342: system stuck when arecord via SCO.

WilsonWang-ai, 2020-06-07:
Hi  @Arkq 
### Only after the first time of burning and writing can sometimes the recording succeed and exit successfully.

bluez-alsa: 9c4ab2b
steps:
1)bluez-alsa &
2)bluetooth  
3)arecord -D bluealsa:SRV=org.bluealsa,DEV=5C_C6_E9_90_E6_76,PROFILE=sco -d 10  capture.wav
issue 1: arecord stuck, unable to go to ba-transport.c:906: New SCO link
issue 2: arecord stuck,Recording for 10s has been unable to exit. unable to go to bluealsa-pcm.c:327:/org/bluealsa/hci0/dev_5C_C6_E9_90_E6_76/hfpag/source: Stopping
issue 3: reboot system,Recording  stuck.

I have attached the  log of the following normal recording . please do refer & help us to analyze.
Thanks!

bluealsa: D: bluez.c:1121: Called: org.bluez.Profile1.NewConnection()
bluealsa: D: ba-rfcomm.c:1263: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
bluealsa: D: sco.c:202: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: bluez.c:1046: HFP Audio Gateway (CVSD) configured for device 5C_C6_E9_90_E6_76
bluealsa: D: ba-transport.c:701: State transition: 0 -> 2
bluealsa: D: ba-transport.c:975: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
bluealsa: D: sco.c:243: Starting SCO loop: HFP Audio Gateway (CVSD)
bluealsa: D: sco.c:79: Starting SCO dispatcher loop: hci0
bluealsa: D: ba-rfcomm.c:880: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
bluealsa: D: at.c:161: AT message: SET: command:+BRSF, value:921
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
bluealsa: D: at.c:161: AT message: TEST: command:+CIND, value:(null)
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
bluealsa: D: at.c:161: AT message: GET: command:+CIND, value:(null)
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 5 -> 7
bluealsa: D: at.c:161: AT message: SET: command:+CMER, value:3, 0, 0, 1
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 7 -> 8
bluealsa: D: ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 8 -> 9
bluealsa: D: at.c:161: AT message: SET: command:+VGS, value:15
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: at.c:161: AT message: SET: command:+VGM, value:11
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: at.c:161: AT message: GET: command:+BTRH, value:(null)
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: at.c:161: AT message: SET: command:+XAPL, value:05AC-1702-0100,7
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,6
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: at.c:161: AT message: SET: command:+BIA, value:0,1,1,1,0,0,0,0
bluealsa: D: ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: aptX
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: SBC
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Unknown vendor codec: d7 00 00 00 25 00 38
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: (null)
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Unknown vendor codec: 0a 00 00 00 04 01 07 08 00 02 c0 ff 8c 84 09 b6
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: (null)
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Unknown vendor codec: d7 00 00 00 26 00 07 06 00 00 38 17 02 35
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: (null)
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Unknown vendor codec: d7 00 00 00 25 00 38
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: (null)
bluealsa: D: bluez.c:1346: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Unknown vendor codec: 0a 00 00 00 04 01 07 08 00 02 c0 ff 8c 84 09 b6
bluealsa: D: bluez.c:1425: Adding new Stream End-Point: 5C_C6_E9_90_E6_76: (null)
[... 31 lines cut ...]
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Initializing SW
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Initializing SW
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Initializing SW
D: bluealsa-pcm.c:437: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Prepared
D: bluealsa-pcm.c:415: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Initializing SW
D: bluealsa-pcm.c:285: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Starting
bluealsa: D: ba-transport.c:701: State transition: 2 -> 2
D: bluealsa-pcm.c:143: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Starting IO loop: 7
D: bluealsa-pcm.c:327: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Stopping
D: bluealsa-pcm.c:111: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: IO thread cleanup
D: bluealsa-pcm.c:407: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Freeing HW
D: bluealsa-pcm.c:357: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Closing
root@OpenWrt:~# bluealsa: D: ba-transport.c:948: Closing PCM: 17
bluealsa: D: sco.c:330: Releasing SCO due to PCM inactivity
bluealsa: D: ba-transport.c:924: Closing SCO: 20

root@OpenWrt:~# bluealsa: D: bluez.c:1435: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: bluez.c:1435: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()

root@OpenWrt:~# arecord -D bluealsa:SRV=org.bluealsa,DEV=5C_C6_E9_90_E6_76,PROFILE=sco -d 10  capture.wav
bluealsa: D: bluealsa-dbus.c:186: Called: org.bluealsa.Manager1.GetPCMs()
D: bluealsa-pcm.c:671: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Setting constraints
Recording WAVE 'capture.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
D: bluealsa-pcm.c:368: /org/bluealsa/hci0/5C_C6_E9_90_E6_76/hfpag/source: Initializing HW
bluealsa: D: bluealsa-dbus.c:392: Called: org.bluealsa.PCM1.Open()

gitdachong, 2020-09-06:
hello, have you solved this problem ? i have the same situation on my raspberry pi 4 ,64bit system

borine, 2021-07-17:
There has been significant change to the SCO code in bluez-alsa since this issue was raised. So the issue may not be present any more, and even if it is, the logs here are no  longer helpful in tracing the cause. So I am closing it now; however if testing with the latest code from master branch reveals similar behaviour, please open a new issue with updated error logs.

closed 2021-07-17 by borine