arkq/bluez-alsa #344: Unknown PCM

foxupdown, 2020-06-13:


borine, 2020-06-13:
> ALSA lib dlmisc.c:254:(snd1_dlobj_cache_get) Cannot open shared library /lib/alsa-lib/libasound_module_pcm_bluealsa.so

` /lib/alsa-lib/libasound_module_pcm_bluealsa.so` is unreadable by aplay. It may: 
- be missing
- have the wrong permissions
- be the wrong format.

Please check:

`ls -l /lib/alsa-lib`

and

`file /lib/alsa-lib/*`

foxupdown, 2020-06-13:
to verity your questions:
ls /lib/alsa-lib/libasound_module_pcm_bluealsa.so 
/lib/alsa-lib/libasound_module_pcm_bluealsa.so
file lib/alsa-lib/libasound_module_pcm_bluealsa.so 
lib/alsa-lib/libasound_module_pcm_bluealsa.so: ELF 32-bit LSB  shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, not stripped

borine, 2020-06-13:
Your response does not show the file permissions. I assume your system is 32-bit ARM ?

foxupdown, 2020-06-13:
1 .ls -l /lib/alsa-lib/libasound_module_pcm_bluealsa.so 
-rwxr-xr-x    1 root     root         45768 Dec 27 09:07 /lib/alsa-lib/libasound_module_pcm_bluealsa.so
2. my system is 32-bit-arm (armV7)

borine, 2020-06-13:
Ok, I just thought of one other possible cause:
- missing dependency

please check

`ldd /lib/alsa-lib/libasound_modue_pcm_bluealsa.so`

foxupdown, 2020-06-13:
ok 
in arm arm-himix100-linux-readelf -d libasound_module_pcm_bluealsa.so 

Dynamic section at offset 0x7ef0 contains 30 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libpthread.so.0]
 0x00000001 (NEEDED)                     Shared library: [libasound.so.2]
 0x00000001 (NEEDED)                     Shared library: [libdbus-1.so.3]
 0x00000001 (NEEDED)                     Shared library: [libm.so.0]
 0x00000001 (NEEDED)                     Shared library: [libc.so.0]
 0x0000000e (SONAME)                     Library soname: [libasound_module_pcm_bluealsa.so]
 0x0000000f (RPATH)                      Library rpath: [/system/lib]
 0x0000000c (INIT)                       0x1d80
 0x0000000d (FINI)                       0x6cd8
 0x00000019 (INIT_ARRAY)                 0x17e94
 0x0000001b (INIT_ARRAYSZ)               4 (bytes)
 0x0000001a (FINI_ARRAY)                 0x17e98
 0x0000001c (FINI_ARRAYSZ)               4 (bytes)
 0x00000004 (HASH)                       0xd4
 0x00000005 (STRTAB)                     0xdec
 0x00000006 (SYMTAB)                     0x51c
 0x0000000a (STRSZ)                      2489 (bytes)
 0x0000000b (SYMENT)                     16 (bytes)
 0x00000003 (PLTGOT)                     0x18000
 0x00000002 (PLTRELSZ)                   912 (bytes)
 0x00000014 (PLTREL)                     REL
 0x00000017 (JMPREL)                     0x19f0
 0x00000011 (REL)                        0x18e0
 0x00000012 (RELSZ)                      272 (bytes)
 0x00000013 (RELENT)                     8 (bytes)
 0x6ffffffe (VERNEED)                    0x18c0
 0x6fffffff (VERNEEDNUM)                 1
 0x6ffffff0 (VERSYM)                     0x17a6
 0x6ffffffa (RELCOUNT)                   26
 0x00000000 (NULL)                       0x0

I find the  Library rpath: [/system/lib] is not normal ， it means I need to comiple the blue-alsa-master  ?

borine, 2020-06-13:
> I find the Library rpath: [/system/lib] is not normal 

Hmm, my builds do not have the RPATH entry. According to `man 8 ld.so` use of RPATH in the dynamic section is deprecated. However, the dynamic linker should continue to search the other standard locations if a dependency is not found in the RPATH.

If you have `ldd` on your system, I thank that should report where all dependency libraries were found (including recursive dependencies) and explicity identify any missing ones.

foxupdown, 2020-06-13:
then I try this way to verity
arm-himix100-linux-ld -d libasound_module_pcm_bluealsa.so 
arm-himix100-linux-ld: warning: libasound.so.2, needed by libasound_module_pcm_bluealsa.so, not found (try using -rpath or -rpath-link)
arm-himix100-linux-ld: warning: cannot find entry symbol _start; defaulting to 0001022c
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_ioplug_delete@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_ioplug_set_param_list@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_avail_update@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_lib_error@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_format_physical_width@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_get_id@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `eventfd_read'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_ioplug_mmap_areas@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_iterator_end@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `eventfd_write'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_stream@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_get_string@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_output_printf@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_iterator_next@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_get_integer@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_ioplug_set_param_minmax@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_ioplug_create@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_pcm_sw_params_get_boundary@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_iterator_entry@ALSA_0.9'
libasound_module_pcm_bluealsa.so: undefined reference to `snd_config_iterator_first@ALSA_0.9'
It seems that some interfaces is not find

borine, 2020-06-13:
> arm-himix100-linux-ld: warning: libasound.so.2, needed by libasound_module_pcm_bluealsa.so, not found (try using -rpath or -rpath-link)

ld is not the dynamic linker. You need to find out if, and then why, ld-linux.so is not finding some dependencies at runtime. `ldd` is the best tool for that. If it reports some dependencies not found, then you will know what you need to install to fix this. If it reports all dependenices found, then perhaps `libasound_module_pcm_bluealsa.so` is actually corrupted in some way.

foxupdown, 2020-06-13:
I try int 32/64-bit virtual machine 
ldd libasound_module_pcm_bluealsa.so 
        not a dynamic executable
[https://github.com/foxupdown/20200502/commit/e7cfb1bdfddf45a8a53303daea477ec6a44ec492](url)  the dynamic library’s url
either you coule try to parse it

borine, 2020-06-13:
on the machine that you ran aplay for the original comment, please can you run `aplay --version`

foxupdown, 2020-06-13:
aplay --version  is 1.1.1. thanks

borine, 2020-06-13:
OK, so it looks like libasound on the host machine is compatible with the one the bluealsa was built with.

So the remaining possibilities are that some runtime dependency is not installed on the host machine or the bluealsa module is corrupt. The only way I know to find a missing runtime library dependency is to run 

`ldd /lib/alsa-lib/libasound_module_pcm_bluealsa.so` 

__on the host machine (that is, the one on which aplay is failing) with the same account, in the same environment, as you ran aplay with__.

foxupdown, 2020-06-14:
thanks 
I will try and troubleshoot this issue.

foxupdown, 2020-07-06:
this issue finished 
the cross compile is based by uclibc
change the interface such as pthread_setname_np ,  exp20 , eventfd_read / wirte
that solved
thanks！

closed 2020-07-06 by foxupdown