arkq/bluez-alsa #345: Connecting to bluealsa from inside Docker container

jamesyc, 2020-06-15:
I am currently running a Docker container, on a raspberry pi 3b running stretch. 

Bluealsa is installed by default on Raspbian, and is configured fine on the host. Running `aplay -D bluealsa test.wav` works fine. 

Inside the docker container, the PCMs from the host isn't visible. Running `aplay -L` doesn't list the bluealsa PCMs.  I added the option `--device /dev/snd ` to `docker run`, but I think this only allows the docker container to access the hardware sound cards. 

In addition, bluealsa isn't installed within the docker container. The docker container has a Debian `/etc/apt/sources.list`, and doesn't seem to be able to install bluealsa via apt. (The package is straight up missing from the package manager), even though Raspbian has bluealsa preinstalled.

Is there a way to connect to the host bluealsa from within the Docker container? Preferably without needing to install bluealsa in the container. If I need to install bluealsa, should I build from source from the repo here or from the raspbian repo?

Arkq, 2020-06-15:
See these issues: #313, #335

And I think you will need bluez-alsa installed inside container (at least ALSA plug-ins).

arantes555, 2020-10-06:
@jamesyc I managed to make it work, with a bit of trickery:
```bash
docker run \
-v /etc/asound.conf:/etc/asound.conf:ro \ # sharing the alsa config with the container
-v /usr/lib/arm-linux-gnueabihf/alsa-lib:/usr/lib/arm-linux-gnueabihf/alsa-lib:ro \ # sharing the bluealsa libraries with the container (simpler than installing them there ... Yeah, it's not really pretty^^)
-v /run/dbus/system_bus_socket:/run/dbus/system_bus_socket \ # sharing the system D-Bus so the container's ALSA can communicate with the hosts' bluealsa
your_image
```

closed 2021-01-10 by Arkq