arkq/bluez-alsa #350: missing release tarballs

gearhead, 2020-07-03:
Maybe I am just crazy, but I've been building the release tarball for this and was looking to see if there was an update and cannot find any tar balls. 
Used to go here:
https://github.com/Arkq/bluez-alsa/archive/v${pkgver}.tar.gz
the last version I built was 2.1.0. Have you moved to git updates only or are there still releases? If it is right in front of me and I cannot see it, please let me know where to get it.

Arkq, 2020-07-05:
I did not change anything regarding archives on github. I'm not sure if it is even possible to modify such thing :D ZIP files and tarballs are listed here: https://github.com/Arkq/bluez-alsa/releases e.g. https://github.com/Arkq/bluez-alsa/archive/v2.1.0.tar.gz

gearhead, 2020-07-05:
My bad. I do not what is going on. Maybe it is a github update thing.  There used to be a 'Releases' tab and it is gone. You have to type it by hand, as you showed, I cannot figure out how to navigate there from the main github repository page. 

Try going to Nginx and find the releases. If you go to the code then switch between the tags to releases, you get the code view, but I have not figured out how to navigate to the tarballs other than typing it in by hand.
Sorry!

Arkq, 2020-07-05:
[image]

closed 2020-07-05 by Arkq

gearhead, 2020-07-05:
Like I said, I am a doofus. It is right in front of me, just in a different location. Thanks!