arkq/bluez-alsa #352: assistance debugging arecord "Couldn't get BlueALSA PCM: PCM not found"

dannaughton, 2020-07-06:
I could use some assistance finding a way to get arecord to work. I can't seem to configure this to make a device available to record from. (Playing works fine - BT connection from phone audio to Raspberry PI board a2dp-sink with I2S amplifier/speaker using alsa). I just can't get access to read from the stream. I can see the mixer in alsamixer has a capture PCM - I just can open it. Thanks in advance for any advice.

error:
arecord -D bluealsa -d 1 /tmp/test.wav
ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found

amixer -D bluealsa
Simple mixer control 'galaxys9 - A2DP',0
Capabilities: cvolume cswitch
Capture channels: Front Left - Front Right
Limits: Capture 0 - 127
Front Left: Capture 94 [74%] [on]
Front Right: Capture 94 [74%] [on]

arecord -l
**** List of CAPTURE Hardware Devices ****
<nothing?>

arecord -L
null
Discard all samples (playback) or generate zero samples (capture)
jack
JACK Audio Connection Kit
pulse
PulseAudio Sound Server
default
mono
softvol
dmixer
soundcard
dsnooped
bluetooth
usbstream:CARD=sndrpihifiberry
snd_rpi_hifiberry_dac
USB Stream Output

ps aux | grep alsa
root 250 0.0 0.4 3676 2056 ? SNs 10:08 0:00 /usr/sbin/alsactl -E HOME=/run/alsa -s -n 19 -c rdaemon
root 284 7.6 1.1 34888 5084 ? Ssl 10:08 7:05 /usr/bin/bluealsa -p a2dp-sink
dan 359 6.2 0.7 15504 3200 ? SLsl 10:09 5:45 /usr/bin/bluealsa-aplay 00:00:00:00:00:00

Alsa-info can be found here: [https://pastebin.com/5WhLsaXv](url)

borine, 2020-07-06:
>arecord -L
...
pulse
PulseAudio Sound Server
...
bluetooth
...

You cannot have both pulseaudio and bluealsa accessing bluetooth audio; did you disable the pulseaudio bluetooth modules?
What is the PCM called `bluetooth` ? please can you post the definition of this PCM.
 
>ps aux | grep alsa
...
dan 359 6.2 0.7 15504 3200 ? SLsl 10:09 5:45 /usr/bin/bluealsa-aplay 00:00:00:00:00:00

If bluealsa-aplay is capturing from your 'phone, then arecord will not be able to capture at the same time.

Arkq, 2020-07-06:
From README:

> Please note that this PCM device is based on the ALSA software PCM I/O plugin - it will not be available in the ALSA Kernel proc interface.

Hence, connected devices will not be available in the `aplay -l` nor `arecord -l`. In order to get the list of available devices please use `bluealsa-aplay -l` or `bluealsa-aplay -L`.

> Playing works fine.

What kind of playing? You have smartphone connected to RPi and you can play audio from smartphone via bluetooth on RPi internal sound card using bluealsa-aplay, is that right?

If the above statement is true, then there is an issue with bluealsa ALSA plugin. Please see this comment: https://github.com/Arkq/bluez-alsa/issues/347#issuecomment-654038985

Arkq, 2020-07-06:
> If bluealsa-aplay is capturing from your 'phone, then arecord will not be able to capture at the same time.

@borine trying to open PCM twice results in "EBUSY" error, e.g.:

```sh
$ bluealsa-aplay 10:B1:F8:E6:10:D4 >/dev/null &
$ arecord -D smartphone
D: bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_10_B1_F8_E6_10_D4/a2dpsnk/source: Setting constraints
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
D: bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_10_B1_F8_E6_10_D4/a2dpsnk/source: Initializing HW
D: bluealsa-pcm.c:375: /org/bluealsa/hci0/dev_10_B1_F8_E6_10_D4/a2dpsnk/source: Couldn't open PCM: Device or resource busy
```

Arkq, 2020-07-06:
@dannaughton 

> arecord -D bluealsa -d 1 /tmp/test.wav

Do you have defined `bluealsa` in your local .asoundrc file? If not, then it will not work, because by default there is no DEV specified. Please, look at the README how to correctly setup .asoundrc file.

dannaughton, 2020-07-06:
Thanks for you tips.

pulse is not installed or running.    ("ps aux | grep pulse" to confirm)

"If bluealsa-aplay is capturing from your 'phone, then arecord will not be able to capture at the same time" - is there another way?

bluealsa-aplay-l or -L is not supported.  bluealsa-aplay version  1.4.0
bluealsa-aplay  -L
bluealsa-aplay: invalid option -- 'L'

"You have smartphone connected to RPi and you can play audio from smartphone via bluetooth on RPi internal sound card using bluealsa-aplay, is that right?" - that is correct.

Do you have defined bluealsa in your local .asoundrc file?  - using asound.conf  -  

https://pastebin.com/RASWVmTb

I'm not an expert at alsa - the docs said to use asym - one for capture - one for playback.  Playback works.  Playback goes from asym ==>  mon ==> softvol ==> dmix ==> hw.  Capture goes from asym ==> dsnoop ==> bluealsa  ==> hw

Maybe my asound.conf is wrong?

Arkq, 2020-07-06:
> asym ==> dsnoop ==> bluealsa ==> hw

I'm afraid dsnoop is not supported by bluealsa. Also, I advise to use fresh master snapshot - bluez-alsa-1.4.0 is pretty much outdated.

dannaughton, 2020-07-06:
OK - downloaded, built, installed from git.  Streaming from phone to RPI over bluetooth still works - that's good.

bluealsa-aplay  -V
v2.1.0-65-gf80cd48

bluealsa-aplay  -L
bluealsa:SRV=org.bluealsa,DEV=14:9F:3C:E7:48:54,PROFILE=a2dp
    galaxys9, trusted phone, capture
    A2DP (SBC): S16_LE 2 channels 44100 Hz

bluealsa-aplay  -l
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****
hci0: 14:9F:3C:E7:48:54 [galaxys9], trusted phone
  A2DP (SBC): S16_LE 2 channels 44100 Hz

arecord -D bluealsa -d 1 /tmp/test.wav
ALSA lib ../../../src/asound/bluealsa-pcm.c:802:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
arecord: main:828: audio open error: No such device

I removed the dsnoop from the asound.conf  https://pastebin.com/FEQVhZsQ
still not sure if that asound.conf is correct.

I think I'm closer....:

arecord -D bluealsa:SRV=org.bluealsa,DEV=14:9F:3C:E7:48:54,PROFILE=a2dp -d 1 /tmp/test.wav
D: ../../../src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Setting constraints
Recording WAVE '/tmp/test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:375: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Couldn't open PCM: Device or resource busy
arecord: set_params:1403: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  U8
SUBFORMAT:  STD
SAMPLE_BITS: 8
FRAME_BITS: 8
CHANNELS: 1
RATE: 8000
PERIOD_TIME: (92879 92880)
PERIOD_SIZE: [743 743]
PERIOD_BYTES: (743 744)
PERIODS: (4 5)
BUFFER_TIME: 464375
BUFFER_SIZE: 3715
BUFFER_BYTES: 3715
TICK_TIME: 0
D: ../../../src/asound/bluealsa-pcm.c:357: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Closing

Arkq, 2020-07-06:
> Couldn't open PCM: Device or resource busy

Yes, that's the case where you try to open PCM more then once, which is currently not possible. If you kill bluealsa-aplay (or some other application which might use bluealsa), you shall be able to use arecord.

dannaughton, 2020-07-06:
It seems to be working now. It writes out a .wav file.  The file seems to only have the wav header in it - but that's a minor detail.  Thank you for your assistance and advice.

Unfortunately, I think what I wanted to do is not possible - use a raspberry pi board for a bluetooth speaker with a built in audio spectrum visualizer.  The bluetooth speaker part works - but there is no way to access the audio stream once it's running to connect the visualizer.

sudo arecord -D bluealsa:SRV=org.bluealsa,DEV=14:9F:3C:E7:48:54,PROFILE=a2dp -d 5 /tmp/test.wav
D: ../../../src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Setting constraints
Recording WAVE '/tmp/test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:400: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Selected HW buffer: 5 periods x 16384 bytes == 81920 bytes
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Prepared
D: ../../../src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Starting
D: ../../../src/asound/bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_14_9F_3C_E7_48_54/a2dpsnk/source: Starting IO loop: 7

Arkq, 2020-07-06:
> The bluetooth speaker part works - but there is no way to access the audio stream once it's running to connect the visualizer.

Currently it's not possible out of the box. But I think you shall search through bluez-alsa issues for `alsaloop` keyword. I've never created setup with bluez-alsa and ALSA loopback, but that might be the way to go (e.g. https://github.com/Arkq/bluez-alsa/issues/48). Otherwise, you might make some modifications in the bluealsa-aplay for hooking visualizer :)

dannaughton, 2020-07-06:
Thanks Arkq.  I think I might be at the end of my skills with this one.  Whoever put this bluealsa stuff together knows 1000X more than I do.  What would take them an hour would take me the rest of my life. I'll check back periodically, maybe I just shelve this idea for now.  I guess this issue is closed out - thanks.

borine, 2020-07-07:
> use a raspberry pi board for a bluetooth speaker with a built in audio spectrum visualizer

I haven't tried this myself, but something like this might work:
```
arecord -D bluealsa:SRV=org.bluealsa,DEV=14:9F:3C:E7:48:54,PROFILE=a2dp -f cd - | tee /tmp/test.wav | aplay -f cd -
```

or, if you want to pipe the stream directly into a spectrum visualizer program:
```
PIPE=$(mktemp -u)
mkfifo $PIPE
arecord -D bluealsa:SRV=org.bluealsa,DEV=14:9F:3C:E7:48:54,PROFILE=a2dp -f cd - | tee $PIPE | aplay -f cd -
spectrum-visualizer < $PIPE

rm $PIPE
```

borine, 2020-07-07:
> I haven't tried this myself

Actually, just tried it on my laptop with cava (https://github.com/karlstav/cava)

cava requires the FIFO name to be put in its config fie, so I put in the `[input]` section of `~/.config/cava/config`
```
method = fifo
source = /tmp/bluealsa.fifo
sample_rate = 44100
sample_bits = 16
```
then tried it with:

```
mkfifo /tmp/bluealsa.fifo
arecord -D bluealsa:SRV=org.bluealsa,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp -f cd - | tee /tmp/bluealsa.fifo | aplay -f cd - &
cava
```

cool !!

borine, 2020-07-07:
Here's another solution, this time using bluealsa-aplay. This one requires an entry in ~/.asoundrc

```
pcm.fifo_copy {
    type plug
    slave {
        pcm {
            type file
            slave.pcm "default"
            file "/tmp/bluealsa.fifo"
            format "raw"
        }
        format "s16_le"
        channels 2
        rate 44100
    }
}
```

A fifo will block if nothing is reading from it, so we need to start the visualizer before bluealsa-aplay; that's easier if we do it in a different terminal window.

```
mkfifo /tmp/bluealsa.fifo
cava
```

```
bluealsa-aplay -D fifo_copy --profile-a2dp 00:00:00:00:00:00
```

I've used `rate 44100` in both these examples, because you posted
>bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=14:9F:3C:E7:48:54,PROFILE=a2dp
galaxys9, trusted phone, capture
A2DP (SBC): S16_LE 2 channels 44100 Hz

but you could also use `rate 48000` 

I like this solution better, because you do not need to hard-code the 'phone bluetooth address into any config files.

.

dannaughton, 2020-07-07:
I think I found something I can work with.  I used the fifo_copy approach from borine (thanks borine).  I loaded snd_aloop into the module.conf.    When you read the stream from the fifo with a simple file reader  (cat /tmp/bluealsa.fifo works too) , the audio plays through the speaker.  I couldn't get cava working, but I'll work on that later - looks like a cool program.  But I think I can make this work - thanks all.

borine, 2020-07-08:
>  I loaded snd_aloop into the module.conf.

Loopback is not needed in this solution; that's only required for dsnoop with bluealsa, and this method is intended to avoid the complexity of all that.

BTW I wasn't trying to recommend cava - I have no experience with (or knowledge of) audio spectrum visualization. I was just looking for a very simple application I could use for testing, and cava was the first hit I found. I'm sure that just about any audio visualizer will work with the fifo_copy method.

dannaughton, 2020-07-08:
After a day of using it - it works okay.  If you are not reading from the named pipe file, bluetooth-aplay will terminate after a while.  And if you stop reading from the pipe - same thing - blueotooth-aplay will terminate.  So the blocking in a pipe and bluetooth-aplay stopping is something you have to work around.  (killall bluetooth-aplay & restart) - Thanks All

borine, 2020-07-09:
A weakness of the fifo approach is that you have to make sure the visualizer is running at all times, otherwise playback will be blocked waiting for it to start, and any application using the fifo output will be terminated wehn the visualizer terminates.

Do really need to monitor the input stream? You would have much more flexibility if you can monitor the output instead. Here is another solution that visualizes the audio playing though the i2c soundcard. This one does require the Loopback device, so you would need to load the snd-aloop kernel module. The alsa configuration looks like this:

```
pcm.!default {
    type asym
    playback.pcm "monitored"
    capture.pcm "null"
}

ctl.!default {
    type hw
    card sndrpihifiberry
}

# software mixer for HiFiBerry
pcm.dmixer {
    type dmix
    ipc_key 1024
    ipc_perm 0666
    slave {
        pcm "hw:sndrpihifiberry"
        channels 2
        rate 44100
        format "s16_le"
    }
    bindings {
        0 0
        1 1
    }
}

# software mixer for loopback device
pcm.dmixerloop {
    type dmix
    ipc_key 1025
    ipc_perm 0666
    slave {
        pcm "hw:Loopback,1,0"
        channels 2
        rate 44100
        format "s16_le"
    }
    bindings {
        0 0
        1 1
    }
    hint.show off
}

# duplicate stream to both HiFiBerry and Loopback
pcm.monitored {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "dmixer" }
            b { channels 2 pcm "dmixerloop" }
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]
        [ 0 1 0 1 ]
    ]
}
```

The visualizer then needs to read the stream from the device hw:Looback,0,0. For example, using cava, the config would be:
```
[input]
method = alsa
source = hw:Loopback,0,0
```
and you can run bluealsa-aplay as

`bluealsa-aplay --profile-a2dp 00:00:00:00:00:00`

Now you can stop and start the visualizer as you wish, playback will continue unaffected.

dannaughton, 2020-07-09:
thanks borine - I just tried it out and I like that a lot better.  bluealsa-aplay works as expected.  And my application can pick up the loopback independent of the bluealsa-aplay.  It even works with the standard java audio interface - I can record the stream to a wav file, and everything checks out.  Now I can use the data for the visualizer.    you rock - thanks.

borine, 2020-07-09:
If you need to both record and visualize, then you can wrap the loopback capture device with dsnoop so that multiple applications can capture from it:

```
pcm.hifiberry_monitor {
    type dsnoop
    ipc_key 1026
    ipc_perm 0666
	slave {
        pcm "hw:Loopback,0,0" 
        channels 2
        rate 44100
        format "s16_le"
    }
    bindings {
        0 0
        1 1
    }
}
```

visualizer should then use the dsnoop device:

```
[input]
method = alsa
source = hifiberry_monitor
``` 

and you can record from it at the same time, eg:

`arecord -D hifiberry_monitor -f cd recorded.wav`

borine, 2021-06-21:
This issue appears to be resolved

closed 2021-06-21 by borine