arkq/bluez-alsa #353: Is it possible to arecord via a2dp-sink?

robertd99, 2020-07-06:
My current setup looks like this : 
<pre>
                                                  ---->3.5mm Audio-Output
                                               /
Smartphone(Bluetooth) ---> A2DP-Dongle ---> Pi2 
                                               \
                                                 ----->HDMI Audi-Output

</pre>
I use multi and route to achieve the routing of the Audio to the 2 hw outputs.
With "bluealsa-aplay XX:XX..." im perfectly able to stream sound from my Phone to the Speakers, but i want to be able to arecord it to a file or pipe that directly to aplay. I do not want to use bluealsa-aplay. 

My initial goal was to have a setup that looks like this : 
<pre>
                                                   -->Bluetooth Dongle 1-----> Audio Output 1(Speaker 1) 
                                                 /
Smartphone(Bluetooth)--> Bluetooth Dongle 0--->RPi
                                                 \ 
                                                   -->Bluetooth Dongle 2-----> Audio Output 2(Speaker 2)
</pre>
But i had to learn that it is not possible to have both, a A2DP-Sink AND A2DP-Source with bluealsa at the same time.
So now i want to seperate it in 2 Steps: recording with arecord and replaying with aplay.

Now is it even possible to do that? Can i use a A2DP-Sink to arecord? Are there more obstacles that i should know about? I know these are pretty basic questions but i couldn't find any answer to them.  This is my first project with Audio sorry for those basic questions :/

Arkq, 2020-10-12:
Yes, you can simply enable sink and source A2DP profile and connect phone to one dongle, then setup "forwarding" with `bluealsa-aplay` and ALSA `multi` plugin. More complicated setup would involve two instances of bluealsa service - one for sink A2DP and other for source A2DP (when using only one bluealsa service, all 3 dongles will report sink and source capabilities, you will have to recognize them by MAC address or customized names set by BlueZ).

See this links:
- https://github.com/Arkq/bluez-alsa/issues/379#issuecomment-707309026
- https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa.8.rst#example
- https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-aplay.1.rst

borine, 2021-06-24:
This question was answered, and there has been no follow-up questions for 9 months since then, so closing. Please re-open if you are still working on this and have more questions,

closed 2021-06-24 by borine