arkq/bluez-alsa #354: build fails on uClibc-ng

hthiery, 2020-07-12:
When compiling with buildroot and uClibc-ng is used as C library the build fails.

```
libtool: compile:  /home/hthiery/sources/mainline/buildroot/output/host/bin/i686-buildroot-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I../.. -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -I../../src -I/home/hthiery/sources/mainline/buildroot/output/host/bin/../i686-buildroot-linux-uclibc/sysroot/usr/include/dbus-1.0 -I/home/hthiery/sources/mainline/buildroot/output/host/bin/../i686-buildroot-linux-uclibc/sysroot/usr/lib/dbus-1.0/include -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Os -MT ../shared/log.lo -MD -MP -MF ../shared/.deps/log.Tpo -c ../shared/log.c  -fPIC -DPIC -o ../shared/.libs/log.o
../shared/log.c:13:10: fatal error: execinfo.h: No such file or directory
   13 | #include <execinfo.h>
      |          ^~~~~~~~~~~~
```

The include of `execinfo.h` is needed for `backtrace_symbols`. But this is only used when DEBUG is defined.

Should the file be included only in case of DEBUG?

Arkq, 2020-07-12:
Yes, this functionality is not available in all libc libraries, so indeed it might be in the `#if DEBUG`. However, I think that the better idea is to actually check for backtrace functionality with autotools. I'll try to fix that. Thanks for reporting this issue :)

closed 2020-07-13 by Arkq

hthiery, 2020-07-14:
Hi, retested with current HEAD and now it builds fine. Thank you again.

commit e5acb73f (2020-07-13): Auto-detect libc call stack backtrace support

Fixes #354