arkq/bluez-alsa #355: bluez-alsa codec selection weirdness

rodan, 2020-08-02:
I'm testing two different versions of bluez-alsa (2.1.0 and latest git snapshot) with different codecs enabled during compile time. my wish is to have one of the more modern ones: aptx-hd or ldac. bluealsa is used as source for a FiiO Q5s DAC, which should be supporting Bluetooth 5.0 AAC/SBC/aptX/aptX LL/aptX HD/LDAC/HWA

with version 2.1.0 if I compile with 
 --enable-rfcomm --disable-aac --enable-debug --disable-ofono --disable-static --disable-test --enable-hcitop --enable-ldac --disable-upower --disable-aptx --disable-aptx-hd
then SBC is the one and only codec that gets presented during the initial handshake. and that one works.
I tried to force the invalidation of SBC in bluez_endpoint_select_configuration() in the hope of getting a different codec during the next invocation, but I seem to only get SBC presented twice. not sure what the life cycle of the codec list is and who gets to generate it in the first place. based on the rather limited wireshark capture looks like bluealsa is picking a codec and FiiO accepts it.
if I also compile aptx, aptx-hd then aptx-hd gets picked but I can only hear long beeps and get presented with a demo-library warning in the debug output. fair enough.

with the latest git snapshot I get this debug info: https://pastebin.com/WfSXzig3
most vendor codecs are now presented to bluealsa, one of them is not recognized, aptx-hd is picked, but playing thru it is not possible - cmus returns 'Error: opening audio device: No such device'.
if I disable aptx* during compilation, then SBC is picked, since apparently LDAC (which is compiled in) is the one not recognized as a valid codec - "Unknown vendor codec: d7 00 00 00 02 00 32 02 00 00 00 00 00 00 00 00 00"
and not even SBC is now working with cmus - same error as above. does alsa need a different setup then with v2.1.0?
is the latest snapshot in a working state?

I will be digging some more and add my findings to this ticket.

Arkq, 2020-08-02:
As ar as I know, the current master snapshot should be fully functional. I can guarantee that it works with SBC and AAC (these codecs are supported by my headset, and I'm using it all the time), but for other codecs it "should" work.

Current master snapshot allows to select A2DP codec on runtime (with bluez >= 5.52), however, there is no user-friendly tool for that (unles someone has made some). You can use any D-Bus client for switching to desired codec, e.g.:
```sh
# get the list of connected device
gdbus call --system -d org.bluealsa -o /org/bluealsa -m org.bluealsa.Manager1.GetPCMs
# get the list of available codecs (supported by bluealsa and by remote BT device)
gdbus call --system -d org.bluealsa -o '/org/bluealsa/hci0/dev_1C_48_F9_9D_81_5C/a2dpsrc/sink' -m org.bluealsa.PCM1.GetCodecs
# select desired codec (in that case SBC with configuration selected by bluealsa)
gdbus call --system -d org.bluealsa -o '/org/bluealsa/hci0/dev_1C_48_F9_9D_81_5C/a2dpsrc/sink' -m org.bluealsa.PCM1.SelectCodec 'SBC' '{}'
# select SBC with (valid) configuration provided by us
gdbus call --system -d org.bluealsa -o '/org/bluealsa/hci0/dev_1C_48_F9_9D_81_5C/a2dpsrc/sink' -m org.bluealsa.PCM1.SelectCodec 'SBC' '{"Configuration":<[byte 0x88, 0x85, 0x0f, 0x35]>}'
```

From the provided log file I can see, that indeed bluealsa recognizes a lot of codecs (I think that you are right that  unrecognized codec might be aptx-LL but with BT_COMPID_QUALCOMM_TECH vendor ID).

> cmus returns 'Error: opening audio device: No such device'.

Firstly, check what is the device ALSA name with (this will list all available bleualsa PCMs):
```
bluealsa-aplay -L
```

If you are using cmus, then, if bluez-alsa was compiled with debug enabled, you should get debug logs from bluealsa ALSA plug-in - scattered in cmus terminal window (ALSA logs are not friendly with ncurses cmus window). In these logs you can find selected bluealsa PCM device. Make sure that selected PCM is indeed available. Also make sure, that after installing bluez-alsa-9999, ALSA plug-ins are also updated (e.g. restart cmus), because bluez-alsa-9999 is not backward compatible with bluez-alsa-2.1.0 (next bluez-alsa release will be bluez-alsa-3.0.0).

rodan, 2020-08-02:
Hello!

apparently FiiO is using Quallcom CSR8675 as the bluetooth chip and LDAC is not supported by it, so I'm unsure how exactly they can sink LDAC.

just managed to get the git snapshot working as good as v2.1.0 (SBC and AAC are good), but looking at the logs I was in for a dose of disappointment. https://pastebin.com/RhDQ9G8B
until now I guessed that the log warning was generated by the older bluealsa, but I now realize it's actually part of the openaptx. which raises questions on so many levels.

did someone manage to link ffmpeg's encoding with bluez-alsa?

Arkq, 2020-08-03:
@rodan, please, look at PR #356 This should allow integration with [libopenaptx](https://github.com/pali/libopenaptx) library, which allows aptX and aptX encoding. When creating PR for libopenaptx, I've made only offline verification, because I do not own headset with aptX/HD capabilities. Since you've got DAC capable of aptX/HD decoding, could you please verify this (HD and non-HD variant)?

rodan, 2020-08-04:
I'm speachless. I really am.

you managed to integrate libopenaptx in order to support a protocol you cannot test and it works first-try. congrats.
everything checks out: both the debug log and the capture shows protocol 0xd7:0x24 being selected and the rgb led on the Q5s blinks yellowish-green which means it's connected either with aptX-HD or HWA. the sound is great with no artefacts.
I placed the capture here if you wanna see it: https://subdimension.ro/upload/aptxhd.tar.xz

if you need testing in the future, please feel free to ping me. also, beer is in order for you my friend, so please provide me your paypal address.

my very best regards,
peter

rodan, 2020-08-05:
great news, everyone!

aptX (non-HD) works as well.

logs look good, capture shows vendor id 0x004f (APT ltd), codec 0x0001, DAC blinks purple (aptX/aptX-LL).
link for aptX debug archive: https://subdimension.ro/upload/aptx.tar.xz

more beer can be dispatched if you help me track down why LDAC ain't working :)
I see there's no InterfaceAdded() for the LDAC protocol (which in theory should be also supported by the DAC). can you please tell me which packet id from my captures contain the explicit codecs/capabilities that the DAC is advertising before handshake?

cheers,
peter

Arkq, 2020-08-05:
> I see there's no InterfaceAdded() for the LDAC protocol (which in theory
should be also supported by the DAC). can you please tell me which packet
id from my captures contain the explicit codecs/capabilities that the DAC
is advertising before handshake?

I think, that since some recent version of bluez (I don't remember which),
the endpoint query is cached, so during reconnection there is no packet
with such data. You might try to remove device from bluez, then (maybe
restart bluez just to be sure) start tcpdump and process with pairing and
connection. Maybe you will see desired data. I have not tried this method
by myself, though.

When I've got similar issue, I've simply downgraded bluez (to the version
before SEP support).

janro1, 2020-10-03:
Hi @Arkq,

I'm probably facing a related problem. I've build bluez-alsa from PR #356 and also from master with your newest ffmpeg based openaptx, but in both cases the selected codec when pairing my headphones is SBC. My Headphones supporting SBC, AAC, aptX, aptX HD and LDAC, but only SBC support seems to be identified. Same happens when I build bluez-alsa with LDAC support.

Here are some [logs](https://pastebin.com/QsSE0nvv). Looking into the code I see that in `bluealsa_pcm_get_codecs` the variable  `seps` is null, but I had not enough time to find the reason for it.

Arkq, 2021-07-31:
Support for aptX via libopenaptx is already in master. I'm closing this issue, then.

> My Headphones supporting SBC, AAC, aptX, aptX HD and LDAC, but only SBC support seems to be identified. 

If your headset is a brand of Sony, then you will have to enable "audio quality mode" in smartphone Sony's app. Otherwise, the default, is "audio stability" which disables all codecs except SBC.

> the variable seps is null

You need BlueZ >= 5.52 in order to be able to select SEPs. For lower versions of BlueZ, the sep variable will be NULL.

closed 2021-07-31 by Arkq