arkq/bluez-alsa #357: bluealsa and mpv

nsklaus, 2020-08-13:
i have two issues i cannot find a solution for. 
the first one is: i am forced to always use sudo if i want bluealsa to output anything.

and the second is: bluealsa works for me like this:
`sudo aplay -D bluealsa:SRV=org.bluealsa,DEV=2C:41:A1:83:CE:22,PROFILE=a2dp ../Wav_868kb.wav`
but from there i have no idea how i can use mpv with bluealsa.
also i am not sure how to make proper ~/.asoundrc
i tried to copy what the others did, and put my bluetooth headset address 2C:41:A1:83:CE:22
but that does not work. the only thing that work for me is aplay -D bluealsa as shown above.

i can list connected bluetooth device
```
$ sudo bluealsa-aplay -l
[sudo] password for klaus:
**** List of PLAYBACK Bluetooth Devices ****
hci0: 2C:41:A1:83:CE:22 [hqc35], trusted audio-card
  A2DP (AAC): S16_LE 2 channels 48000 Hz
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 2C:41:A1:83:CE:22 [hqc35], trusted audio-card
  SCO (CVSD): S16_LE 1 channel 8000 Hz
```

i can get mixer control for bluetooth
( sudo alsamixer -D bluealsa:SRV=org.bluealsa <-- this also works )
```
$ sudo amixer -D bluealsa scontrols
[sudo] password for klaus:
Simple mixer control 'hqc35 - A2DP',0
Simple mixer control 'hqc35 - SCO',0
```

my ~/.asoundrc i tried to make, looking at what other people had:
```
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "2C:41:A1:83:CE:22"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000

pcm.!default {
  type plug
  slave {
    pcm {
      type bluealsa
      device "2C:41:A1:83:CE:22"
      profile "a2dp"
#      delay 1000
    }
  }
  hint {
    show on
    description "Mpow Judge"
  }
}
```

but from there i don't understand what to do.
my asoundrc is probably wrong and i don't know how to correct it, 
and i have no idea how to use mpv with bluealsa.

borine, 2020-08-13:
> i am forced to always use sudo if i want bluealsa to output anything

Either user `klaus` is not a member of the group `audio`, or the bluealsa installation is incomplete, so group `audio` has not been granted necessary D-Bus permissions. Please post output of:
```
groups klaus
```
and
```
cat /etc/dbus-1/system.d/bluealsa.conf
```

nsklaus, 2020-08-13:
in /etc/group i can see:
```
audio:x:994:klaus
```

$ cat /etc/dbus-1/system.d/bluealsa.conf
```
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

</busconfig>
```

borine, 2020-08-13:
OK, I think (judging by the edits to your last comment) you just need to reboot - that will refresh both the group and dbus status as seen by all processes (not just your login shell); then everything should work OK.

I am not an mpv user, but hopefully some else can help you with that.

nsklaus, 2020-08-13:
but, should i put something at all in ~/asoundrc ? or what i already have is correct ?

borine, 2020-08-13:
I recommend to reboot, re-connect the bluetooth device, then try:
```
aplay Wav_868kb.wav
```
If it works, your .asoundrc is OK. If it does not work then post any error message here.

nsklaus, 2020-08-13:
here is what i'm getting:
`aplay -D bluealsa:SRV=org.bluealsa,DEV=2C:41:A1:83:CE:22,PROFILE=a2dp ../Wav_868kb.wav` works 
`aplay Wav_868kb.wav` doesn't work

closed 2020-08-13 by nsklaus