arkq/bluez-alsa #358: Error compiling bluealsa

mackowiakp, 2020-08-13:
I try to compile bluealsa according to the procedure:

`cd /home/guido
git clone https://github.com/Arkq/bluez-alsa.git
cd bluez-alsa/
autoreconf --install
mkdir build && cd build
../configure --disable-hcitop --with-alsaplugindir=/usr/lib/arm-linux-gnueabihf/alsa-lib
make && sudo make install `

 but I got an error:

`*******

checking for ALSA... no
configure: error: Package requirements (alsa) were not met:

No package 'alsa' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables ALSA_CFLAGS
and ALSA_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details`

Maybe because the name of alsa package is named in my system as alsa-base:

`root@NanoPi-NEO:~# apt list|grep alsa

*********

alsa-base/xenial,now 1.0.25+dfsg-0ubuntu5 all [installed]

*********`

How can I resolve that problem?

borine, 2020-08-13:
On Debian based systems, the alsa package you require is called `libasound2-dev`. See the wiki article for information on required packages: https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

mackowiakp, 2020-08-13:
Will try when I back home.
BTW. Is it possible to use aplay command concurrently for streaming radio via Bluetooth and reproduce annunciations concerning home automatic via wire connected amplifier using onboard audio chip?

mackowiakp, 2020-08-13:
OK, WORKS !!! THX !