arkq/bluez-alsa #359: Unable to install hw params (80db29f)

hrc-amilcar, 2020-08-17:
Hello,

I've updated from e5acb73 to 80db29f and now I'm unable to play to SCO with error "aplay: set_params:1382: Unable to install hw params"

I noticed this error from bluealsa in the logs:
**src/sco.c:507: PCM poll error status: 0x10**

Here is the output from bluealsa from 80db29f:
```
./bluealsa: D: src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/1
./bluealsa: D: src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
./bluealsa: D: src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/2
./bluealsa: D: src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/2
./bluealsa: D: src/bluez.c:785: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
./bluealsa: D: src/bluez.c:719: Registering hands-free profile: /org/bluez/HSP/AudioGateway
./bluealsa: D: src/bluez.c:785: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
./bluealsa: D: src/bluez.c:719: Registering hands-free profile: /org/bluez/HFP/AudioGateway
./bluealsa: D: src/main.c:397: Acquiring D-Bus service name: org.bluealsa
./bluealsa: D: src/main.c:402: Starting main dispatching loop
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/dbus.c:57: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
./bluealsa: D: src/ba-rfcomm.c:1263: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
./bluealsa: D: src/sco.c:200: Created SCO dispatcher [ba-sco-dispatch]: hci0
./bluealsa: D: src/bluez.c:611: HFP Audio Gateway (CVSD) configured for device E4:22:A5:8B:35:06
./bluealsa: D: src/ba-transport.c:750: Starting transport: HFP Audio Gateway (CVSD)
./bluealsa: D: src/sco.c:77: Starting SCO dispatcher loop: hci0
./bluealsa: D: src/sco.c:241: Starting SCO loop: HFP Audio Gateway (CVSD)
./bluealsa: D: src/ba-rfcomm.c:880: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
./bluealsa: D: src/ba-transport.c:1032: Created new thread [ba-sco]: HFP Audio Gateway (CVSD)
./bluealsa: D: src/at.c:161: AT message: SET: command:+BRSF, value:25
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
./bluealsa: D: src/at.c:161: AT message: TEST: command:+CIND, value:(null)
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
./bluealsa: D: src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
./bluealsa: D: src/bluez.c:999: Adding new Stream End-Point: E4:22:A5:8B:35:06: SBC
./bluealsa: D: src/at.c:161: AT message: GET: command:+CIND, value:(null)
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 5 -> 7
./bluealsa: D: src/dbus.c:57: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
./bluealsa: D: src/at.c:161: AT message: SET: command:+CMER, value:3, 0, 0, 1
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 7 -> 8
./bluealsa: D: src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 8 -> 9
./bluealsa: D: src/dbus.c:57: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/SBC/Source/1
./bluealsa: D: src/at.c:161: AT message: SET: command:+VGS, value:08
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/bluez.c:915: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
./bluealsa: D: src/dbus.c:57: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/Source/1
./bluealsa: D: src/a2dp.c:665: Selected A2DP SBC bit-pool range: [2, 53]
./bluealsa: D: src/bluez.c:293: A2DP Source (SBC) configured for device E4:22:A5:8B:35:06
./bluealsa: D: src/bluez.c:295: Configuration: channels: 2, sampling: 48000
./bluealsa: D: src/bluez.c:490: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/3
./bluealsa: D: src/bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/3
./bluealsa: D: src/at.c:161: AT message: SET: command:+VGM, value:15
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/at.c:161: AT message: SET: command:+XAPL, value:047F-22-6b,1
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,6
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
./bluealsa: D: src/at.c:161: AT message: GET: command:+BTRH, value:(null)
./bluealsa: D: src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
[... 132 lines cut ...]
```
Here is the output from aplay using e5acb73:
```
aplay --verbose -Dbluealsa:SRV=org.bluealsa,DEV=E4:22:A5:8B:35:06,PROFILE=sco -fS16_LE -r8000 -c1 test.raw
D: src/asound/bluealsa-pcm.c:671: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Setting constraints
Playing raw data 'test.raw' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
D: src/asound/bluealsa-pcm.c:368: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Initializing HW
D: src/asound/bluealsa-pcm.c:394: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:400: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Initializing SW
D: src/asound/bluealsa-pcm.c:437: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Prepared
D: src/asound/bluealsa-pcm.c:415: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Initializing SW
Plug PCM: BlueALSA PCM: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink
BlueALSA BlueZ device: /org/bluez/hci0/dev_E4_22_A5_8B_35_06
BlueALSA Bluetooth codec: CVSD
D: src/asound/bluealsa-pcm.c:285: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Starting
D: src/asound/bluealsa-pcm.c:143: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Starting IO loop: 8
D: src/asound/bluealsa-pcm.c:327: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Stopping
D: src/asound/bluealsa-pcm.c:111: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: IO thread cleanup
D: src/asound/bluealsa-pcm.c:327: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Stopping
D: src/asound/bluealsa-pcm.c:407: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Freeing HW
D: src/asound/bluealsa-pcm.c:357: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Closing

```

borine, 2020-08-18:
This may be a bug in the new bluealsa pcm plugin code. Please can you post your alsa-lib version:
`aplay --version`
 
using 80db29f , what happens if you omit the args `-fS16_LE -r8000 -c1` from the aplay command line?

borine, 2020-08-18:
I've tested with alsa-lib 1.1.3 and alsa-lib 1.1.8 and am unable to reproduce this error

Arkq, 2020-08-18:
With alsa-lib 1.2.2 I was also not able to reproduce it...

hrc-amilcar, 2020-08-18:
Version:
```
# aplay --version
aplay: version 1.1.4 by Jaroslav Kysela <perex@perex.cz>
```

Playing a 44.1Khz WAV instead of 8Khz RAW:
```
aplay --verbose -Dbluealsa:SRV=org.bluealsa,DEV=E4:22:A5:8B:35:06,PROFILE=sco Tone_1kHz_Amplitude1_16.wav                          
D: src/asound/bluealsa-pcm.c:761: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Setting constraints
Playing WAVE 'Tone_1kHz_Amplitude1_16.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Mono
D: src/asound/bluealsa-pcm.c:397: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Initializing HW
D: src/asound/bluealsa-pcm.c:417: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: FIFO buffer size: 4096
D: src/asound/bluealsa-pcm.c:426: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: src/asound/bluealsa-pcm.c:441: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Initializing SW
aplay: set_params:1382: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 44100
PERIOD_TIME: 125000
PERIOD_SIZE: (5512 5513)
PERIOD_BYTES: (11024 11026)
PERIODS: (3 5)
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 44100
TICK_TIME: 0
D: src/asound/bluealsa-pcm.c:433: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Freeing HW
D: src/asound/bluealsa-pcm.c:386: /org/bluealsa/hci0/dev_E4_22_A5_8B_35_06/hfpag/sink: Closing
```
I've also tried a different device altogether with same result.

borine, 2020-08-18:
very strange .. I've just tested with alsa-lib 1.1.4 and get no error.

From your aplay debug, it appears that bluealsa_sw_params() is returning an error code; but that should not be possible from with snd_pcm_hw_params() as the buffer size is guaranteed to be bigger than the period size, and that is the only case that causes non-zero return from bluealsa_sw_params().

I am confused !

borine, 2020-08-18:
@hrc-amilcar can you step through the aplay run in a debugger? I would like to know the path taken leading up to bluealsa_sw_params() being called, and the values of params->avail_min, pcm->io_avail_min, io->buffer_size and io->period_size inside the call to bluealsa_sw_params(). As I am unable to reproduce the error, it is impossible for me to understand what is wrong here. 

The function call tree when aplay calls snd_pcm_hw_params() is (or should be):
```
snd_pcm_hw_params()
	_snd_pcm_hw_params_internal()
		snd_pcm_plug_hw_params()
			_snd_pcm_hw_params_internal()
				snd_pcm_ioplug_hw_params()
					bluealsa_hw_params()
				snd_pcm_sw_params_default()
				snd_pcm_sw_params()
					snd_pcm_ioplug_sw_params()
						bluealsa_sw_params()
		snd_pcm_sw_params_default()
		snd_pcm_sw_params()
			snd_pcm_ioplug_sw_params()
				bluealsa_sw_params()
	snd_pcm_prepare()
		snd_pcm_ioplug_prepare()
			bluealsa_prepare()
```
as you can see, bluealsa_sw_params() should be called twice before snd_pcm_hw_params() returns, but snd_pcm_hw_params() is returning before the second call is made.

hrc-amilcar, 2020-08-20:
@borine, this is a custom embedded device running Yocto with pre-built alsa-lib and utils. I've been trying to get gdb working to get you that info, but I end up stepping around blindly in disassembly, trying to find the call to `snd_pcm_hw_params`.

I manually built `aplay` v1.1.4 in debug and can modify the code to print out values, but building alsa-lib for the embedded device is proving difficult because of missing dependencies in its toolchain.

I can keep plugging away at trying to build a debug version of alsa-lib for the device, but that may take a while.

Do you have any other debugging recommendations? I'd really like to try out your improved bluealsa-pcm module!

borine, 2020-08-21:
> Do you have any other debugging recommendations?

If using gdb is not feasible, you could try to put some more logging statements into the bluealsa plugin code. To get some idea of what is happening in the function bluealsa_sw_params(), edit the file src/asound/bluealsa-pcm.c, and insert a new debug2() statement at line 447:
```
static int bluealsa_sw_params(snd_pcm_ioplug_t *io, snd_pcm_sw_params_t *params) {
	struct bluealsa_pcm *pcm = io->private_data;
	debug2("Initializing SW");

	snd_pcm_sw_params_get_boundary(params, &pcm->io_hw_boundary);

	snd_pcm_uframes_t avail_min;
	snd_pcm_sw_params_get_avail_min(params, &avail_min);
debug2("params->avail_min=%zu, pcm->io_avail_min=%zu, io->buffer_size=%zu,  io->period_size=%zu", avail_min, pcm->io_avail_min, io->buffer_size, io->period_size);

	if (avail_min != pcm->io_avail_min) {
		if (avail_min > io->buffer_size)
			return -EINVAL;
		debug2("Changing SW avail min: %zu -> %zu", pcm->io_avail_min, avail_min);
		pcm->io_avail_min = avail_min;
	}

	return 0;
}
```

borine, 2020-08-21:
> this is a custom embedded device running Yocto

... just a thought - are you compiling bluealsa on a different machine? Perhaps the alsa header files on the build machine are not compatible with the alsa library on the runtime machine?