arkq/bluez-alsa #363: [Feature Request] save typing with alsa plugin default changes

borine, 2020-08-24:
This is for discussion: 2 potential changes to the pcm plugin inspired by the bluealsa-aplay change in commit df8a75ba48c345276a21ed085f057b027123b96b

**1. pcm.bluealsa parameters order**
    alsa allows arguments to be bound by order, as well as by name. so for example:
    `bluealsa:DEV=00:11:22:33:44:55,PROFILE=a2dp`
    is equivalent to
    `bluealsa:org.bluealsa,00:11:22:33:44:55,a2dp`
   but note the need to always specify SRV when using the "by-order" method. A simple change to 20-bluealsa.conf would make it much easier to type out bluealsa pcm names - just list the parameters in a different order, as service is the least likely to vary from its default:
   ```
   pcm.bluealsa {
	@args [ DEV PROFILE DELAY SRV ]
   ```
   Now `bluealsa:00:11:22:33:44:55,a2dp` or even `bluealsa:00:11:22:33:44:55` is all that is required.

**2. implement "any" as default device when none or all zeros is specified**
   Just like bluealsa-aplay has default "all", pcm.bluealsa could have default "any" for `DEV=00:00:00:00:00:00`. In this case the first matching pcm returned by GetPCMs() would be selected (a2dp or sco as specified by the user or defaulted). So then
   ```aplay -D bluealsa:DEV=00:00:00:00:00:00```
   or, in combination with item 1 above:
   ```aplay -D bluealsa```
   would select the first a2dp playback pcm returned by GetPCMs(). This would be most useful when only 1 such device is connected, but I think that is possibly the most common case.

Arkq, 2020-08-24:
> 1. pcm.bluealsa parameters order

I like this idea! @borine can you prepare PR for this (params reorder and maybe some info in readme).

> 2. implement "any" as default device when none or all zeros is specified

Specifying "any" MAC address seems more like "play to all connected devices" (at least for me, or maybe I've messed up nomenclature here... because in bluetooth.h there is also BTADDR_ALL, which is all FF) - something like broadcast from BT LE audio specification. Once I thought about implementing "broadcast" feature, but ALSA ´route´ plugin works as well (with BT audio 4.0 - current bluealsa). Also it would be nice if indeed firstly connected BT device would be selected, but right now it will be non-deterministic - GetPCMs iterates over hashmap. So maybe GetPCMs should return devices in order of connection? 

Anyway, if you think that this feature will be beneficial to others, we can go with it as well :)

graham8, 2020-08-24:
1.  I've always wondered why SRV is needed at all.  Under what circumstances would it be different from org.bluealsa?
2.  Good idea.  It should not be necessary to type out 00:00:00:00:00:00 for either bluealsa or bluealsa-aplay.  AFAIK streaming to multiple devices doesn't happen unless there are some smarts in the sending device.  Otherwise, I think it's normal for the _last_ device to attempt to connect to grab the connection.  I may be wrong.

borine, 2020-08-24:
> can you prepare PR for this (params reorder and maybe some info in readme).

will do.

As for item 2, I think I'll consider it a bit longer. My first thought was that the default should be non-deterministic - If the user wants a specific device they should specify it explicitly. I can see now that either the first or last connected device (with the correct profile) would also make sense, but would require server-side changes. I'm not keen on the "all devices" default, as they would not be in sync so this option is most unlikely to be what a typical user wants. I may come back to this suggestion, but for now I'll leave it as a discussion topic.

Arkq, 2020-08-24:
> AFAIK streaming to multiple devices doesn't happen unless there are some smarts in the sending device.

Agree, broadcast mode is not something that most ppl will use.

> Otherwise, I think it's normal for the last device to attempt to connect to grab the connection.

+1 for *last* device, but also I'm wondering how such problem is resolved in some commercial products (beside these ones where it is not possible to connect more than one BT device :D).

> I can see now that either the first or last connected device (with the correct profile) would also make sense, but would require server-side changes.

Making server-side change is not a problem as long as it is backward compatible. And even if it's not, right now is the window for breaking backward compatibility :) I was planing to tag new release in the next few days, but I can wait until this feature request is finalized.

graham8, 2020-08-24:
Just an FYI, I have a JBL Flip 3 which offers the function to use them in pairs for stereo.  There's a smartphone app that does this trick. They discontinued that model before I could save up for a second one so I've never tried it.  I think there are other BT speakers that offer this stereo mode but I've no idea how it's achieved.

borine, 2020-08-25:
I've raised PR #364 for changing the order of pcm (and ctl) parameters, with very slight change to the wording in README.md

I am working (slowly) on more detailed documentation for the alsa plugins and will add that to the PR when ready for review. As the documentation is for this project site, and not a distributed file, it should not affect the timing of the next release.

For coding changes to the server, I need larger blocks of free time in which to concentrate, and that is unlikely to happen for several weeks yet. So I would suggest to leave that out for this next release.

To avoid breaking backward compatibility, instead of modifying GetPCMs() we could introduce a new method, such as `GetLastPCM(stream, profile)` to return the most recently connected pcm of the specified type. That way the change would not affect existing clients so could be put out as a minor update release.

Arkq, 2020-08-25:
> 1. I've always wondered why SRV is needed at all. Under what circumstances would it be different from org.bluealsa?

(I forgot to answer that before) It is possible to have multiple HCI interfaces on the system. In such case, by default, bluealsa binds to all of them. However, one can specify which HCI should be controlled by bluealsa instance. Then one can run multiple number of bluealsa instances with different D-Bus service name like follows:
```
bluealsa -B sink -i hci0 -p a2dp-sink &              # org.bluealsa.sink
bluealsa -B source -i hci1 -i hci2 -p a2dp-source &  # org.bluealsa.source
```

Buuuuuuttttttttt, there is one drawback. BlueZ currently does not allow to register HSP/HFP handler for a specific HCI.... So, if one would like to use hci0 for HFP and hci1 for A2DP only, it will not be possible, because `-p hfp-ag` will bind to all available HCI interfaces (in BlueZ) - it will be reported to BT devices that hci0 and hci1 support HFP audio gateway.

Such feature would also allow to use e.g. one HCI with bluealsa and other with e.g. puleaudio (if pulseaudio would not bind to all interfaces). This feature is similar to every other well designed network utility, which allow to bind to only one iface if desired.

> To avoid breaking backward compatibility, instead of modifying GetPCMs() we could introduce a new method, such as GetLastPCM(stream, profile) to return the most recently connected pcm of the specified type.

I think, that it is not required. Right now  GetPCMs() returns elements in non-deterministic order, making it deterministic will not break anything (e.g.: ALSA control plugin sorts elements by name by itself). Other more simple solution might be to add "counter" value (e.g. connection timestamp), then client could sort PCMs and selects last of first.

borine, 2021-04-18:
I've made a tentative stab at having the most recently connected device be available to ALSA clients as bluealsa:00:00:00:00:00:00
The code is in a branch on my fork here: https://github.com/borine/bluez-alsa/tree/default-pcm
@Arkq if you still think this would be a worthwhile feature, I can raise a PR; although it may be best to wait until your current refactoring work is complete.

Arkq, 2021-04-19:
@borine I think that this feature and my refactoring do not have any conflicts between them, so you may safely create PR.

commit 04d90432 (2021-04-19): Use most recently connected PCM device as default

Introduces new PCM property "Sequence" and uses this to identify the
most recently connected device. Clients can then use 00:00:00:00:00:00
MAC address to select that device.

Resolves #363 and closes #435
closed 2021-04-19 by Arkq